OpenCPN/OpenCPN #2186: Build errors on Win10 in VS2017

Hakansv, 2021-03-14:
After last commits to master I can no longer build in VS2017 on one of my machines. (Another is working)
Any hints what to look for?
These are the errors:
```
20>ser_ports.cpp
20>C:\Builds\OCPN\OpenCPN\src\multiplexer.cpp(28): fatal error C1083: Cannot open include file: 'libgen.h': No such file or directory
20>C:\Builds\OCPN\OpenCPN\src\ser_ports.cpp(52): fatal error C1083: Cannot open include file: 'dirent.h': No such file or directory
20>Done building project "opencpn.vcxproj" -- FAILED.
```

[image]

Hakansv, 2021-03-14:
If connected?
These "find_path..." are new in CmakeList: (dirent.h itself is nothing new,) 
find_path(HAVE_UNISTD_H NAMES unistd.h)
find_path(HAVE_LIBGEN_H NAMES libgen.h)
find_path(HAVE_DIRENT_H NAMES dirent.h)

leamas, 2021-03-14:
Yes, they are new. last large commit, re-factoring the serial ports code.

Datapoint: It builds on appveyor, so it's not broken on Windows per se.

One thing I can think of is that cmake's  find_path() finds it in some odd location and flags it as existing. And that the standard include path used later in the build doesn't find it. TBH, this is not a scenario I have considered until now. 

Thoughts?

Hakansv, 2021-03-14:
Is there a cmake command I can test to echo what find_path() found?

leamas, 2021-03-14:
You can add` message(STATUS "Dirent.h  found in ${HAVE_DIRENT_H}")` after the `find_path(HAVE_DIRENT_H NAMES dirent.h)`

Hakansv, 2021-03-14:
OK. Nothing OpenCPN would utilize I suppose?
-- libgen.h found in C:/MinGW/include
-- Dirent.h found in C:/MinGW/include
So we better make some more definite search if possible?

Hakansv, 2021-03-14:
Actually I don't remember why I have C:/MinGW but it explains why it was working on another machine i.e without any C:/MinGW installed.

leamas, 2021-03-14:
This is actually not that easy. Obviously, current situation isn't what we want. Some possible remedies:

  - We teach cmake to add all paths containing used headers to the include paths. It will probably compile, but not link.
  - You determine that your current setup is not suitable for building opencpn, and moves mingw out of the way.
  - We decide that windows never can have these headers. Sort of unsatisfactory, I assume there are implementations available also for windows out there...

Looking at https://cmake.org/cmake/help/v3.17/command/find_path.html it seems that there are several cmake variables affecting the search like CMAKE_INCLUDE_PATH, CMAKE_FRAMEWORK_PATH, CMAKE_FIND_USE_CMAKE_SYSTEM_PATH etc. Can you fix the build using some sane combination of these that would certainly be the best.

Hakansv, 2021-03-14:
Now cmake commands are not my "table" so far. 
But something like CMAKE_CURRENT_SOURCE_DIR may be suitable?
How would a edited: find_path(HAVE_DIRENT_H NAMES dirent.h) look in that case? 
I will of course make some tests but tomorrow. It has been a long day today.
God natt och tack så länge
Håkan

leamas, 2021-03-14:
Yes, it 's late. That said, the more I think about this I think cmake does it's job. How can it know that exact that sample of dirent.h shouldn't be taken account for?

When I looked at the docs I *think* it should be possible to clear all default search paths used by cmake using some of the variables, and then "re-add" those you think contains valid headers using other variables. If possible, this should be the right way. After all only you, the user, knows where sane headers exist. We really cannot make much assumptions here... 
.

leamas, 2021-03-15:
> But something like CMAKE_CURRENT_SOURCE_DIR may be suitable?
How would a edited: find_path(HAVE_DIRENT_H NAMES dirent.h) look in that case?

No, this does not fly. cmake's very task here is to locate libraries outside the source tree. So CMAKE_CURRENT_SOURCE_DIR doesn't really make sense here. 

The issue is that your system exposes headers which looks like they are installed, and cmake takes the bait. OTOH, having mingw installed is certainly not unique and we need to figure out how to cope with it.

Hakansv, 2021-03-15:
Understood!
But... (I've not investigated what we do with these headers)
If I comment out:
#find_path(HAVE_LIBGEN_H NAMES libgen.h)
#find_path(HAVE_DIRENT_H NAMES dirent.h)
All builds works fine and I see no runtime trouble so far.
Do we need them on WIN32?

leamas, 2021-03-15:
> Do we need them on WIN32?

There are arguments against just disabling them on Windows. One is that there certainly are working  implementations of dirent.h on Windows, and using it a user can achieve other things. It might be possible to add or more options disabling these, but it might become messy. If possible, we should keep the code clean (which it actually is now).

If your experiments fails, we can certainly do a patch. But using the code as it is, just changing some cmake parameters,  would be the best. If the setup is complicated. we could certainly wrap it in a single option.

leamas, 2021-03-15:
From another POV: the problem here is that cmake searches in directories you don't want it to scan. As long as this situation persists, there could be other surprises. The proper fix would be to limit the directories cmake scans, and this is what these variables are all about.

Hakansv, 2021-03-15:
I may find a way to limit directories to not find e.g dirent.h on my system. like a simple NOT C:\MinGW\, but since I've no knowledge about the header I don't know about a plausible location where to find a useful header on "every" system.
And I'm not dexterous enough with this to make clever tries. I can though see this code builds fine on a couple of RPi's so different tests I could very well perform. 
Sorry for a "hide behind" answer?

leamas, 2021-03-15:
Right. Is this a blocking error for you now, or can you move around it by moving/removing c:\\mingw ? I'm just a bit overloaded, and would need to postpone this a little...

leamas, 2021-03-15:
Or, of course, you can of course just patch CMakeLists.txt until we have a proper fix?

Hakansv, 2021-03-15:
No problem for me at all. I know how to handle it for myself. My concerns is only to not let it out when released. Future home brewers may be confused. 
I'm sure you will find a nice solution when suitable. I'll be happy to test whatever you may need. 
I let this issue open but on halt?
Thanks

Hakansv, 2021-04-12:
This may not be a common trouble. I know how to handle locally. Thanks for the discussion.

closed 2021-04-12 by Hakansv

leamas, 2021-04-12:
It's on my agenda anyway, closed or open...

bdbcat, 2021-09-30:
As a practical workaround, I patched CMakeLists to disable these checks on "non-UNIX" platforms.
15636794348825e2c2b7b28de060650d70a1db36

Hakansv, 2021-09-30:
Double close  ;)
Thanks

leamas, 2021-09-30:
hm... I'm obviously not happy with Dave's patch for reasons described above. OTOH, this has been my desk and I have forgotten it. I might file a new bug and look into somewhat more sane solution, taking into account that the core issue here is not about Windows or VS but rather an inconsistent (but not uncommon ) developer setup.

We should IMHO start assigning bugs so we (I) don't miss things like this.

bdbcat, 2021-09-30:
This patch was prompted by a real live "inconsistent (but not uncommon ) developer setup."
https://www.cruisersforum.com/forums/f134/first-time-build-on-windows-256187.html

bdbcat, 2021-09-30:
IMHO, the issue relates largely to a commitment to full cross-platform functionality.
The patch which introduces the problem is mainly meant to implement some functionality that simply does not, and likely never will, exist on Windows MSVC (e.g. accessing serial ports by FD).
We can face that fact, and embrace "#IFDEF"  as a legal and honorable solution.  
An alternative to conditionals in CMakeLists.text is a fully abstracted OCPN library for these kinds of functions, which itself will contain the conditional instructions.
But the conditional build logic will exist somewhere.

leamas, 2021-09-30:
The build should work, agreed. And if I want to make a more clean patch, I'd have to propose with some code. Talk is cheap.,I shouldn't talk more on this. Might or might not come up with some code...