OpenCPN/OpenCPN #2192: command to update all plugins to latest needed

mgrouch, 2021-03-31:
There is a need for a command line utility to update all plugins to latest.
(Without running opencpn and non-interactively).

Does it exists?

Thanks,
--MG

nohal, 2021-03-31:
No, it does not. Why is there a need for it? Such a thing can be scripted pretty easily using eg. Python, but what is the usecase?

mgrouch, 2021-03-31:
the use case is to update all plugins
from a script non-interactive

nohal, 2021-03-31:
But what is the usecase for updating the plugins of a GUI program from a script non-interactively? Can you give me an example of other programs implementing this, please?

mgrouch, 2021-03-31:
Pretty much all the software I know GUI or no has support for unattended non-interactive updates.

leamas, 2021-03-31:
The vast majority of OpenCPN users, most of which on Windows, certainly don't expect *any*  CLI functionality. This also applies to Linux and MacOS end-users.

Still, this issue could be investigated in other directions:
- Is an "Update All Plugins" GUI button motivated?  I'd say no until we have user reports indicating such a need.
- Is there a need to manipulate plugins from the CLI? I have considered the need to make sure that a plugin actually can be loaded by OpenCPN as an automated, build-time check.

Hakansv, 2021-03-31:
Alec..
re: "I have considered the need to make sure that a plugin actually can be loaded by OpenCPN as an automated, build-time check."
I've got requests about such a reminder/invitation for OpenCPN itself. So if your clever mind is touching these areas.........?
Sincerely yours. 
Håkan

rgleason, 2021-03-31:
Plugin manager has an update catalog button. 
I think Hakan's suggestion comes from the  Android builds which tend to be less successful with no good easy  way to test.

Hakansv, 2021-03-31:
Rick..
No, my humble request is to check also for an OpenCPN update e.g. when the update plugin catalog is running.  There are a lot users out there running an old version.

leamas, 2021-03-31:
I had actually other things in mind i. e., that plugin can be *loaded* successfully. Not downloaded.

 @Hakansv: perhaps you should file a bug against the plugins project, they are responsible for the catalog

rgleason, 2021-03-31:
I think Alec's idea is excellent because we do expect the plugin to work, and it is very difficult to check each plugin environment by the plugin dev as they may not run those environments normally.

Hakan's is a friendly iconic suggestion in the Plugin Manager interface that there is an updated plugin available.  That would be useful, but users need to be familiar with the update catalog button too.

mgrouch, 2021-03-31:
> But what is the usecase for updating the plugins of a GUI program from a script non-interactively? Can you give me an example of other programs implementing this, please?

> The vast majority of OpenCPN users, most of which on Windows, certainly don't expect _any_ CLI functionality. This also applies to Linux and MacOS end-users.
>
> Still, this issue could be investigated [...]

I’m a user of OpenCPN and I do need to update it non-interactively and I need to be able to update plugins from scripts. 
Moreover how do you automate tests
if even installation and updates can’t be scripted?
Thanks

CarCode, 2021-04-01:
It is very easy to implement an updater for macOS with Sparkle.framework. It is available for Windows also.
[image]

rgleason, 2021-04-22:
Issues is not for feature requests.  Please close this and Enter it in Tracker. Thanks.

mgrouch, 2021-05-07:
I’m trying to understand what needs to be done to script it. 
I can script downloads for all plugins for a given architecture. 
What needs to happen after that with .tar.gz file? Assuming OpenCPN is not running. 
I do not see them exploded after install. I do see some other files in
~/.opencpn/install_data. 
is populating plugins/install_data and  plugins/cache/tarballs enough?
What needs to go into install_data?

thanks

rgleason, 2021-05-07:
mgrouch   The newer Dev Manual that alec mike and I have been working on may help. https://rgleason.github.io/ocpn-manuals/ocpn-dev-manual/5.3.1/intro-AboutThisManual.html

Alec also has written a lot about the inner workings here
https://rgleason.github.io/ocpn-manuals/plugin-installer/0.0.1/Home.html

Which I believe should help with destination paths for the plugins and various OS.

rgleason, 2021-05-07:
Mike, I will let Alec answer that question, because he thought quite a bit about how to do this,  but there is also a cache for tarballs, forexample in windows that is located here   C:\ProgramData\opencpn\plugins\cache

nohal, 2021-05-07:
Because they are thirdparty software and there is no infrastructure and manpower to manage it. They sure could be stored exactly the way you outline and it would also work. If there was someone available and willing to do the work you have to do yourself now. There is not such a person.

mgrouch, 2021-05-07:
/usr and /use/local prefixes are removed when copied to .local?

what happens with /etc?

I do not see it under .local for oesenc
But it is there in tar
Is it simply skipped?

Is there a contract on location on internet of the main catalog?

What should be correct URL?

Thanks

nohal, 2021-05-07:
I am not sure what you mean by a "contract on location on internet of the main catalog", but the catalogs reside in https://raw.githubusercontent.com/OpenCPN/plugins/@BRANCH@/ocpn-plugins.xml managed in the https://github.com/OpenCPN/plugins project (replace `@BRANCH@` with the respective branch you are interested in)

I am afraid I do not understand the rest of your question neither as a randomly selected plugin tarball from a sane developer does not conatain anything you talk about:

```
# tar zxvf Radar-5.2.1-+2104111315.3d2de39_raspbian-10-armhf.tar.gz
x Radar-5.2-raspbian-A64-10/
x Radar-5.2-raspbian-A64-10/lib/
x Radar-5.2-raspbian-A64-10/lib/opencpn/
x Radar-5.2-raspbian-A64-10/lib/opencpn/libradar_pi.so
x Radar-5.2-raspbian-A64-10/share/
x Radar-5.2-raspbian-A64-10/share/opencpn/
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/alarm.wav
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/radar_hidden.svg
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/radar_active.svg
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/radar_searching.svg
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/radar_standby.svg
x Radar-5.2-raspbian-A64-10/share/opencpn/plugins/radar_pi/data/radar_seen.svg
x Radar-5.2-raspbian-A64-10/share/locale/
x Radar-5.2-raspbian-A64-10/share/locale/pt_BR/
x Radar-5.2-raspbian-A64-10/share/locale/pt_BR/LC_MESSAGES/
x Radar-5.2-raspbian-A64-10/share/locale/pt_BR/LC_MESSAGES/opencpn-radar_pi.mo

...

x Radar-5.2-raspbian-A64-10/share/locale/et_EE/
x Radar-5.2-raspbian-A64-10/share/locale/et_EE/LC_MESSAGES/
x Radar-5.2-raspbian-A64-10/share/locale/et_EE/LC_MESSAGES/opencpn-radar_pi.mo
x Radar-5.2-raspbian-A64-10/metadata.xml
```

mgrouch, 2021-05-07:
oesenc plugin has /etc directory which goes nowhere apparently 

twocan has /usr prefix inside tar

nohal, 2021-05-07:
Feel free to open issues on them, IMO there should not be any irrelevant stuff in the tarballs. But if there is, it is an issue with the respective plugin, not with OpenCPN or it's plugin system, so it is out of scope of any discussion here.

leamas, 2021-05-07:
The /usr prefix is fine, it is handled (removed). /etc stuff is dropped. There is some more on these things at https://github.com/leamas/OpenCPN/wiki/Tarballs.

To download all plugins one need to download the catalog and parse it. Getting the urls from the parsed data structure is trivial. There is example code  (javascript) at https://github.com/leamas/ocpn-download.

As of now, stuff like /etc which is dropped goes into UserUnknownPrefixDir, see #1567. So, it's not really lost but rather unused.
.

leamas, 2021-05-07:
> Feel free to open issues on them, IMO there should not be any irrelevant stuff in the tarballs.

Indeed. Cleaning up such stuff might be a task for next cycle.

mgrouch, 2021-05-24:
I looked into scripting it. However there are too many cases for handling /etc, having /usr prefix or not, etc. Catalog xml data quality is also an issue. Choosing latest version from it and proper binary for a given platform for example. I can’t see making a script for it under these circumstances.

leamas, 2022-04-10:
@mgrouch: yes, scripting this is not really a viable path. Once upon a time there was such scripts, basically test tools, but they were not possible to maintain.

The fast track here is probably to add "Update all plugins" button to the pluginmanager. _pluginmanager.cpp_ has all information about installed plugins and available updates. It has also access to the actual installation routines, mostly in PluginHandler. I assume that if anyone made a PR based on this it would have a chance to be accepted.

A more ambitious approach would be to split the current codebase in two parts, one low-level stuff including the installation routines and one which perhaps could be called GUI but practically just is "the rest". Such a split would have many advantages  when it comes to testing and also overall handling of the dependencies.

If such a split was done it's trivial to make CLI routines which installs plugins, checks updates, etc. using the "normal" code. I have made such test shots, basically showing that's it's possible. However, it's a very long path before it's a PR which could be merged, and I'm unsure about walking this path.

leamas, 2024-02-19:
Back later... We now have opencpn-cmd which provides a basic CLI interface to list installed, list available and actually update plugins. It's still sketchy, and documentation is basically non-existent. `opencpn-cmd --help` gives the basics, though.

mgrouch, 2024-02-19:
> Back later... We now have opencpn-cmd which provides a basic CLI interface to list installed, list available and actually update plugins. It's still sketchy, and documentation is basically non-existent. `opencpn-cmd --help` gives the basics, though.

It doesn't seem to work even with plugins which load in OpenCPN UI fine:

````
user@lysmarine:~ $ opencpn-cmd list-plugins
Error: Failed to load shared library '/home/user/.local/lib/opencpn/libJavaScript_pi.so': /home/user/.local/lib/opencpn/libJavaScript_pi.so: undefined symbol: GetAISTargetArray
Warning: iCCP: known incorrect sRGB profile
Error: Failed to load shared library '/home/user/.local/lib/opencpn/libtactics_pi.so': /home/user/.local/lib/opencpn/libtactics_pi.so: undefined symbol: _ZN15PlugIn_WaypointC1EddRK8wxStringS2_S2_
Error: Failed to load shared library '/home/user/.local/lib/opencpn/libmeteovache_pi.so': /home/user/.local/lib/opencpn/libmeteovache_pi.so: undefined symbol: SetToolbarToolViz
Error: Directory '/plugins/' couldn't be created (error 13: Permission denied)
Error: Directory '/plugins/ocpn_draw_pi/' couldn't be created (error 2: No such file or directory)
Error: Directory '/plugins/ocpn_draw_pi/data/' couldn't be created (error 2: No such file or directory)
Error: Directory '/o_charts_pi/' couldn't be created (error 13: Permission denied)
terminate called after throwing an instance of 'wxSQLite3Exception'
Aborted
````

leamas, 2024-02-19:
This is what I get on a Bullseye machine with some plugins installed:
```
 opencpn-cmd list-plugins
Error: Directory '/o_charts_pi/' couldn't be created (error 13: Permission denied)
Error: /home/al/.local/lib/opencpn/libJavaScript_pi.so: undefined symbol: GetAISTargetArray
Autopilot Route          0.4.13.0
ais-vd                   1.8.0
ShipDriver               3.2.3+8146.e6145f0
FindIt                   1.3.2.0
o-charts                 2.0.3.1
Radar                    v5.3.4
Calculator               v4.0.0
GRIB                     4.1
Dashboard                1.2
WMM                      1.1
ChartDownloader          1.4
javascript               v2.0.4a
```
The problem with Javascript seems like a "real" error: It links to a symbol which is not available in my OpenCPN build i . e. a  plugin API/ABI mismatch.

If I install the Tactics plugins from the GUI, my opencpn misbehaves (toolbar is almost without icons). I can uninstall it from the command line using `opencpn-cmd uninstall-plugin Tactics`  which reveals a _/home/al/.local/lib/opencpn/libtactics_pi.so: undefined symbol: _ZN15PlugIn_WaypointC1EddRK8wxStringS2_S2__ message.

TODO:
  - Could you possibly try to create a backtrace from the crash?
  - I'll try to look into the `Error: Directory '/o_charts_pi/' couldn't be created ` message.

leamas, 2024-02-19:
And on bookworm `opencpn-cmd list-plugins` does not display anything. Will look into this as well

leamas, 2024-02-19:
Fixed the linkage error in current master at bc4beb204. Output now on bookworm:
```
$ opencpn-cmd  list-plugins
ShipDriver               v3.3.3
o-charts                 2.0.1.8
Tactics                  1.0.80.0
pypilot                  0.40.30.0
AIS Radar view           1.4.10.0
MeteoVache               1.1.4+774.08b438a
javascript               v2.1.6-e
FindIt                   1.3.50.0
ChartScale               0.3.11.0
Radar                    v5.5.0
ChartDownloader          1.4
WMM                      1.1
GRIB                     4.2
Dashboard                1.2
```

Possible errors are usually visible in the log file.  One bug to fix is that failed installations still returns 0 in some situations.

leamas, 2024-02-21:
@mgrouch :
That is, I think we have a CLI interface  which is reasonably simple to use in a script which provides the functionality requested here. Or?

mgrouch, 2024-02-21:
> @mgrouch : That is, I think we have a CLI interface which is reasonably simple to use in a script which provides the functionality requested here. Or?

Can it be made simpler to use for upgrading all installed plugins to latest?

Like with Debian apt to upgrade everything only two commands needed. apt update / apt upgrade.

Thanks

leamas, 2024-02-21:
It certainly can. But this is just one of many use cases, and I hesitate to add this at this point.  

Writing a bash script should actually be simple, and leave more control to the user. Are you OK with writing scripts, or would you be helped if I created a sketch?

mgrouch, 2024-02-21:
> It certainly can. But this is just one of many use cases, and I hesitate to add this at this point.
> 
> Writing a bash script should actually be simple, and leave more control to the user. Are you OK with writing scripts, or would you be helped if I created a sketch?

Does current code downloads in parallel into some staging directory before installing a plugin?

Interrupted download will not install incomplete files? Will it clean temp files?

thanks

leamas, 2024-02-21:
For better or worse it works exactly the same way as the GUI, the same functions are used. This includes not installing broken downloads (there is a checksum). I'm not aware of left over temp files.

There are no parallel downloads. However, the downloads are so small I don't really see the need. (~500k average, max ~2.2 M)

leamas, 2024-02-21:
> Does current code downloads in parallel into some staging directory b

Downloads are done into a cache, cached downloads are used when re-installing the same plugin + version. The GUI has methods for cleaning the cache, but opencpn-cmd has not (should be simple to add if need be).

leamas, 2024-02-26:
@mgrouch : ping?

mgrouch, 2024-02-26:
@leamas 
I’m ok with scripting around it. Thanks

leamas, 2024-02-26:
OK,  let's close this then.

closed 2024-02-26 by leamas