OpenCPN/OpenCPN #2193: Plugin ability to determine OpenCPN version number

Rasbats, 2021-04-06:
Probably missed it but is there a way for a plugin to determine the version of OpenCPN it is being run on? If not, could this be added to the API.

(Needed for work being done by @antipole2 (JavaScript_pi))

leamas, 2021-04-06:
No, I don't think there is such a method. It's a bit sad, without it a plugin can just declare "I need API level x.y work properly" instead of using a less restrictive API level and adapt to the actual core version. So I second your proposal.

OTOH, if this was introduced now it will only be available in next version. Practically, it's a year or so until a plugin could use this new function without excluding many, many users.

What you could do, extremely clumsy but still: parse the log file which normally has the entry like `13:44:02.083    INFO logger.cpp:169 ------- OpenCPN version 5.2.4 restarted at 2021-03-15 -------` . This is not reliable, log file could be truncated,  user  might change the log-level to *warn* or what not. It will also be slow, so doing it in a thread might be necessary.

But it should work most of the times, and with some safety fallbacks it might even be acceptable.

canne, 2021-04-11:
An example: https://github.com/canne/dashboard_tactics_pi/blob/a3ec5884b552410e15065e6788d44951c3d75a95/src/dashboard_pi.cpp#L2808

leamas, 2021-04-11:
I stand corrected. Seems that code in  src/chcanv.coo around line  9800 sends a configuration message with all sorts of info about the host. Since messages are broadcasted, any plugin should be able to pick up this.

I don't fully understand  all the conditionals in core OpenCPN, but presumes that this message will always be sent at least once (?)

antipole2, 2021-04-11:
@canne thanks for pointing this out.  Extracting from "OpenCPN Config" requires ending a message to request it and then parsing the result and extracting the information.  Possible, I agree.  But  other configuration info is available through the API, such as

> #define API_VERSION_MAJOR           1
> #define API_VERSION_MINOR           16

and I think it would be helpful to have the OpenCPN version accessible too in such a simple manner.  I could use that to check if the host version had some issue fixed without having to write lots of code and arrange call-backs etc.

leamas, 2021-04-11:
It just can't be. There is no static inkage between the plugin and the core, so the core version must be established in runtime. In particular, a header file used when compiling the plugin can *not* be the source of truth.

OTOH, someone could of course wrtite a wrapper function or two which handles the messaging stuff and makes retrieving the version simpler

@rasbats original request for a function where plugin can retrieve the version from the core is certainly possible and would indeed be simpler. Problem is just that it's at least a year away if implemented now.

antipole2, 2021-04-11:
Ah - I see that now - it has to be fetched at run-time.  A 'getConfig' call could return the version and maybe other configuration stuff.  Yes I know it can be got via a message but a simple call to retrieve would be so much easier and simpler to use :-)
In the JavaScript plugin, scripts may want to check which OpenCPN version they are running under before assuming some recent feature/fix is available.   The script does not exist at plugin activation time and has missed the config message sent to the plugin at activation time.  So it would have to send another request and then set up a call-back to process it once received.  Possible  but a hassle.

leamas, 2021-04-11:
> . A 'getConfig' call could return the version and maybe other configuration stuff. 

Not my call, but I guess a PR adding  such a function to the API might be appreciated. But, again: even if it was included in next release, a plugin can not link to it and use it until "most" users have updated their OpenCPN installations. Realistically, this means waiting at least a year, probably more.

canne, 2021-04-11:
Meanwhile, @antipole2 , to help you to move forward without touching the OpenCPN core, and while we are waiting for the completely revamped API: since you already have subscribed to WANTS_CONFIG (line 136) why not capture and register the provided configuration JSON structure which you will get at startup to _SetPluginMessage()_ thanks to the above subscription?

The version of OpenCPN is not likely to change during runtime (one never knows...) so why you need to 'ask' for that particular message dynamically since you already asked it to be sent once while you registered your plugin with OpenCPN?

You would provide the single-shot stored JSON structure to your JS client objects once they show up their nose in this world through a new "OpenCPN APIs" of yours (which you need to develop, page 8 of your document).

The example code link above is doing precisely that, a static one shot inquiry for the configuration information at startup: in addition, you just need to store the JSON structure in the C++ object's attributes and provide a method for your client JS objects to retrieve it whenever they need it. JSON parsing in JS is natural born, so no such parsing in C++ is needed, IMO: you can ignore the parser in my code. But if you like, provide a JS method for your users to extract the version information from the JSON if they are interested only in that and/or do not feel at ease with JSON for some reason.

antipole2, 2021-04-11:
@canne...  thanks for your in-depth suggestions.  Yes, I was thinking this morning about capturing the info when it was delivered at plugin registration.  As you clearly realise, parsing JSON in C++ is hard work compared to in JavaScript.

I am now capturing the message in the plugin when seen and returning it as JSON on demand.
So the script
`print(OCPNgetOCPNconfig(), "\n");`
now prints, in my case, 
```
{
   "OpenCPN S52PLIB ShowImportantTextOnly" : false,
   "OpenCPN S52PLIB ShowNationalText" : true,
   "OpenCPN Version Full" : true,
   "OpenCPN S52PLIB DeclutterText" : true,
   "OpenCPN S52PLIB ShowAnchorConditions" : false,
   "OpenCPN S52PLIB BoundaryStyle" : 78,
   "OpenCPN S52PLIB UseSCAMIN" : true,
   "OpenCPN S52PLIB ShowQualityOfData" : false,
   "OpenCPN S52PLIB GlobalReconfig" : true,
   "OpenCPN Version Patch" : 4,
   "OpenCPN S52PLIB ColorShades" : 0,
   "OpenCPN S52PLIB MetaDisplay" : true,
   "OpenCPN Version Date" : true,
   "OpenCPN S52PLIB SymbolStyle" : 82,
   "OpenCPN Version Minor" : 2,
   "OpenCPN Version Major" : 5
}
```
A script could still send a request for the configuration to get an updated one if something might have changed. 

Thanks for the exchange.

leamas, 2021-04-13:
@canne: On a sidenote: We are trying to rewrite the Developer Guide. From the discussion here, I  understand that the messaging API is a central part of the interface between plugins and the core. Since you seem to have figured this out: could you possibly write a draft for a page with a description of the messaging facilities?

canne, 2021-04-14:
@leamas  - 👍 for the effort put up a new Developer Guide. However, your estimation about my knowledge about the message API is exaggerated since I have written but one plug-in and used a debugger to find about those non-documented messages I needed in it. With regret, I have to decline your invitation since that would require to scan through numerous (too many) core source code files and I do not have time for that.

IMO the call-back function API is the primary interface of plug-ins which are directly attached into the OpenCPN main thread and therefore its name space. It shall be developed further (as suggested above and in many FlySpray posts) or redesigned from scratch.

The usage of messaging "API" (is there one?) with largely arbitrary message structures does make sense only if the plug-ins are allowed to be free of the OpenCPN execution thread (and even the computer on which it is running) and communication with OpenCPN would happen through some socket or another serialized interconnection.

rgleason, 2021-04-14:
Ocpn_draw uses odAPI for communicating textpoint, boundary and quard zone GUID with other plugins such as watchdog and weather_routing but that is a separate messaging system from OpenCPN and that has been moved to Ocpn_Draw plugin pages.

rgleason, 2021-04-14:
Off Topic to answer @Leamas your writeup partially answered https://github.com/OpenCPN/OpenCPN/issues/1070

leamas, 2021-04-14:
@canne: Can you drop me a PM to leamas dot alec AT gmail?

bdbcat, 2021-04-29:
For all the reasons elucidated above, I do not plan to add a RequestVersionString() call to the plugin API.
I endorse the notion of capturing the JSON config object string at plugin Init(), storing statically, and parsing as necessary.  The required wxJSON source files are readily available from any of a number of OCPN plugins on github, and it really is not so hard to write the 20 or so lines of code capturing the JSON messages.  Just copy/paste.

leamas, 2021-04-29:
@rasbats: Your original request seems answered, there is a way to retrieve the version. Also, Dave has stated that there is no need no expand the API with a new method for this, since it's not needed.

Time to close?

Rasbats, 2021-04-29:
We have the ability. Closing.

closed 2021-04-29 by Rasbats