OpenCPN/OpenCPN #2197: Raspberry Pi can hang in Options->Plugins

transmitterdan, 2021-04-12:
On Raspberry Pi (Raspbian Buster) most recent master I get a hung UI condition like this:

[image]

The Options toolbar icons are missing and the buttons CANCEL, OK, APPLY do nothing.  The close window (X) does nothing either.  The only way out is to kill O.

This is not new, I can replicate it quit a few commits back.  I'll have to do a bisect to see when this happened I guess.

bdbcat, 2021-04-12:
Dan...
Please do the bisect.  It is probably a specific plugin that is hanging.
Thanks
Dave

transmitterdan, 2021-04-12:
Dave,

The only enabled plugins are the 4 standard.  It hangs before loading any other plugins.

This is some kind of wxWidgets/GTK3 issue I suspect.

Dan.

closed 2021-04-13 by transmitterdan

bdbcat, 2021-04-13:
And what did you find?

transmitterdan, 2021-04-13:
It has something to do with a non-managed plugin.  I cannot confirm it is a bug within O but it could be.  Will take a while to find out for sure. It does not appear to be happening with the 4 standard ones or managed plugins.

I will keep trying to find if there is a bug in O or not.  Don't think this issue should be a major concern.  I'll reopen if I can confirm a real bug exists.