OpenCPN/OpenCPN #2198: RPI4 deactivate network connection at Options ->OK

Hakansv, 2021-04-13:
Rpi4 Buster as of below spec.
OCPN master code.
A new phenomenon experienced on my Pi 4. A "Options -> OK" closes the network socket. Independent of used Options tab the OK button closes any network connection. 
I can reestablish a connection by a switch to another and "Apply" but it's closed again on "OK"
If I restart O the connection is established again until I visit "Options and click "OK".
Could anyone make a O code connection or suggest a debug method I could try?
The Pi is recently reestablished by a SD card backup after a card crash. (Probably due to power off while SD writing was not ready)

More data:
pi@RPi4:~ $ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 180.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 

Log file examples:

```
Options -> Wiew -> OK
09:59:04.190 MESSAGE NetworkDataStream.cpp:501 Closing NMEA NetworkDataStream TCP:signalk.stupan.se:10112
09:59:04.190 MESSAGE datastream.cpp:274 Closing NMEA Datastream TCP:signalk.stupan.se:10112

In options I change network connection.
> click "Apply"
10:00:45.788 MESSAGE datastream.cpp:135 makeDataStream Signal K:192.168.10.229:3000
10:00:45.788 MESSAGE datastream.cpp:213 ConnectionParams CTOR
10:00:45.788 MESSAGE SignalKDataStream.cpp:617 Opening Signal K WebSocket client: Signal K:192.168.10.229:3000
..............
10:00:46.295 MESSAGE SignalKEventHandler.cpp:65 Connected to Signal K server version: 1.39.0

> click "OK"
10:00:55.788 MESSAGE SignalKDataStream.cpp:516 Closing Signal K WebSocket DataStream 
10:00:55.988 MESSAGE datastream.cpp:274 Closing NMEA Datastream Signal K:192.168.10.229:3000

Restart O, connection established and works as it should until:.........
Open Options.  > click "OK"
10:08:12.074 MESSAGE SignalKDataStream.cpp:516 Closing Signal K WebSocket DataStream 
10:08:12.275 MESSAGE datastream.cpp:274 Closing NMEA Datastream Signal K:192.168.10.229:3000
```

pi@RPi4:~ $ ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:2c:b8:74  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 33145  bytes 10677396 (10.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 33145  bytes 10677396 (10.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.229  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fe80::dea6:32ff:fe2c:b875  prefixlen 64  scopeid 0x20<link>
        inet6 fd47:4bf5:454b:0:dea6:32ff:fe2c:b875  prefixlen 64  scopeid 0x0<global>
        inet6 2001:2042:7e0b:3300:dea6:32ff:fe2c:b875  prefixlen 64  scopeid 0x0<global>
        ether dc:a6:32:2c:b8:75  txqueuelen 1000  (Ethernet)
        RX packets 17288  bytes 3835178 (3.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1996  bytes 254660 (248.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan9: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.10.1  netmask 255.255.255.0  broadcast 10.10.10.255
        inet6 fe80::dea6:32ff:fe2c:b875  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:2c:b8:75  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 712  bytes 82252 (80.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Hakansv, 2021-04-13:
A clarification.
The Button "OK" is actually doing the "Apply" function. And, yes, the "Apply" button yields the same effect as the "OK button. In my "Apply" > "OK" example above a second "Apply" consequently yields the same network failure.

bdbcat, 2021-04-14:
@leamas...
Please check this logic.  AFICT, readlink() does not terminate the returned string buffer.  And all error returns (-1) leave the returned buffer undefined.
Anyway, what is this readlink() call intended to do?  What link is it supposed to find? I have not seen it succeed yet.....

closed 2021-04-14 by bdbcat

leamas, 2021-04-14:
This readlink() comes into play when users do what's described in LINUX_DEVICES.md. In short:
  - A device name like /dev/ttyUSB0 is not stable. On my laptop, it typically changes to /dev/ttyUSB1 after a suspend/resume cycle
  - The remedy is to make udev create a stable device name like /dev/nmea. Such names are symlinks maintained by udev.

But for the last bullet to work, OpenCPN must handle symlinks, and understand the actual device they refer to.

Your patch looks fine, and the unterminated target is a terrible error. "blushes"  Filed #2202 with some refinements.

Hakansv, 2021-04-14:
- The patch fixed it on my RPi4. Thanks.
- "A device name like /dev/ttyUSB0 is not stable. " That's why Openplotter - Serial have the opportunity to use alias for serial ports.
[image]

leamas, 2021-04-14:
Most likely, yes, And if those alias names are listed by `ls -l /dev/` they should now work just fine also in OpenCPN

commit a22a89fe (2021-04-14): Correct logic surrounding readlink() in multiplexer. Closes #2198