OpenCPN/OpenCPN #2205: SignalK protocol connection gets lost and doesn't recover

mgrouch, 2021-04-17:
The values I get via signalK protocol still displayed in OpenCPN but they are stale.
Barometric pressure doesn't move.
Same barometric pressure changes in SignalK KIP dashboard.

If I restart OpenCPN for some time I get  correct values via SignalK.

It seems signalK protocol implementation is not fully ready yet

Hakansv, 2021-04-17:
If I for example switch my internet connection off and after a while on again my SK connection is reestablished as expected. (My SK server is at home)
In current master code the watchdog time out for barometric pressure is 30 seconds. The SK parsed value is "environment.outside.pressure" and the instrument values are wiped after 30 seconds of no data.
BTW: I'm using master code.What's your O-version?

mgrouch, 2021-04-17:
5.2.4.  Latest release available. 
I’ve just observed it again. In this case
It went to no value (null).

Hakansv, 2021-04-17:
OK, there have been some changes to Dashboard in this matter. If also for the SK connection methods I don't remember. But there's a beta test and later release of version 5.5.0 coming up soon. 
My suggestion is to close this issue and reopen if still valid in next beta release.
Thanks
Håkan

mgrouch, 2021-04-17:
30 seconds is too fast to expire air temperature and barometer.
SignalK measures only once a minute. That's why they were disappearing.

Hakansv, 2021-04-17:
Is that so. I read somewhere the non navigational messages, like this and e.g. motor data and levels, was updated once every 20 seconds that's why that watchdog was set to 30 seconds. No problem to change but where did you find this info?

mgrouch, 2021-04-17:
BMP sensor plugin default setting in
SignalK

tkurki, 2021-04-17:
The "Signal K way" for this is to use `timeout` in metadata and if not specified by the SK server you're on your own.

https://signalk.org/specification/1.5.0/doc/data_model_metadata.html

We really should add some sensible defaults to the SK server ( or the specification?) and add a way to edit metadata, including timeout values, to the SK server UI.

Hakansv, 2021-04-17:
The conclusion would be there is no general setting in SK and for the moment now way to obey an expected timeout for any SK value. The future may have something in sight. This based on a recent discussion with SK devs. 

Now we have to decide when we judge a value for old and unsure and wipe it out from the Dashboard instrument. 
My idea was 30 seconds for non navigational data like temperatures, air pressure and tank levels. So that's the present code.
Any other bids? 60 seconds is in the air here.
Håkan

rgleason, 2021-04-17:
Hakan, is there a way  to have that setting under preferences or is that too messy?

Hakansv, 2021-04-17:
Yes, I think so. We have enough more or less understandable settings. My opinion. 
But for these not critical data 60 seconds would do?

mgrouch, 2021-04-17:
60 seconds might not work if 60 is in SignalK. Boats travel on surface. Pressure can’t change that fast like in airplane. 5 minutes will be just fine.

Hakansv, 2021-04-17:
Well, we are talking about how often an instrument is reporting data, Not if the value is the same as before. Even if the value is the same we need a value to detect the instrument is alive.

closed 2021-04-17 by mgrouch

Hakansv, 2021-04-19:
@mgrouch 
Just to be sure here. The estimation of the update period for any not critical instrument to 30 seconds is based on examples of some temperature and tank level instruments. If you've other examples please advice. 
The watchdog time would not be longer than necessary because a Dashboard instrument value is expected to disappear for example when a connection is lost. (And the Dashboard plugin does not monitor any physical data connection. The plugins just receive data from OpenCPN)

transmitterdan, 2021-04-19:
@Hakansv, I think leaving the displayed values of air temp, barro press, etc. on display for longer time is ok. These do not often change rapidly. It might be useful to have an "instrument" that provided some info on the SignalK server itself. Perhaps a simple 'it's alive' indicator similar to the NMEA rotating slash that O uses. Then a user could know that the data has stopped flowing by a means other than the data disappearing.

mgrouch, 2021-04-19:
Might be crossed font or some shade of gray on expired readings in display

Hakansv, 2021-04-19:
Opinions:
Wherever a supervision is constructed it has to be time related to last received value. In Dashboard we call it Watchdog and for now set to 30 seconds. Without any "Watchdog" the DB instrument value will be persistent until next OCPN restart.

Instead of clearing out the data when the Watchdog expires we could of course change the font in some matter but what to do with graphical instruments like the Barometer "compass"and not less the Baro history graph. What else than a straight line? A (small) font change can still fool anyone not enough observant.  -  "Still plenty of Diesel." -  A wiped out value is all time very clear.

The Watchdog timer for not critical instruments could though be easily changed. Not less now before a Beta release. Suggestions?

transmitterdan, 2021-04-19:
For sure, we need a watchdog to prevent stale data confusing the user.

I was thinking that we could allow a longer watchdog time for slow changing data if we also had a new instrument that just indicated Signal-K data is flowing or not.  A user could use that instrument to observe if new data is coming in or not.  That could have a short watchdog time. I just think 30 seconds is much too short for barro pressure.  Most people only look at barro once every 15 or 30 minutes.

I agree that trying to show via font or some other visual cue that the data is "stale" will not work well.  Many sailors have too much going on to notice a change in shade, font, etc.

But having a flashing icon or perhaps even an audible alarm when Signal-K data has stopped might make some people comfortable with longer watchdog times for individual instruments like air temp.

Hakansv, 2021-04-25:
If most data are coming from SK there's not a problem to detect a socket failure. Boat position, compass bar and other nav-instrument will lost it's data in 6 seconds.

I agree one is not studying the barometer pressure very often but once you do you will know it's a reliable value? 
I don't think this is about how often anyone use the data but how often a common sensor will transmit it's data. My 30 seconds was based on examples from tank level and air temperature sensors using 20 seconds. If there are other examples, like 60 seconds, we could use that instead?

Hakansv, 2021-04-29:
No suggestions.
@transmitterdan 
What if we set the timer to two minutes. Will cover most sensors and wiped out or change to another source in due time I suppose