OpenCPN/OpenCPN #2206: Announce: Travis CI now adding credits to open-source account.

leamas, 2021-04-17:
Got a message from travis ci support:   excerpt:

> After reviewing your open source qualifications, I am pleased to notify you that 25k credits were added to your account allowing you to resume running OSS builds. When your credits begin running low again, please reach back out to the Support team.

Perhaps this means we can continue to use the Travis builders, perhaps not.  The reply was to a message I sent February, 5.  FWIW, the reply also included some "sincere apologies" so the overall impression  is that they actually want their free open-source users back .

leamas, 2021-04-17:
This is an example on a message that really should be distributed in some other forum. Sorry...

leamas, 2021-04-22:
This is not actionable, agreed. OTOH, this is important information for devs, and right now this is the only place to make it known.

> Travis is not being a reliable partner for Open Source

You certainly cannot partner with Open Source ;)

That said: Travis seems to be back in open-source business, at least they try to if I get this right.

rgleason, 2021-04-22:
Ok, so I ask for more credits, maybe they give me some, then what after they are used up?  Everything is broken again. Forget it if there is a more reliable way.  Managing one or two repositories is different than multiples.

leamas, 2021-04-22:
Well, that is you. Perhaps there are others...

leamas, 2021-04-22:
I might add that in the message they asked to come back to their support when the credits are running low.  They are trying,  from what I see right now, to make this work.

rgleason, 2021-04-22:
So I make a push to github to build and deploy, travis runs out of credits, I then ask for more, wait, then have to make another push to the get the full deployment.  And this will happen on 20 plugins?  No thank you.

leamas, 2021-04-22:
We'll see. Somebody should make some tests.

rgleason, 2021-04-22:
In this case a recent Trusty failed, but there are other consistent failures... perhaps travis is running and signed up ok, but it is worth checking.

leamas, 2021-04-23:
I would be happy to move this to Discussions, but seems that I cannot do this myself (?). If anyone else can, please do.

rgleason, 2021-04-23:
To "Show and Tell" Topic?

closed 2021-04-23 by nohal