OpenCPN/OpenCPN #2207: OpenCPN picks up wrong magnetic variation (0) even if it knows correct value

mgrouch, 2021-04-17:
Somehow GPS reports magnetic variation value of zero.
OpenCPN knows magnetic variation from its own magnetic variation plugin (correct value)
but somehow in all calculations for 'true' it uses 0 value from GPS.

Sometime it's even non zero but some many huge digits number.

As result all 'true' values are calculated wrong.

Thanks

Hakansv, 2021-04-17:
If any source is reporting exactly "0" as magnetic variation OCPN assume it's not a real value and lets WMM take over instead. If the value is not exactly zero that value is used. If the data reported to OCPN is not correct in any other matter it has to be corrected by the transmitting device. So, bad data in - bad data out.

See for example here: https://github.com/OpenCPN/OpenCPN/blob/1b353268160ef9e73f7b7bbf3c05637ac5855ac6/src/chart1.cpp#L9078

mgrouch, 2021-04-17:
Might it be instead of 

0.0 != m_NMEA0183.Rmc.MagneticVariation

Check abs difference with some low threadhold. 

Like 0.0000000001 or something

Hakansv, 2021-04-17:
If there's is a value 0.0 we assume it's an old GPS with outdated WMM table. In many devices you've the opportunity to set the variation to a specific value or just 0.0
So, any other value can be real.

mgrouch, 2021-04-17:
0.0 serialized over network or parsed might not be == your local 0.0 binary representation anymore.

Hakansv, 2021-04-17:
Maybe? I've used it like that in my dated GPS for many years now and have never seen it fail.

transmitterdan, 2021-04-18:
@mrgrouch, 0.0 is the same in all IEEE floating point compliant systems.  I can think of no system that represents 0.0 as anything other than identically 0.  If you know of such a system please tell what it is.  Don't make us go in an Easter Egg Hunt to find it.

rgleason, 2021-04-19:
Mike, time to close this please. Thanks.

closed 2021-04-19 by mgrouch