OpenCPN/OpenCPN #2208: Add change indicator (arrow up/down) for parameters displayed on dashboard

mgrouch, 2021-04-17:
Add change indicator (arrow up/down) for parameters displayed on dashboard.
Like on a stock ticker (green up, red down) or might be without color. just arrow.
Thanks

Hakansv, 2021-04-17:
@mgrouch 
Is this a bug report or a feature request? In the latter case please use the [Flyspray](https://www.cruisersforum.com/forums/f134/tracker-flyspray-when-and-how-to-use-it-110519.html#post1327968url)
Thanks

rgleason, 2021-04-17:
Mike   See tracker here https://opencpn.org/
You will need to register and then  Add a new item  and then pick Feature Request.  
After you've done that VOTE for it and get others to do so too.
We have a Vetted list once a featture request gets enough votes.

Also pllease please close this Issue.  We cannot do that only moderators.  We are trying to get the number down to below 100!

Thanks
Rick

mgrouch, 2021-04-17:
feature req

rgleason, 2021-04-27:
Mike, It turns out I  was wrong about Feature Requests, sorry.  They should just start with RFE  (Request for Enhancement)

Hakansv, 2021-05-12:
@mgrouch 
If you're not talking about the already existing function maybe you could describe a bit more what you're suggesting?

[image]

transmitterdan, 2021-05-12:
I think what is meant is for instruments such as DEPTH, there should be an arrow showing the direction depth is changing over time.  If DEPTH is going deeper then a DOWN arrow should be shown.  If DEPTH is unchanging then no arrow is shown.

Many commercial instruments have this feature.

Hakansv, 2021-05-12:
Ahaa. Misunderstanding from my side then. Since I always use the depth graph where changes are obvious that thought didn't hit me. And yes, my sounder instrument has these arrows. Mostly rather nervous up/down since the bottom profile is changing all time here. It may need some well designed hysteresis?
Thanks TDan.