OpenCPN/OpenCPN #2209: In case of multiple GPS connections failover to the one which has a fix doesn't work

mgrouch, 2021-04-17:
In case of multiple GPS connections failover to the one which has a fix doesn't work.

I set them different priorities. But it seems it only switches to another one when TCP link goes down.
If TCP link stays up but fix is lost it doesn't switch to the one which has a GPS fix.

Thanks,
--MG

Hakansv, 2021-04-17:
That's a correct behavior so it works at it should. The aim for multiplexer is just to prioritizing between equal NMEA messages, like RMC. The multiplexer can't judge the content of a received message. But If a device stop sending e.g. RMC or you filter it out the prioritization will kick in. 
If your device has a failure you may switch it off or change the prioritization to favor a better one.

mgrouch, 2021-04-17:
I'm not talking about multiplexer. OpenCPN charts loose fix even when it is available.

Hakansv, 2021-04-17:
Yes, you do as I understand it. You are talking about "connections failover" and its prioritization. That's handled by OCPN's multiplexer

mgrouch, 2021-04-17:
I do think reasonable users would expect not to lose GPS fix on chartplotter if GPS fix is available

rgleason, 2021-04-17:
Dear Mike, 
Your knowledge, enthusiasm and involvement is great. 

I think this report really belongs in the Cruiser's Forum, until we know it is a bug for sure, and not some hardware problem or other issue. The forum has a broader range of individuals many of whom have good technical skills. I think it would be best if you would make a new Topic  in Cruiser's Forum  link is also on the website under Resources  https://opencpn.org/OpenCPN/info/resources.html

The topic could be something like  "GPS Failover Issue"  If you would post your OS,  Opencpn version and other pertinent details, such as GPS, driver etc,  I am certain that Hakan and others will respond in an effort to figure out what is happening.  Once it is confirmed as a bug, you would then post it here with greater certainty. 

Thanks.

Hakansv, 2021-04-17:
Mike..
re: I do think reasonable users would expect not to lose GPS fix on chartplotter if GPS fix is available
That's another thing. If we receive what's not a fix in e.g. a RMC message O tries to find it in another message like GGA or GLL  But we can't judge a valid position to be "bad" if received. 
If a connection to a device fails and there's another supporting the same NMEA messages the multiplexer will switch.

rgleason, 2021-04-17:
So who is going to close this now? Not me. I can't, nor can Hakan.  Its the OP or moderator who closes it.

rgleason, 2021-04-25:
Mike wrote:
I set them different priorities. But it seems it only switches to another one when TCP link goes down.
If TCP link stays up but fix is lost it doesn't switch to the one which has a GPS fix.

Hakan wrote:
That's a correct behavior so it works at it should. The aim for multiplexer is just to prioritizing between equal NMEA messages, like RMC. The multiplexer can't judge the content of a received message. But If a device stop sending e.g. RMC or you filter it out the prioritization will kick in.
- If your device has a failure you may switch it off or change the prioritization to favor a better one.

Mike wrote:
I do think reasonable users would expect not to lose GPS fix on chartplotter if GPS fix is available

So prioritization is only checking that the TCP link is working?  Why not also check that there is a valid GPS sentence too?

rgleason, 2021-04-28:
Is this a BUG or should it be move to Discussion?

bdbcat, 2021-05-02:
Rick...
I'll mark it as enhancement.
The issue is understood, but it is not clear what the proper behavior should be. In any case this change would require significant re-factoring of the NMEA prioritization logic.