OpenCPN/OpenCPN #2213: isCompatible no longer works with linux mint

jongough, 2021-04-22:
A change as introduced to handle variants of ubuntu and the possibility of running different built versions on other platforms. This change allowed plugins that were built for the same environment as OCPN to be installed as if OCPN could run then the plugin could run irrespective of the information obtained from the runtime environment.

Change 17e2d0a2 by Alec Leamas, 03/15/2021 08:30 AM (The initial check is just optimization. Given that there are
problems to get the overall check right, drop this for now.) appears to have purposely removed this capability. I can now no longer run debugging on a linux mint machine even though I built OCPN on the machine and the plugin, the plugin is deemed to be incompatible.

The capability needs to be put back into the code in line with the xml schema used for plugins.

leamas, 2021-04-22:
Reading the header, I assume that your core problem is that you cannot debug on your Mint machine (?).

However, the code removed in  17e2d0a cannot fix this. It does *not* mark a plugin as compatible in any case,  although it does mark some as incompatible in an attempt to optimize.

As a starter: can we agree on this?

jongough, 2021-04-22:
My core problem is that compatibility is based on what the runtime can determine rather than what is determined at build time. If a particular build of OCPN is running then it must be compatible with the the operating system. If a plugin is built on the same platform as OCPN it is then compatible. Checking at runtime for build time compatibility will ensure that there is no issue with derived operating systems.

I fixed this issue last year and gave a way forward to help compatibility testing. I currently do the builds on linux mint, OCPN and the plugins so they should all be compatible, but they are now rejected by plugin manager. There is no build time compatibility checking as this was removed. 

It worked at this commit: 2938acd6, it was changed in commit: 428b5ab9, changed again in commit: 17e2d0a2, and appears to have been completely removed in commit: 7230f0ef. The changes are all in PluginHandler.cpp.

bdbcat, 2021-04-22:
Jon...
Is it not possible for you to configure your builds (using the tuple definition) to allow building and running on Mint?

rgleason, 2021-04-22:
Why was it necessary to change coded compatibility checks done at the beginning, to be done at the end?

bdbcat, 2021-04-22:
I don't understand your question.
Please answer mine.
Thanks
Dave

jongough, 2021-04-23:
I can of course change my build environment, but that is missing the point. This is hypothetical but illisrtates the problem.
If you build OCPN for Ubuntu 16.04 but run it on Ubuntu 20.04 should Plugin Manager not accept the plugin because it is built for Ubuntu 16.04 as that is not compatible. Or should plugin manager accept that if OCPN is running and it knows what it was built for, Ubuntu 16.04, then a plugin also built for Ubuntu 16.04 should be accepted. 

In my case, with the changes introduced by the mentioned commits, Plugin Manager will say its not compatible, but if I copy the lib file to the requisite place it runs quite happily. It is only the installation process that does not work. 

I made an update to PluginHandler and the XSD/XML to know about 'build environment' and also check for this, and this was accepted in Dec 2020. This check has now been removed, hence the issue. I can quite easily recreate the patch needed, but I have no confidence that it will not be summarily removed in a future commit, hence why I am not doing the work.

bdbcat, 2021-04-23:
"If you build OCPN for Ubuntu 16.04 but run it on Ubuntu 20.04 should Plugin Manager not accept the plugin because it is built for Ubuntu 16.04 as that is not compatible."
The answer to that must be "the plugin should be rejected."  There is no telling if a plugin built for 16.04 will run in a 20.04 environment.  In fact, likely not, since the GTK2-GTK3 migration issue arises on this particular pair.  Almost certainly, all possible permutations of this logic will not have been tested in any meaningful way.  When wxWidgets 3.1.5 (or 3.2) becomes available in the linux repos, this adds yet another variable.

We are electing to enforce fairly tight plugin compatibility tests because we have thousands of users in the field, with a huge variety of configurations and experience level.  We want things to work when they say they will, with high reliability, right "out of the box" with no tweaking required.

You are a special case.  An experienced developer choosing to build and run OCPN on a non-standard platform.  You might be expected to have to customize your build in some ways.  An inexperienced user running Mint would likely be advised to use flatpak, to avoid such problems.

rgleason, 2021-04-23:
Perhaps this should be moved to "Discussion"  Show & Tell topic?

leamas, 2021-04-23:
Or closed as a duplicate of #1943.

jongough, 2021-04-26:
Just trying to test installation of plugin it is possible to select the download compatibility from the Plugin Catalog Settings. This allows the user to download 'compatible' plugins which it does but during the installation process it says the plugin is not compatible. I am using a derivative of Ubuntu 20.04, the plugin lib file works perfectly OK, but the Plugin Manager does not believe the plugin is compatible.

leamas, 2021-04-26:
>  the plugin lib file works perfectly OK, but the Plugin Manager does not believe the plugin is compatible.

As Dave [pointed out](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-825341743) this is as things are intended to be. The issue about Mint plugins has also been ["beaten to death"](https://github.com/OpenCPN/OpenCPN/discussions/2224#discussion-3334592), and this issue IMHO should thus be closed as NOT-A_BUG since status quo is the resolution.

jongough, 2021-04-26:
Can't be bothered. Working code changed to non-working, no real testing done, causing disengagement. Got better things to do that fight for a better product.

rgleason, 2021-04-26:
I am still curious, Is there any answer to:

Why was it necessary to change coded compatibility checks done at the beginning, to be done at the end? 
Or why were the changes necessary in that manor ... put at the end?

rgleason, 2021-04-26:
In other words, is it possible to meet all the objectives with a little added work?

leamas, 2021-04-26:
See discussion in #2185.

rgleason, 2021-04-26:
I enjoyed bdbcat's humor here:

"I have lots of boat electrical hardware that "works occasionally". If I thought software was subject to the same rules, I'd give up...."

I note that on March 15 you wrote this:

IMHO, this whole function needs a redesign due to:

- There should not be a need to consult the runtime version info to figure out the host. We know already at configure time if the code is aimed for for example Android-eabi-v7a, flatpak-aarch64, etc.
- The only exception is the list of compatible OS:es, this is only available in runtime. _**OTOH, these checks primary used for Mint doesn't really work today.**_  [my highlight]
- Not having proper target types for Android has meant introducing some kind of "subtypes" for armhf/arm64 adding more complexity
- Overall, it's just too much code in a single method to maintain (as can be seen here).

In other words, it's too complicated so lets simplify and the end result was not checking at the beginning, thus leaving out Mint?

leamas, 2021-04-26:
The initial check could not save Mint. That check was just marking some plugins as not acceptable, it did not mark *any* plugin as acceptable, neither Mint nor anything else.

What broke Mint was that I resorted to having the ABI determined in configure time rather than runtime. As you can see from [Dave's comment](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-825341743) there are good reasons for this.

rgleason, 2021-04-26:
The code removed by the "**[isCompatible: Drop initial architecture check](https://github.com/OpenCPN/OpenCPN/commit/17e2d0a2872764ccc63bdd9445cfe4ae39c21f63)**"   your March 15 Commit, had comments at the beginning

```
 // For linux variants....
    // If the plugin architecture is defined, we can eliminate incompatible
    // plugins immediately
```

However you state:
**_The initial check is just optimization._**  (my emphasis) Given that there are
problems to get the overall check right, **_drop this for now_**  (what was the intent here?)

You stated "the initial check is just optimization".   - Was it?

leamas, 2021-04-26:
Yes

leamas, 2021-04-26:
> what was the intent here? Work it back in

When I wrote it, I wasn't sure. But as it turned out, there was no need.

rgleason, 2021-04-26:
Dave wrote:

`We are electing to enforce fairly tight plugin compatibility tests because we have thousands of users in the field, with a huge variety of configurations and experience level. We want things to work when they say they will, with high reliability, right "out of the box" with no tweaking required.`

Which makes perfect sense to me, being 1/2 user.

Is there some way to have a CMAKE override trigger for "experienced programmers"  or is that a bad idea?

leamas, 2021-04-26:
What we have ir the various ways to override the default compat setting: The OPENCPN_COMPAT_TARGET env variable, the config file settings and  the new catalog GUI for setting the target. These are designed to aid the developer. One of the core objectives of the PR we discuss was to restore this functionality which was broken, partly due to the initial test you broght up [above](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-826698455)

rgleason, 2021-04-26:
Alec, can you give an example for how that might be used with Mint?  That would be very helpful. Thank you.

leamas, 2021-04-26:
IT's depends on mint version. in general:

  - env variable OPENCPN_COMPAT_TARGET should be  *abi:version*, for example *ubuntu-x86_64:20.04* which should make OpenCPN attempt to load plugins compiled for ubuntu version 20.04. Note that this might not work  if the linkage fails, the logs should reveal the truth. The trick is to select the right version.
  - The configuration variables *CompatOS* and *CompatOsVersion* in the *[Settings]* section works likewise if modified manually before starting (CompatOS is the same as ABI)
  - The catalog Settings GUI has a dropdown for selecting the compatibility target. It has the same limitations, it will try to load plugins, but it might fail. As a bonus, it displays the number of available plugins for each platform.

rgleason, 2021-04-26:
Thank you, I am not sure I understand what ABI means  = compatOS  as compared to compatOS.version in PluginHandler.cpp
but perhaps it does not matter (it's pretty complicated) and at the heart of managing plugins and well above my head.

I have a side question, should this be documented in the Dev Manual somewhere?

leamas, 2021-04-26:
the list of allowed ABI specs can be found in ocpn-plugin.xsd (plugins project). Here, you have `<xs:element name="target">` which includes a list of all usable ABI definitions.

Yes it should be documented. Why not file a bug against ocpn-dev-manual so we don't forget it? Perhaps with a link to this issue.

leamas, 2021-04-26:
After tjhis diversion: IMO this bug should be closed as of earlier  [post](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-826569607)

jongough, 2021-04-26:
Nope. Build OCPN on Linux Mint, build plugin on Linux Mint, run OCPN, install via plugin manager, fails due to incompatible plugin. Therefore not working.

bdbcat, 2021-04-27:
Jon...
In order to replicate, we need more info:
1. "build plugin on Linux Mint,"  How did you do this, exactly, outside of the established CI plugin build process?

jongough, 2021-04-27:
I have KDevelop on my desktop, currently Linux Mint 20.1, which I normally use for build and debug. For this test I created a new build directory for OCPN and built OCPN commit 8dcf8ec5 using "cmake ..", "make -j8", "make package". I then moved to a clean virtual mint 20.1 where 'lsb_release -a' gave:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20.1
Release:	20.1
Codename:	ulyssa
```
I removed the old OCPN install with "sudo apt purge opencpn", then installed this one with "sudo apt install ./opencpn_5.5.0-1_x86_64.deb" which was generated by the previous build commands.  I then ran OCPN using "./opencpn" from the build directory. Help/About gave:
[image]
I build OD in a clean directory using the following commands:
```
cmake -E env CLOUDSMITH_API_KEY='123' cmake ..
make -j8
make package
./cloudsmith-upload.sh
```
This built OD and created the deployable *gz file. The xml included was:
```
<?xml version="1.0" encoding="UTF-8"?>
<plugin version="1">
<name> OCPN Draw </name>
<version> 1.8.6.15 </version>
<release> 15 </release>
<summary> Drawing Plugin to allow extra objects to be drawn and used within OCPN </summary>

<api-version> 1.16 </api-version>
<open-source> yes </open-source>
<author> Jon Gough </author>
<source> https://github.com/ </source>

<description>
OCPN_DRAW (OD) is a plugin that extends the capability of OCPN to include non-standard object: Boundaries, Boundary Points, Text Points, Dead Reckoning paths, Electronic Bearing Lines (EBL) and Guard Zones. The plugin also has an API to allow other plugins to directly interact with it.
</description>

<target>linuxmint-gtk3-x86_64</target>
<build-target>linuxmint</build-target>
<build-gtk>gtk3</build-gtk>
<target-version>20.1</target-version>
<target-arch>x86_64</target-arch>
<tarball-url>
https://dl.cloudsmith.io/public/opencpn/ocpn_draw-alpha/raw/names/ocpn_draw_pi-1.8.6.15-linuxmint-x86_64-20.1-tarball/versions/1.8.6.15+1.20d6c99/ocpn_draw_pi-1.8.6.15-linuxmint-x86_64-20.1.tar.gz
</tarball-url>
<info-url> https://opencpn.org/OpenCPN/plugins/draw.html </info-url>
</plugin>
```
I then tried to install the plugin using Plugin Manager "import plugin" and got told the plugin was incompatible.
I can also use circleci local to build exactly the same as would be done by git commits/tags.
For this process I go to the root directory of OD and issue the following command:
`circleci local execute --job build-bionic-gtk3 -e CIRCLECI_LOCAL=true -e CLOUDSMITH_API_KEY="123" -e LOCAL_DEPLOY=true`
which builds OD in a docker image and instead of deploying to cloudsmith will deploy to a local directory. I also did it for focal with this command:
`circleci local execute --job build-focal-gtk3 -e CIRCLECI_LOCAL=true -e CLOUDSMITH_API_KEY="123" -e LOCAL_DEPLOY=true`
None of these worked and told me the plugin was incompatible.

leamas, 2021-04-27:
Could you please:
  - start opencpn using the --loglevel debug option.
  - Try to import your plugin, as described above.
  - Make a clean exit
  - Attach the logfile here so we can sort things out.

jongough, 2021-04-28:
I have identified the issue, it is with the new 'Settings/Select new flavour' button and associated processing, if I leave it as default I can install the mint build of OD plugin. If I change it to anything else the it will not install. This means that the compatibility check is now very specific to a particular flavour of linux and version. This is a strange change as it is picking on one of the less problematic area, i.e. version of linux, as many components built for Ubuntu 18.04 will run quite happily on Ubuntu 20.04 & Ubuntu 20.10. This is even more problematic with the length of time it can take to update each plugin to be 'compatible' with new versions. This applies to windows as well (I am running windows xp code on windows 10 quite happily).

There are more than enough builds currently going on for each change to a plugin, currently frontend2 does 15. It would make more sense, particularly now plugins statically link all the libs they need, to focus on what will cause a break, i.e. version of wxWidgets. Information that is available at build time can be added to the XML to help with identifying issues. And, if a particular build of OCPN is running, then by definition is must be compatible with the OS, so a plugin built to the same environment will be able to run. Also, newer builds of plugins 'may' run if they don't change or required critical items, i.e. version of wxWidgets, OpenGL (ES 2.x, Vulcan, etc.) which is not supported by OCPN.

So, now I can test installation of my plugins, but have to change the flavour each time I try to install a plugin not built by me, i.e. watchdog, weather_routing, squiddio, etc. which may be needed for compatibility testing of the ODAPI or JSON messages. This is all getting way too complicated and introducing new constraints for every change made. And no, flatpak, appimage, snap, etc. is not the answer as they all have a performance hit, take more memory, disk, internet bandwidth to download, etc. Many instances of OCPN are running on older equipment, or constrained environments, i.e. raspberry Pi, Phones, Tables, old lapops, etc.

leamas, 2021-04-28:
> So, now I can test installation of my plugins, 

Great!

> This means that the compatibility check is now very specific to a particular flavour of linux and version. 

As pointed out by Dave [above](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-825341743) this is the intended behaviour for reasons described in the same comment. I concur with his reasoning. 

Since the late testing issue is resolved, this is another example of a bug [beaten to death due to honest differences of engineering opinion, but without resolution](https://github.com/OpenCPN/OpenCPN/issues/2199#issue-857421833). According to this classification, this bug should IMHO be closed, leaving things at status qua.

jongough, 2021-04-28:
Not very user friendly or centric.

jongough, 2021-04-28:
More correctly, status confringetur

bdbcat, 2021-04-28:
Jon...
We expect developers to have knowledge and ability to adjust their system parameters for specific build or test tasks.  We do not expect "normal" users to do so, at least initially.
In general, we will always prioritize "user friendliness" over "developer friendliness".  That is a guiding policy.

Think of the basic linux user workflow:
1.  Enable the ubuntu PPA, for their running version of linux.
2. $sudo apt install opncpn
3. et voila, all the available plugins are present, and compatible.  No fuss...

jongough, 2021-04-28:
I am referring to users who are using other variants of Linux. I use Mint and get problems, there are many other variants out there that provide different desktops and pre-packaged applications that will have the same problem. The current process requires that you add 'CatalogExpert=1' to the config file (are users really catalog experts?) to be able to select different 'flavours'. If you do do this then you end up with a different constrained list of plugins depending on when builds were done. The base 'flavour' is ignored to display the different 'flavour' plugins. This will cause confusion., it did to me until I read the code and saw what it was doing.

The ppa will quite happily install Ubuntu 18.04 OCPN onto Ubuntu 20.10 as all the dependencies are met. The user does not need to know it is 'old'. Plugins however now react differently. Hence the need now to generate plugins for all available versions of linux.

bdbcat, 2021-04-28:
Jon...
Not trying to get the "last word", but I really do not understand your points.
From an earlier post:
"I have identified the issue, it is with the new 'Settings/Select new flavour' button and associated processing, if I leave it as default I can install the mint build of OD plugin. If I change it to anything else the it will not install. "
Fair enough, as intended.
Now you say users need to be "Catalog Experts" to select different flavors.  But why would a normal user (explicitly NOT a developer working on a plugin) need/want to select a different flavor, when we know that it will not work?  Indeed, that is precisely why we have the CatalogExpert config option available _only_ by editing the config file, presumably by an experienced user.
Please clarify...

bdbcat, 2021-04-28:
And this one:
"The ppa will quite happily install Ubuntu 18.04 OCPN onto Ubuntu 20.10 as all the dependencies are met. The user does not need to know it is 'old'. Plugins however now react differently. Hence the need now to generate plugins for all available versions of linux."
If a user configures their repos for the "wrong" ubuntu version in order to load an alternate version of OCPN, then one may reasonably expect undefined behavior, even if the app somehow manages to satisfy dependencies and install.  Can't think why one would do that, except to stress test the installation logic, and then report corner-case bugs.

closed 2021-05-03 by bdbcat

jongough, 2021-05-03:
Usability