OpenCPN/OpenCPN #2214: FS#2529 - Function of the hybrid display ENC (Navaids, Spot Soundings, Text) on MBTiles (SATIMG) set

bc38, 2021-04-22:
Overlaying ENC Objects over mbtiles set
Conditions
While using ENC / CM93 with MBTiles set in one Chart Groups.
inclusion of an option in the Chart Panel Options
[FS#2529](https://opencpn.org/flyspray/index.php?do=details&task_id=2529&opened=342&type%5B0%5D=&sev%5B0%5D=&due%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&reported%5B0%5D=
[image]
[image]

)

rgleason, 2021-04-22:
I believe this is a feature request. Please close this issue and post it in Tracker.  Issues are for active issues with the program, not feature requests.

bc38, 2021-04-22:
Created on 03/12/2019!
But CPN has remained from technology for 5 years!

rgleason, 2021-04-22:
It may be a good idea, but it is a Feature Request.  Please close it and use Tracker.

closed 2021-04-22 by bc38

rgleason, 2021-04-22:
Thanks, very much. hope you added it to tracker with a link to this post to save time.