OpenCPN/OpenCPN #2233: RFE: Re-define and automate portable mode.

leamas, 2021-04-24:
The portable -p mode is an important usecase. However, the semantics of it has become unclear after the introduction of the plugin installer. Also, the setup requires a lot of documentation, a sure sign of something being not that mature.

An improved UX could be
  - Users can create a portable directory (typically on a  usb stick) with a single command.
  - After that, user could take the stick, put it on another machine and just click on the opencpn icon to start it.

Perhaps this could be achieved by:

  1. Lean on the promise that opencpn -p will "just work" as expected.
  2. Define if user data, notably downloaded plugins, should be part of the things which should "just work".
  3. On Linux, define environment variables (notably XDG_DATA_DIRS) setup in portable mode. 
  3. Automate copying required data to the portable dir.
  4. Trick the opencpn binary copied to the portable dir to use -p by default.

transmitterdan, 2021-04-25:
Should this go into "Discussions"?

transmitterdan, 2021-04-25:
Portable mode has another important use case. Specifically, debugging on platforms like Windows. Portable mode is about the only decent way to debug without having to conditionally compile or edit the sources. I would seriously argue for preserving that. 

Can you list or link to some specific issues that are found in -p mode?  I haven't seen any in a while so I just want to review some examples.

Hakansv, 2021-04-25:
Apart from debug in for example another version of O Ive also seen the portable mode used in boats when a second screen is connected to the same PC. The portable runs on the second screen instead of duplicating the views.

rgleason, 2021-04-25:
@bdbcat @nohal   Yes I think it should be in Discussions  topic Q&A?

rgleason, 2021-04-25:
Perhaps Discussions  topic  "Ideas" is best.

leamas, 2021-04-25:
This is a targeted RFE. As such, it very much actionable and does not belong in Discussions. See #2216.

Note the workflow when a dev plan to do something: she first files an RFE to get more input, and later at some point files a PR. In this flow, the RFE is an important companion to the once-to-be-filed PR.

In other words: these kind of RFE serves the same role as (good) bug reports, which also (sometimes)  leads to a PR.

rgleason, 2021-04-25:
What is RFE?  Request for ?

leamas, 2021-04-25:
Request For Enhancement, common acronym for  these kind of issues.

leamas, 2021-04-25:
> Can you list or link to some specific issues that are found in -p mode? 

First and foremost: it's to complicated to setup. A  [simplified  UX](https://github.com/OpenCPN/OpenCPN/issues/2233#issue-866776735) is desirable..

Also: the current setup is from before the new plugin installer, and as such does not take downloaded plugins into account. This is perhaps as it should, perhaps not. But it needs to be discussed and defined.

Searching for issues gives us things like #2152, #1802.

rgleason, 2021-04-25:
This should be in the discussion thread.... however
It's too complicated to setup  -  No that is not true.
A simplified UX is desirable - Yes and we can set that up.  We will want multiple templates probably
   Bug
   Feature
   Etc
Also using templates will provide the same features as

"Note the workflow when a dev plan to do something: she first files an RFE to get more input, and later at some point files a PR. In this flow, the RFE is an important companion to the once-to-be-filed PR."

leamas, 2021-04-25:
@rgleason: this bug is a the portable mode, nothing else. Please stick to issue.

rgleason, 2021-04-25:
You are the one who diverged.

rgleason, 2021-04-25:
This "RFE" is not "actionable" in its current state and requires discussion.

leamas, 2021-09-03:
This is also related to IDE setup, the IDE needs basically to run in portable mode.

transmitterdan, 2021-09-03:
Especially so for Windows. Portable mode allows one to debug different branches and to more rapidly use git bisect to find bugs.

leamas, 2021-10-17:
Given use-cases discussed it seems that a portable installation should have it's own set of plugins as well. Note that current urban myths does not take this into account, the setups described so far only handles legacy plugins.

----
Can this be accomplished with 5.6.0?
  - Windows: portable mode uses executable path as homedir, might work already today.
  - Linux: sort of,  in portable mode plugins are installled in the data directory, and there is OPENCPN_PLUGIN_DIRS in the environment
  - MacOS: perhaps by manipulating $HOME, but that carries all sorts of risks.
  - Flatpak: yes, using XDG_CONFIG_HOME.

To summarize, MacOS might need some support like OPENCPN_PLUGIN_DIRS to implement portable plugin directories, but otherwise it should work (?)

Hakansv, 2021-10-18:
Just for info wo arguments:
Before the multicanvas function I used a portable instance all time. 
I made me an update routine to copy all needed files with my different PCs where Win-XP was involved. 
Of course maintained plugins was not involved at that time
[UpdatePortable.bat.txt](https://github.com/OpenCPN/OpenCPN/files/7362091/UpdatePortable.bat.txt)
.

leamas, 2021-10-18:
Since the basic logic in portable mode is to store plugins in the configuration directory also on Linux this is related to #1022

transmitterdan, 2023-04-18:
So are we saying this situation is normal or is there a bug?

leamas, 2023-04-18:
Frankly, I don't know.  The logic and thinking about portable mode is since before the managed plugins, and we have really not defined how these things relates to each other