OpenCPN/OpenCPN #2234: Build of OCPN 5.5.0 on Ubuntu gives wrong build architecture of x86_64

jongough, 2021-04-24:
Building OCPN 5.5.0 on Ubuntu 18.04 `cmake ..`  gives:
```
-- cmake version: 3.10.2
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- *** Host Build Architecture is x86_64
-- Setting C++11 standard via cmake standard mechanism
```
and 20.04 `cmake ..` gives:
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- *** Host Build Architecture is x86_64
-- Setting C++11 standard via cmake standard mechanism
-- Default compiler options:
```
However to install the build architecture needs to be amd64. When trying to build and install OCPN on a native Ubuntu desktop the following error is reported:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'opencpn:x86_64' instead of './opencpn_5.5.0-1_x86_64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 opencpn:x86_64 : Depends: libarchive13:x86_64 (>= 3.0.4) but it is not installable
                  Depends: libbz2-1.0:x86_64 but it is not installable
                  Depends: libc6:x86_64 (>= 2.27) but it is not installable
                  Depends: libcairo2:x86_64 (>= 1.2.4) but it is not installable
                  Depends: libcurl4:x86_64 (>= 7.16.2) but it is not installable
                  Depends: libelf1:x86_64 (>= 0.131) but it is not installable
                  Depends: libexif12:x86_64 (>= 0.6.21-1~) but it is not installable
                  Depends: libexpat1:x86_64 (>= 2.0.1) but it is not installable
                  Depends: libgcc1:x86_64 (>= 1:3.0) but it is not installable
                  Depends: libgl1:x86_64 but it is not installable
                  Depends: libglib2.0-0:x86_64 (>= 2.12.0) but it is not installable
                  Depends: liblz4-1:x86_64 (>= 0.0~r113) but it is not installable
                  Depends: liblzma5:x86_64 (>= 5.1.1alpha+20120614) but it is not installable
                  Depends: libpango-1.0-0:x86_64 (>= 1.22.0) but it is not installable
                  Depends: libpangocairo-1.0-0:x86_64 (>= 1.14.0) but it is not installable
                  Depends: libsqlite3-0:x86_64 (>= 3.7.3) but it is not installable
                  Depends: libstdc++6:x86_64 (>= 6) but it is not installable
                  Depends: libtinyxml2.6.2v5:x86_64 but it is not installable
                  Depends: libwxbase3.0-0v5:x86_64 (>= 3.0.4+dfsg) but it is not installable
                  Depends: libwxgtk3.0-gtk3-0v5:x86_64 (>= 3.0.4+dfsg) but it is not installable
                  Depends: libx11-6:x86_64 but it is not installable
                  Depends: zlib1g:x86_64 (>= 1:1.2.3.4) but it is not installable
                  Recommends: xcalib:x86_64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
The install package was built with the following commands:
```
cmake ..
make -j8
make package
```
Then an install of the built package on the machine it was created was done:
```
sudo apt install ./opencpn_5.5.0-1_x86_64.deb
```

nohal, 2021-04-25:
Jon...
Do you have some real reason to build the DEB package and install it this way instead of `make install`? Not that the broken dependencies are not annoying, but to address it is virtually impossible and it solves itself "for free" on Launchpad....

jongough, 2021-04-25:
I am testing the installation process for plugins and I want to be sure I know what OCPN code I am using, hence building OCPN and the plugins on Ubuntu 18.04 & 20.04 with nothing added except the build capability. Doing that I get the issue, if I build and run in my IDE it is all fine, but I am trying to checkout what the user will experience when installing some plugins.

nohal, 2021-04-25:
Does this solve it?
```
diff --git a/cmake/GetArch.cmake b/cmake/GetArch.cmake
index 61445e165..e750b4a9b 100644
--- a/cmake/GetArch.cmake
+++ b/cmake/GetArch.cmake
@@ -30,6 +30,7 @@ function(GetArch)
     endif ()
     if (EXISTS /etc/debian_version)
       set (PACKAGE_FORMAT "DEB")
+      set (ARCH "amd64")
       set (PACKAGE_RECS "xcalib,xdg-utils")
       set (LIB_INSTALL_DIR "lib")
       set(TENTATIVE_PREFIX "/usr/local")
```

leamas, 2021-04-25:
We have had a long discussion about this. Long story short, as things are now the ARCH needs to be x86_64, since this is what the plugins uses (and the xsd file defines). If you need to tweak the  CPackage .deb package using another architecture this could certainly be done, but changing ARCH to amd64 is to open a really big can of worms

nohal, 2021-04-25:
If it makes `make package` completely unusable, and there are people who obviously insist on using it, hence this issue, we have to either drop the package target or somehow fix it.

leamas, 2021-04-25:
Indeed. But it should probably be a "local" patch only affecting CPack, not a redefinition of ARCH which is used in all sorts of places..

leamas, 2021-04-25:
A quick git grep gives me (for oesenc):
    
    $  git grep CPACK | grep -i arch
    cmake/PluginPackage.cmake:  set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})

This line could perhaps be patched along the lines above, without re-defining ARCH

nohal, 2021-04-25:
`CPACK_DEBIAN_PACKAGE_ARCHITECTURE` must be `amd64` on everything Debian based, so if we need to keep `ARCH` set to `x86_64` there, we simply have to introduce another variable populated in`GetArch.cmake` that will be used in https://github.com/OpenCPN/OpenCPN/blob/master/CMakeLists.txt#L2777 instead of `ARCH`

nohal, 2021-04-25:
But FWIW we already have exactly what I did above for `arm64` / `aarch64` on everything non-Debian :)

leamas, 2021-04-25:
Indeed. But these paths are sort of dead ends, there are no modern plugins available. Which means that for  redhat/fedora or Suse we can whatever we like, not constrained by the xsd plugin definitions. The situation for Debian is constrained by the XML/XSD metadata,  though

leamas, 2021-04-25:
@nohal: something like this?

    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index 96ec6a6cc..289658002 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -2773,7 +2773,11 @@ if (UNIX AND NOT APPLE)
       # autogenerate additional dependency information
       set(CPACK_DEBIAN_PACKAGE_SHLIBDEPS ON)
       set(CPACK_DEBIAN_PACKAGE_RECOMMENDS ${PACKAGE_RECS})
    -  set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    +  if ("${ARCH}" STREQUAL "x86_64")
    +    SET(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
    +  else ()
    +    SET(CPACK_DEBIAN_PACKAGE_ARCHITECTURE ${ARCH})
    +  endif ()
       set(CPACK_DEBIAN_PACKAGE_VERSION "${CPACK_PACKAGE_VERSION}")
       set(CPACK_DEBIAN_PACKAGE_SECTION "misc")
       set(CPACK_DEBIAN_COMPRESSION_TYPE "xz")

nohal, 2021-04-25:
Yes, this should work, the only little downside is the increased amount of "logic" in the main `CMakeLists.txt`, but we can probably afford it in this trivial case.

PR #2240 (2021-04-26): Debian package: use correct amd64 arch (#2234).
Debian uses amd64, not x86_64 as architecture name. OTOH, we are
these days using x86_64 for plugins, also on Debian (since some
users were confused by the original amd64 name, as they had
Intel processors).

Stick to x86_64 for overall architecture name, but use the correct
amd64 for the Debian package.

Closes: #2234.
commit 9cf7ac57 (2021-04-26): Debian package: use correct amd64 arch (#2234).

Debian uses amd64, not x86_64 as architecture name. OTOH, we are
these days using x86_64 for plugins, also on Debian (since some
users were confused by the original amd64 name, as they had
Intel processors).

Stick to x86_64 for overall architecture name, but use the correct
amd64 for the Debian package.

Closes: #2234.
closed 2021-04-26 by nohal