OpenCPN/OpenCPN #2235: FR Dashboard - additional instruments

rgleason, 2021-04-25:
Way to expand the number of instruments when needed.
[Dashboard Instrument Capture Enum Full #1940](https://github.com/OpenCPN/OpenCPN/issues/1940#issuecomment-826095307)

nohal, 2021-04-25:
@rgleason can you please explain to me why are you doing this? What is the point of this issue and why having this empty shell better than the original issue you closed?
Please, really please, calm down and leave the issues alone for a moment as there are changes all over the place and we need them to settle down and see what works for us and what does not.

rgleason, 2021-04-26:
Good point.  I should have taken time to elaborate.  There are two choices, perhaps 3 to add instruments (and I believe that time is very soon, or past already).

1.  Dashboard Instrument Capture Enum Full #1940.  This change should probably occur at the same time as major release or change in wxWidgets or something, to avoid issues with users being left with an unworkable Dashboard.
2.  Adopt [DashboardT  2.0 by Petri (Canne)](https://github.com/canne/dashboard_tactics_pi)i,  dropping Dashboard.    DashT is fully tested, enhanced with an expanded feature set complete with documentation and used by many.  (after comparison  of the dashboard instruments for small changes.)  I believe DashT has a limit of 64 instruments, but could be expanded with item 1 above.

rgleason, 2021-05-05:
Thanks TDan!

closed 2021-06-06 by rgleason