OpenCPN/OpenCPN #2236: Plugin Management Focus  "meta-issue"

rgleason, 2021-04-25:
There are 119 Issues, almost  down to 100 Issues, perhaps is not a good measure of management, but it is an indicator.
We have 35 Plugin Management Items listed
12 from 2019.
The rest are mostly  2020.
@jongough @bdbcat @leamas  Can any of these be closed? Have any been resolved?
(Keep in mind I am just looking at numbers here and have not reviewed each issue.)

leamas, 2021-04-25:
I have 6 open issues, none of which about plugin management. I don't see that anyone of these should be closed.
Dave has 2 open issues, not much to focus on.
Jon has 34 open issues.

leamas, 2021-04-25:
OK, counting the plugin management issues:

leamas: 2
jongough: 28
bdbcat: 1

Time to focus.

rgleason, 2021-04-26:
I think we can close this now and continue to work on the list in due course. Thank you Jon for your review.
This closing gets us down to 110 Issues.

closed 2021-04-26 by rgleason