OpenCPN/OpenCPN #2241: [CI] Trusty builds fails

leamas, 2021-04-26:
The current trusty builds fails. At a glance, it seems to be about dependencies which are not available for trusty any more (?) There are two possible strategies to fix the failing build:

  1.  Fix the build errors somehow.
  2.  Drop the trusty build

My personal opinion is that we should drop it.  Trusty was released in 2014, and any user running it will have serious problems keeping the machine  alive. In particular, there are no security patches available since long.

There is also the argument that we have new Ubuntu release in the pipe, and there is a limit on how many versions we can support. For plugins, the trusty build also causes major headaches; many plugins has already dropped trusty.

In summary: I propose that we drop the trusty build, thus fixing this issue.

leamas, 2021-04-26:
It turns out that it's only the Travis builds which fails while the CircleCI works. This makes it a no-brainer to remove the Travis trusty build.

Still thinks we should drop it completely, though.

rgleason, 2021-04-26:
Jon got circleci builds of trusty going, however I was absolutely ready to drop it before that point. Circleci seems more reliable running trusty than Travis was (which required actions to re-run)

leamas, 2021-04-26:
@bdbcat: does this mean that we should just add new builds as new ubuntu versions are released, never drop anything?

bdbcat, 2021-04-26:
Rhetorical question gets rhetorical answer.
Yes.
But of course, we deal with the cards on the table as they happen.  There are no absolutes...

leamas, 2021-04-26:
Well, we obviously will have to live with the trusty builds for now-- will think about better arguments next time.

The builds work. Closing

closed 2021-04-26 by leamas