OpenCPN/OpenCPN #2248: flatpak version of logbookkonni doesn't start in opencpn

filochard, 2021-04-27:
Hi
I'm using opencpn in Mageia8 
For years, I have been building  rpms of opencpn and plugins for my own use and helped the official maintainer to improve the rpms provided by Mageia

We are thinking about moving to flatpak ... so I tried it
The plugin manager is a real improvement...
 but Logbookkonni-plugin doesn't work when installed and activated : no way to use the preference button to add the Layouts ... and the Layouts.zip file is not provided
The logbook icon is not added in the icons tray

PS If it is of any use here's what spec file we wrote for Mageia :
[opencpn-logbookkonni-plugin.txt](https://github.com/OpenCPN/OpenCPN/files/6385025/opencpn-logbookkonni-plugin.txt)

NB : when using the rpm you can't install automatically the Layouts data in your user directory and you need to add them by looking for the zip file !
With flatpak those Layouts can be included in the plugin directory itself in :
/home/username/.var/app/org.opencpn.OpenCPN/data/opencpn/plugins/logbookkonni_pi/data/

filochard, 2021-04-27:
PS : if the Layouts are automatically provided in the user directory by flatpak, this will need to modify the plugin so that it doesn't need anymore to look for them and add them before starting...

leamas, 2021-04-27:
hm... first of all: this is a problem in the flatpak'ed version of  Logbookkonni.  This is built in the separate project at https://github.com/rgleason/LogbookKonni_pi; please close this bug and file the issue there instead.

That said, I can confirm that the Logbook flatpak plugin is broken, it does not  open the Preferences dialog.

filochard, 2021-04-27:
Hi Ieamas
Yes indeed it's a problem with the flatpak'ed version
and it looks the same as for MacOS
(I didn't have this problem for Linux with my rpms for Mageia )

My first intention was to explain this  issue there : https://github.com/rgleason/LogbookKonni_pi
But I didn't found a way to do this : this project has no "Issues" menu on its code page (only "Pull requests" that doesn't match for what I meant to do...
That's the first time I find a project in github without a way to report an issue,

I thought that the right place was here ...

I don't know what to do...

leamas, 2021-04-27:
PLease stay tuned. We are in the process of changing our policies; there are not only technical bugs but also this kind of stuff.

@rgleason: Can you enable issues on LogbookKonni_pi

rgleason, 2021-04-27:
Yes, I can.  I suppose as a maintainer repository that is OK?  Note: I will need to do this for Sean's and the other plugins for consistency.  I was letting issues be in the OP repository, particularly for Sean's plugins, but Logbook is a special case because Konni is no longer around and Peter Tulip is available but not maintaining right now, so I am.

rgleason, 2021-04-27:
Logbookkonni  Issues https://github.com/rgleason/LogbookKonni_pi/issues

Last I knew Logbook was working on all 4 cylinders. ..Then OpenCPN changed?  This isn't supposed to happen.  That doesn't sound right.

rgleason, 2021-04-27:
And Preferences Open

[image]

rgleason, 2021-04-27:
FYI
The other alternative for filing issues has always been
https://github.com/ptulp/LogbookKonni_pi/issues
since I've started working on Logbook

leamas, 2021-04-28:
To iterate, we should IMHO now:
  - File a new issue at https://github.com/rgleason/LogbookKonni_pi/issues (or possibly  https://github.com/ptulp/LogbookKonni_pi/issues) with a link to this issue.
  - Close this issue

filochard, 2021-04-28:
Thanks Ieamas
I filled an issue to rgleason

closed 2021-04-28 by filochard