OpenCPN/OpenCPN #2249: Linux - Dragging toolbar almost impossible with OCPN 5.5.0

jongough, 2021-04-28:
Trying to drag to toolbar around the screen is very difficult due lag, the toolbar lags behind the mouse pointer, no indication that the toolbar has been selected for dragging (need to move the mouse to see if the toolbar moves, most of the time it does not). Sometimes a popup message box showing "Hide Toolbar" can be seen. Perhaps a "drag handle" could be shown if the mouse pointer hangs over the toolbar?

leamas, 2021-04-28:
See also #2168

bdbcat, 2021-05-04:
Somewhat improved by https://github.com/OpenCPN/OpenCPN/commit/3842d2f9d11e03f7abf39c404e2657d482b99348
Still, one needs to drag the toolbar slowly.  The lag, or speed, depends on the platform, since moving of a top-level-window is a display manager function.
Also, please note that the "hot spot" for dragging the toolbar is _only_ on the "hamburger" icon.  No other tools are responsive to drag operation.