OpenCPN/OpenCPN #2250: PIM RFE Enable is not readily available and requires resizing

rgleason, 2021-04-28:
Ov5.5  Ubuntu 18.04
See screens.  Could  Enable/Disable become a "Button" left justified at the end of Preferences, Website, etc.?
[image]
[image]
[image]

CarCode, 2021-04-28:
Why you call this "enhancement"?

rgleason, 2021-04-28:
Ubuntu 18.04  OpenCPN v5.5  Now, after restart of OpenCPN, I find that the "Enable" checkbox is floating on the right side of the window, provided that there is at least enough width for the Plugin information.  It seems to be working properly now. I am assuming the previous situation was an anomaly, so I am closing.
[image]

closed 2021-04-28 by rgleason