OpenCPN/OpenCPN #2255: Plugin Installer plugins not removed when OCPN is uninstalled on Win10

rgleason, 2021-05-02:
Please refer to https://github.com/OpenCPN/OpenCPN/issues/1539#issuecomment-830740947
A simple question  Remove Plugins?  Yes No  before running install.
The alternative is to put it in the manual which will never be remembered.

leamas, 2021-05-02:
Duplicate of #1556

Hakansv, 2021-05-02:
Both these "issues" are more of a pseudo discussion for a average user. They wont care if the old one is there or lost as long the new one is working. The discussion is also only valid during the transition period to managed plugins that would be over now. If a updated plugin is not working we can mostly offer a tarball for the previous one instead of the old package.
I suggest closing and focus on O version 5.5.... beta test.

rgleason, 2021-05-03:
Closing, leaving gigabytes of plugins on the hard drive.

closed 2021-05-03 by rgleason

bdbcat, 2021-05-03:
Exaggeration. Maybe 100 MBytes, if a _LOT_ of plugins were installed.

rgleason, 2021-05-03:
RPI?

Hakansv, 2021-05-03:
That's you and me and other testers, We know how to delete it. A "average" user may have one two five old ones with no disturbance?

rgleason, 2021-05-03:
I can't understand the resistance to good practice.  But I will leave it there.

bdbcat, 2021-05-03:
Actually, this is not a trivial task to implement and test.  So the "benefit/expense" ratio is small...
Much better spend the minimal programming resources we have on more visible and critical issues at OCPN runtime.

rgleason, 2021-05-03:
OK, sorry