OpenCPN/OpenCPN #2256: PIM  Testing - Need Comprehensive failure list for all  PIM Flatpak plugins

rgleason, 2021-05-03:
Heads up
Flatpak Plugins failing due to user writing data to locked data dir -TESTING Needed

I have not seen reference and advice regarding this in the Dev Manual. Perhaps it exists. Alec just wrote about this in the link below.

This problem is new to me, and because not many Flatpak users are using plugins, the plugins are often a mess with regard to running properly in Flatpak, due to it running in a container and locking the data directory to user configuration writes.

We need a comprehensive  list of PIM plugins that don't work in Flatpak.

https://github.com/rgleason/LogbookKonni_pi/issues/10

leamas, 2021-05-03:
Although this problem is visible on the Logbook flatpak build it might be present also in other plugins. Basically, writing in the installation data directory is a bug on all platforms although only legacy Linux plugins have full operating system support to enforce this. 

To get things right, we need to make sure that plugins:

  - Don't write any data in the directory obtained using GetPluginDataDir().
  - If required, writes data does this into the configuration directory.
  - That  GetpPrivateApplicationDataLocation() function is used to locate the configuration directory
  - That files created by the plugins are placed in `$CONFIGDIR/plugins<plugin name>`

This scheme should keep things sound and also work on all platforms. And yes, it needs to be documented...

filochard, 2021-05-03:
Hi Alec and Rick
You may add watchdog plugin 
https://github.com/seandepagnier/watchdog_pi
the plugin data are in the right place but the user's created files don't 
see https://github.com/rgleason/LogbookKonni_pi/issues/6#issuecomment-830917980

leamas, 2021-05-03:
No point making a list. Please file an issue against the watchdog plugin, perhaps with a link to this issue.

filochard, 2021-05-03:
OK
Thanks

rgleason, 2021-05-03:
News to me.. "writing in the installation data directory is a bug on all platforms"  

I stand corrected. Is any of this in our new Dev Manual somewhere?

leamas, 2021-05-03:
No, but it's basically just common sense. 

On legacy Linux, it's enforced since the data area is read-only. 

On legacy windows, it might be enforced (it should, really) if user runs without administrator privileges.

On other platforms writing data breaks basic assumptions, and also leads to problems since the data files might be removed on plugin uninstall -- which means that an upgrade might destroy all plugin data.

In short: plugins must not write in the data directory, they should use the configuration directory for that. The same is by the way true also for core opencpn.

leamas, 2021-05-03:
> News to me.. "writing in the installation data directory is a bug on all platforms" I was told to use GetPluginDataDir, when did that get changed?

It has not changed. But the data directory obtained by GetPluginDataDir is read-only.

rgleason, 2021-05-03:
Now I know what my problem has been.  My favorite user account has admin privileges, which had some real benefits, except this problem, where I don't experience what the normal user experiences..

rgleason, 2021-05-04:
Not working in Flatpak   need   GetpPrivateApplicationDataLocation()
Watchdog, 
logbookkonni, 
celestial-navigation

filochard, 2021-05-04:
here is an example of how to fix these issues : (the weather_routing plugin doesn't have this problem)
https://github.com/rgleason/LogbookKonni_pi/issues/10#issue-874146894

rgleason, 2021-05-29:
Weatherfax updated and working

rgleason, 2021-06-06:
AIS Radar working.

rgleason, 2021-07-19:
Not working in Flatpak need GetpPrivateApplicationDataLocation()

    Watchdog,
    celestial-navigation

Will continue working on these... closing

closed 2021-07-19 by rgleason