OpenCPN/OpenCPN #2257: Options->Plugins. Selected plugin. Details for update and activation is drawn outside Options window.

Hakansv, 2021-05-03:
On Win10. Recent master code.
The rightmost tools are not viewed in the default dialog width, pic 1. One must rezise Options window Pic 2.
This behavior is rather new. Two weeks?

[image]

[image]

leamas, 2021-05-03:
@hakansv: Have you figured out how to make a git bisect? Using this, you should be able to pinpoint the exact commit which caused this.

Hakansv, 2021-05-03:
Alec.. Yes I know I can walk down commits to find out. My first assumption was though someone may say "Hmm... I was there. Maybe I should have a second check?  :)
But if not I'll of course see what I can find.

rgleason, 2021-05-03:
Hakan, I noticed this and was going to make a report about it, and then the next time I opened Opencpn, the problem was gone for some reason.  I haven't seen it reappear, but will try to notice this.  Is it some initialize issue.

Hakansv, 2021-05-03:
For me on Win10 the commit:  4397d4044  "Improve plugin panel layout rules.  Addresses #1812"
is the one that has changed the description behavior. If I go back to the old code all is good as of the pict.

I understand it was a reason to change but perhaps for another OS and/or another char width?
Could there be another "smart sizing" that fits all? (I tried the "bCompact" code wo success.)

The old code inserted:
```
// This invocation has the effect of setting the minimum width of the descriptor field.
m_pDescription = new wxStaticText( this, wxID_ANY, m_pPlugin->m_short_description, wxDefaultPosition, wxSize( 40 * GetCharWidth(), -1), wxST_NO_AUTORESIZE );
//m_pDescription = new wxStaticText( this, wxID_ANY, m_pPlugin->m_short_description, wxDefaultPosition, wxSize( -1, -1), wxST_NO_AUTORESIZE );
```

[image]

bdbcat, 2021-05-03:
Hakan...
Thanks, yes, that is the important line of code. Different things on different platforms.  I'll take another look.
Probably could do something like:
`m_pDescription = new wxStaticText( this, wxID_ANY, m_pPlugin->m_short_description, wxDefaultPosition, wxSize( g_options.GetClientSize().x / 2, -1), wxST_NO_AUTORESIZE );`

Not pretty code, but that is sometimes needed for wxWidgets cross platform....

Hakansv, 2021-05-03:
Dave..
Yes that did the trick (Although wxSize( g_options->GetClientSize().x / 2, -1)  instead) I can't say I really understand the difference. It was by changes :)
Hopefully OK on other platforms.

Håkan
[image]

bdbcat, 2021-05-04:
How is it on latest commit?

tai-fun, 2021-05-04:
Dave
I have the same issue on Kubuntu 20.04. If anything, the latest code is
worse. The small square indicating the status is mostly outside the window
that is expanded to max size.
[image: Screenshot_20210504_083831.png]

Hakansv, 2021-05-04:
Dave..
On Win 10 it's not good. It's like my first shot here. 
On RPi4 Buster with a "normal" screen it's fine.
On RPi3 Buster with the 7" screen, 212 mm width, it's not good, see pict, 

Strange the "OK, Cancel, Apply" buttons are always  positioned inside the dialog frame on all system but not "our" icons?

[image]

leamas, 2021-05-04:
Flatpak (gtk2)  looks fine,  current master (3842d2f):

[image]

bdbcat, 2021-05-04:
Taifun...
Did not get the screenshot...

tai-fun, 2021-05-04:
Dave
Check my original post on git-hub again. The screenshot should be there now.

bdbcat, 2021-05-04:
Got it.  Thanks

bdbcat, 2021-05-05:
Please check with latest commit.  Fairly large refactor of the plugin list, simplified.

leamas, 2021-05-05:
Flatpak - 18.08 -gtk2: Still OK

Hakansv, 2021-05-05:
Dave..
In Win 10 and RPi4 at default dialog width. If I slim down the width the icons are falling outside but I think we can live with that.
Sorry I can't reach my RPi3 from remote today but yesterday it was similar to my Win.
If nothing else from @tai-fun I'll close.
Thanks.

A note:
Just a fine tune?  wxALIGN_RIGHT  is not needed here:
While debugging for another thing I run into below assert:
for pluginmanager.cpp:6389
itemBoxSizer01->Add(m_itemStatusIconBitmap, 0,  wxALIGN_RIGHT | wxEXPAND|wxALL, 20);

 wxASSERT_MSG
        (
            !(flags & wxALIGN_RIGHT),
            wxS("Horizontal alignment flags are ignored in horizontal sizers")

tai-fun, 2021-05-05:
Kubuntu 20.04
The layout is now back to normal, or as it was a few weeks ago.
The only thing missing is the "roll over" text explaining the rightmost
icon.

Hakansv, 2021-05-05:
Solved by c30498f1

closed 2021-05-05 by Hakansv