OpenCPN/OpenCPN #2269: PluginManager: segfault on OpenCPN as soon as I install a plugin

gromain, 2021-05-13:
Hello,

I'm running OpenCPN compiled from source with https://aur.archlinux.org/packages/opencpn-git/ . The script to build is available here: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=opencpn-git .
I'm running Manjaro (a derivative from Arch Linux).

I have to enable CatalogExpert=1 to have access to the plugin catalog, and changing my os  (since my OS isn't in the standard list of plugins).
As soon as I download and install a plugin, OpenCPN crashes. When I restart it, it works fine, until I go into Settings/Plugins. At which point OpenCPN crashes again.

(Unrelated, and I'll open an issue for that, there is a checksum error for the package oesenc for debian architecture)

This has been reproduced on my Raspberry Pi4, running ManjaroARM 64 bits as well.

In the example, I've tried to install the polar package (after having the checksum failure). Restart happened after 17:34:19.

Console output is:
```
 SENC: nCPU: 12    m_max_jobs :11

(opencpn:78870): Gtk-CRITICAL **: 17:26:38.443: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar

(opencpn:78870): Gtk-CRITICAL **: 17:26:42.455: gtk_tree_model_iter_nth_child: assertion 'n >= 0' failed
[1]    78870 segmentation fault (core dumped)  opencpn -l debug
```
Message is the same everytime, segfault happens when I click on the Plugin tab in the Settings menu.

I attached the full log from `opencpn -l debug`. Restart happens at 17:36. 
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/6473701/opencpn.log)

Also, you will see that the log is full of a repeating pattern:
```
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: darwin, abi_version: 10.13.6, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:327 Returning fail for debian-x86_64
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:333 Found matching abi version 10
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: ACCEPTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: flatpak-x86_64, abi_version: 18.08, major ver: 18
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: mingw, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: msvc, abi_version: 10.0.14393, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:327 Returning fail for debian-x86_64
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: raspbian-armhf, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: raspbian-armhf, abi_version: 9.4, major ver: 9
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-arm64, abi_version: 18.04, major ver: 18
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-gtk3-x86_64, abi_version: 18.04, major ver: 18
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-gtk3-x86_64, abi_version: 20.04, major ver: 20
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-x86_64, abi_version: 14.04, major ver: 14
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-x86_64, abi_version: 16.04, major ver: 16
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED: Plots
17:27:06.994   DEBUG PluginHandler.cpp:198 Host: init: abi: debian-x86_64, abi_version: 10, major ver: 10
17:27:06.994   DEBUG PluginHandler.cpp:173 Plugin: setting up, name: Plots
17:27:06.994   DEBUG PluginHandler.cpp:174 Plugin: init: abi: ubuntu-x86_64, abi_version: 18.04, major ver: 18
17:27:06.994   DEBUG PluginHandler.cpp:220 Checking for debian and ubuntu
17:27:06.994   DEBUG PluginHandler.cpp:341 Found Ubuntu version matching Debian host
17:27:06.994   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: ACCEPTED: Plots
```
This repeats several times until (seemingly) no end.
Is this normal or should I open a new issue for this?

leamas, 2021-05-14:
Segfaulits after using CatalogExport=21 is more or less expected. Using these functions to circumvent the built compatibility  checks basically puts you in an unsupported usecase, so to speak.

The supported way to run OpenCPN on distros like Arch is using Flatpak. Sorry for for being rude in European way.

gromain, 2021-05-14:
Hi @leamas,

Yes, I figured later that Flatpak was now the way to go for those OS. 

> Sorry for for being rude in European way.

So yeah, don't be sorry (and it's not rude), it's perfectly OK to say "sorry, your use case is not supported". I'll close this then, and try to setup Flatpak (though my initial tests were not great to say the least, but I'll open issues if needed once I solve my other problems).

However, just to add something on the way things work. I would understand a segfault coming from me trying to use a plugin compiled for arm64 on a x86_64 machine (for example). But that was not the case here. Everytime I made sure to choose the proper architecture (if not the proper distro). How is a segfault possible if I use the same architecture? Is this expected or could there still be a bug somewhere? Maybe the plugin is killing itself and makes OpenCPN crash if no compatibility is detected by the plugin itself.

Also, is it normal that there is continuous loop of plugin version compatibility checks?

As I see it, the expected behaviour when choosing the same underlying physical architecture would be the plugin to run with no segfault (even though there could be missing libraries or other incompatibilities that would make the plugin not run, but as you said, unsupported use case!).

leamas, 2021-05-14:
Long story short: the plugin is compiled against both the plugin ane the libraries of the operating system. There  are no ABI guarantee between different distro releases, and even less so between different distros. This makes segfaults likely when loading a plugin compiled for another context.

gromain, 2021-05-14:
OK I see, it makes perfect sense then. Not a bug, a feature! :-)

I'll update the packages (both opencpn-git and the plugins) in the Arch User Repository to warn (at least) the users that this method are not officially supported and that Flatpak is the official way to go for this OS (if you want/need support at least).

Thanks for taking the time to explain!

closed 2021-05-14 by gromain

leamas, 2021-05-15:
Thanks for informing your Arch users!