OpenCPN/OpenCPN #2270: PIM  LogbookKonni  -

rgleason, 2021-05-14:
LogbookKonni v1.4.13.36 and v1.4.14.1  both have the same issue in opencpn.log file.

`dlunix.cpp:162 /home/rgleason/.local/lib/opencpn/liblogbookkonni_pi.so: undefined symbol: _ZN9wxControl7SetFontERK6wxFont`

See this old report:
https://www.cruisersforum.com/forums/f134/rpi-no-plugin-to-install-243202.html#post3289423

And other similar ones:
FS#2724  https://opencpn.org/flyspray/index.php?do=details&task_id=2724

About Photolayer  https://githubmemory.com/repo/Rasbats/photolayer_pi/issues/34?page=2

https://www.gitmemory.com/issue/OpenCPN/OpenCPN/1967/655742707

leamas, 2021-05-14:
This is a plugin problem, not related top OpenCPN.

closed 2021-05-14 by rgleason

leamas, 2021-05-15:
OK, plugin closed as it should. 

The basic problem is that these plugins depends on libraries which are not included, neither in the plugin nor in OpenCPN core. A plugin with such dependencies must bundle them with the plugin one way or another. Options includes re-building from source and/or bundling pre-build binary blobs.  IIRC, photolayer contains examples on both strategies.

In the end, this means that all plugins needs to work as windows plugins always have done.