OpenCPN/OpenCPN #2271: Close OpenCPN does not work when Options is Open

rgleason, 2021-05-14:
Why is this a problem?
Often when using [see Ubuntu 18.04](https://github.com/OpenCPN/OpenCPN/issues/1787) the Options Menu gets hidden (there is no known way to get it to reappear)
I need to KILL the Opencpn Application in this situation because a simple CLOSE does not work. OpenCPN appears to be totally frozen, but that is becuase the Options menu is open somewhere but not accessible.

A well behaved application would shutdown in this event.

leamas, 2021-05-17:
This can be reproduced on Flatpak. After opening the Options window, the main window does not respond to the clicking the X in main window upper-right corner i. e., it does respond to the EVT_CLOSE generated for this click. Seems like a plain bug in the event handling.

Although not a major problem as long as the Optioins window is visible it's OMHO still a bug.

leamas, 2021-05-17:
> With a very very very slow virtualbox ubuntu these problems make it impossible to test plugins.

In my experience, the Virtualbox VMs runs in more or less native speed. Are you using Hardware Virtualization? Have you tried assigning it more memory? Have you installed the Guest Additions?

rgleason, 2021-05-20:
Alec, vbox with guest additions and hardware virtualization. I may update to ubuntu 20.04 to see if there is an improvement. I think it is 28 mb memory.

leamas, 2021-05-20:
It should  absolutely run at native speed.  If not, something is severely broken. Hard to say what, though. 

Have you assigned 28 MB to the guest? That's far more than needed. My guests runs fine at 3-4 MB.

leamas, 2021-05-21:
Oops... That is, my guests runs fine at 3-4 Giga Bytes (not mb). Same error on your side? Otherwise, 20Mb certainly isn't enough for a guest to run smoothly.

rgleason, 2021-05-22:
Alec, it's a bit off topic, but I am pretty sure 21gb is dedicated to virtual box. I probably need to do maintenance.

bdbcat, 2021-06-07:
Back to the original issue:
I think that this may be an issue for the DisplayManager (DM) on the OS, and its interaction with wxWidgets.
IIUC, If a modal dialog is active, then clicking any other windows will not generate a wxWidgets event.  Not even a close event.
Simple testing shows this to be true for OCPN on most (all?) linux DMs.  Testing other GUI apps (e.g. kate) shows the same behavior.

So, not much can be done here, without some new understanding.  Rather not have OCPN interact with the DM directly...
Thoughts?

leamas, 2021-06-07:
hm... you are right. A modal dialog by definition freezes input from other windows, including the main window here.  

Perhaps we could just give some visual feedback that the main window is inactive when the options window is open? The user problem is that the main window looks active, but it isn't. If it's greyed or so, the situation would be obvious.

Of, course, we also need to make sure the modal dialog is on top.  A hidden modal dialog is a rather high priority bug as I see it.

rgleason, 2021-06-07:
Maybe this should be closed as "not implemented"?

rgleason, 2021-06-07:
Alec agreed, would be best!
1. Grey out main window
2. Make sure the modal dialog is on top. A hidden modal dialog is a rather high priority bug as I see it.

Hakansv, 2021-06-07:
Historical I think there was an attempt to not disturb an activated route and thus deactivate an Auto Pilot when Options window is open.
Could it be that functionality has made the Options not on top in some system? In Windows it's always on top due to my experiences.

bdbcat, 2021-06-07:
If your memory serves, can you elaborate on this?
"Historical I think there was an attempt to not disturb an activated route and thus deactivate an Auto Pilot when Options window is open."
Indeed, the options dialog is not currently forced "OnTop", but IIRC it used to be.  Wondering why we changed that...

rgleason, 2021-06-07:
Was it for small RPI screens?

Hakansv, 2021-06-07:
Dave..
My memory is not bad just somewhat short......
So I can't elaborate more than I remember the AP was complaining about no received messages years ago but nowadays the Options Window can be open and the AP messages are still flowing thus not disturbing the AP when navigating a route. 
But what was done to achieve that I don't know. My comment was meant to anyone to code carefully if anything is changed about how the Options window is behaving.

transmitterdan, 2021-06-10:
If O stops processing incoming data the messages pile up in a queue. When whatever holds up processing ends, such as new chart scanning,then a flood of messages are processed. That can lead to some undesirable behavior.

If the options dialog were to block data stream processing it could really be a bad thing. The queue of unprocessed messages could grow to a huge length.

I think we just need a way to shut down options dialog when the program tries to exit. It should not be that hard to post a close message to options. The complications are many because there are child dialogs, including plug-ins,that can be stacked on top of Options.

Maybe wxWidgets has a way to post a close request to all children?

O also does some automatic z-order "surface/submerge" calls that may be the cause of certain window managers burying Options. Perhaps it would be an idea to suspend any z-order calls when Options is visible? Is this issue still found in any modern window manager?

leamas, 2021-06-10:
IIUC, at least on GTK a modal dialog basically disables all other windows. This would mean that there are no events generated when accessing other windows and thus no point where we could post the "Close all windows" event.

The correct fix would be to make to Options window non-modal. However, this is probably not trivial (?)

As I said earlier: The basic problem is that the main windows *looks* active, but it is in fact disabled when Options is open. Some visual feedback in the man window, like "greyed-out" would make the situation much more clear for the user, and is is perhaps easier to implement.

That said, the important issue is the z-stacking. We should focus on this -- as long as the Options windows is visible it's not a serious bug, merely an inconvenience. OTOH, I can't help here...

leamas, 2021-06-10:
@rgleason:

> Often when using see Ubuntu 18.04 the Options Menu gets hidden (there is no known way to get it to reappear)
I need to KILL the Opencpn Application in this situation because a simple CLOSE does not work. OpenCPN appears to be totally frozen, but that is becuase the Options menu is open somewhere but not accessible.

As a starter: are there some  reproducible steps which causes this behaviour?

rgleason, 2021-06-10:
Launching!... will report in more detail later

bdbcat, 2021-06-10:
@transmitterdan 
The wxWidgets message queue is fully active while options dialog is open.  So no worries there.
The issue here is that when options is open as modal dialog, the main OCPN frame is disabled for ANY user input.  So there is no way to initiate a shutdown from the main frame.
On GTK, the title bar of the main OCPN window is grey when options dialog is active. That is the only hint available.
I actually think that this is only a problem on modern Ubuntu, with that particular DM and z-order processing.  Possibly also on MacOS, with its totally strange z-order rules.

transmitterdan, 2021-06-11:
@bdbcat I have seen the NMEA stream get bound up and take a long time to unwind itself. I'll do some testing to see what causes it. It might have been the chart ingestion after a chart download or install of new charts.

bdbcat, 2021-06-11:
I'd like to see that demonstrated.
Dave

rgleason, 2021-06-11:
Dave,  I've ended up frozen a dozen times because Options window turns into a thin white line due to my clicking on something (need to find out exactly what) and then Opencpn 5.5.5 won't close at the upper X and the program is essentially frozen so I need to "Kill" it. 
[image]

rgleason, 2021-06-11:
I know the white line is the Options window because when I hover over it I see the green arrow appear and the white sizing curser but I can't grab it to make it wider.  Also I can see the white line has turned to black at the top for the header, so it has got to be the Options Window that I can't seem to get to open more.

rgleason, 2021-06-11:
Now I figured one part of this out. When I grab and pull down and across on a diagonal I can open the Options window up.  So that part is figured out now, but why does this happen?  More work to do on that.

[image]

rgleason, 2021-06-11:
Is there a way to Alt-Tab and walk through all open menus and change the focus to the Options Menu (similar to windows)?

rgleason, 2021-06-11:
I neglected to say this is being run in a virtualbox 6.1.22 and it is slow.

rgleason, 2021-08-02:
Should I close this now?

bdbcat, 2021-08-02:
Yes, close it

closed 2021-08-02 by rgleason