OpenCPN/OpenCPN #2273: warning when installing flatpak version of opencpn and then when running it

filochard, 2021-05-17:
When installing opencpn I got this message

Info: org.freedesktop.Platform//18.08 is end-of-life, with reason:
   The Freedesktop SDK 18.08 runtime is no longer supported as of 31th August 2020. Please ask your application developer to migrate to a supported platform.

When running I get this : 

(opencpn:2): dconf-WARNING **: 12:36:38.823: failed to commit changes to dconf: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: org.freedesktop.DBus.Error.ServiceUnknown
repeated hundred times

rgleason, 2021-05-17:
I guess we should upgrade to 20?

leamas, 2021-05-17:
This has been discussed. We need to update, but when?
  - Updating to 20.08  now will force rebuilds of all plugins, a bad thing since we are late in the release cycle.
  - Updating to 20.08 directly after next release fits our release schedule better, but will leave users with this warning (and related problems when for example updating their bas os which might not support 18.08)

There are no good options. Question is which is less evil. @filechard: What's your base OS/version?

filochard, 2021-05-17:
Hi
I'm using up-to-date Mageia8 (it's near from Fedora 34 : rpm distribution  not deb like debian or ubuntu)
the basis is : 
Kernel    5.10.33
dbus1_3   1.13.18 
Xorg    1.20.10
mesa    20.3.4
nvidia    460.80
libglvnd   1.3.2 (providing a vendor-neutral OpenGL wrapper )
qt    5.15.2
glib2.0   2.66.8 
glibc   2.32
gcc   10.2
boost   1.75.0
gtk    3.24.24
wxgtk   3.1.5
python    3.8..7
java-openjdk    11.0.10
flatpak    1.10.2
fontconfig     2.13.93

leamas, 2021-05-17:
I know Mageia, no problem. Will make a test with Fedora 34 tomorrow to check the situation there.

leamas, 2021-05-18:
Trying Fedora 34, no immediate problems here. In particular, no warning for unsupported 18.08

Looking more carefully at the warning it says _The Freedesktop SDK 18.08 runtime is no longer supported as of 31th August 2020. Please ask your application developer to migrate to a supported platform._

As I read  this is 18.08 supported until  end of August. Tentative conclusions:
- Updating in next release cycle makes a lot of sense.
- We could publish beta builds of main program + 20.08 in the beta channel and corresponding plugin builds against 20.08 in the beta catalog.
- If Mageia 8 does not support 18.08 now: isn't this then a Mageia bug?

filochard, 2021-05-18:
Hi
The warnings appear in the console from which I install and then launch opencpn
(I prefer the use of a console instead of a single command line : this way I can get this kind of messages...)

About the freedesktop SDK :  version 18.08 WAS supported until August **2020**... it is not anymore and now it's the 20.08 version that is supported 
Mageia 8 uses the last Flatpak version 1.10.2 from March 2021 which allows to use the 20.08 version of Freedesktop SDK
(the warnings doesn't prevent opencpn to work but inform that it was built with an old version and needs to be rebuilt with an up-to-date flatpak builder )

leamas, 2021-05-18:
>  version 18.08 WAS  supported until August 2020.

Silly me. Sorry.

As I said, I agree that we need to update. The question is just if we need to update before the traditional spring release  which might happen in a not too distant future. See [above](https://github.com/OpenCPN/OpenCPN/issues/2273#issuecomment-843056018). 

My idea is that we start the next cycle as soon as the release is out of the doors. That should give us some time to rebuild plugins using 20.08. It also probably means that we need to make an new flatpak  20.08 release later this autumn or so, based on the same sources. 

Thoughts?

gromain, 2021-05-19:
Just as food for thoughts, is it possible to do a test build with 20.08 of the plugins and see if anything breaks? Maybe we get lucky and nothing needs to be done to update?

leamas, 2021-05-19:
@bdbcat: Thoughts? How bad would it be to force a rebuild of all flatpak plugins against 20.08? Here and now?

bdbcat, 2021-05-20:
I think we can at least try some plugins, and see what roadblocks appear.
I'll take a PR against oeSENC, as a trial.

leamas, 2021-05-20:
I don't foresee any roadblocks as such, the attempts we have done in shipdriver does not indicate anything of that kind. It's just a question of timing and raw work, I guess.

I'll make a PR. Stay tuned.

leamas, 2021-05-20:
@bdbcat: There is also https://github.com/OpenCPN/plugins/pull/367, required to be able to verify the new metadata with target-arch == aarch64.

leamas, 2021-05-22:
@gromain: There is a test release of oesenc available at https://github.com/bdbcat/oesenc_pi/pull/116. If you have any oesenc charts: could you possibly do some testing?

leamas, 2021-05-22:
@filochard: ^

filochard, 2021-05-22:
Hi
Unfortunately I have no oesenc chart
Nevertheless I'm gonna try the 20.08 version of opencpn ...

gromain, 2021-05-22:
@leamas I'll try that tonight or tomorrow! I'm on the boat now! I have oesenc charts and a USB dongle, so good whole chain testing!

filochard, 2021-05-22:
I installed the opencpn beta 20.08 and tried it
No more warning about an obsolete SDK

But I have lots of warnings about GTK : 
Gtk-Message: 16:19:32.433: Failed to load module "canberra-gtk-module"  
(in spite of the fact it's installed on my OS, perhaps missing in flatpak itself)
and every time I start a plugin 
(opencpn:2): Gtk-CRITICAL **: 16:19:50.348: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
or
(opencpn:2): Gtk-WARNING **: 16:23:44.729: Negative content width -33 (allocation 1, extents 17x17) while allocating gadget (node button, owner GtkButton)
or
(opencpn:2): Gtk-CRITICAL **: 16:23:44.729: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkRadioButton
or 
opencpn:2): Gtk-WARNING **: 16:23:44.638: Negative content width -9 (allocation 25, extents 17x17) while allocating gadget (node button, owner GtkButton)

I have not this kind of warnings with my installed linux version of opencpn when launching it from a console

leamas, 2021-05-22:
Still, I don't think these warnings are that relevant. When we have tracked them, they usually end up in wxWidgets. We might want to use a more recent version than current 3.0, but don't want to have more spinning wheels than we already have. (question is when wxWidgets 3.2.0 will be out, it should be available Real Soon (tm) since long.

leamas, 2021-05-22:
And the important stuff: Thanks for testing! Is there anything else besides these warnings?

transmitterdan, 2021-05-23:
@filochard these warnings are common on various platforms.  They don't affect the operation in any testing I have done.

As time goes on and various libraries get updated they should die down.  But as a practical matter, they seem to be unimportant.  Flatpak trades a stable library eco-system for a modern (ake bleeding edge) library eco-system.  It's a good trade.

leamas, 2021-05-24:
This has also been discussed and tested off-bug (is there such a word?) together with Dave. The conclusion is that that we should move to 20.08. This should resolve the  issue reported in this bug.

To not break things for existing users which uses 18.08 we will still be building 18.08 plugins for some time to allow for a smooth update period. All in all, w r t flatpak plugins needs to support:
  - flatpak-18.08/x86_64
  - flatpak 20.08/x86_64. 
  - flatpak-20.08/aarch64  (a. k. a. arm64)

As I see it, we should be able to close this bug (?)

gromain, 2021-05-24:
> ... and the x86_64 build against 20.08 is here: https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/oeSENC-4.2-flatpak-x86_64-20.08-tarball/versions/4.2.19.28/oeSENC-4.2.19.28_flatpak-20.08-x86_64.tar.gz
(from https://github.com/bdbcat/oesenc_pi/pull/116 )

No problem for me running this on my x86_64 machine, with the latest beta Flatpak build. I don't have the USB dongle with me now, I'll test it tomorrow with the dongle (and the charts). I'll also test on my raspberry running on an arm64 OS.

leamas, 2021-06-04:
@gromain: ping?

gromain, 2021-06-04:
Hi, sorry!
I just got back to the boat, with finally the right cable!

So I managed to install and run the latest OpenCPN beta flatpak build. However, I ran in troubles with the plugins.
Oesenc (from flatpak-aarch64/20.08
https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/oeSENC-4.2-flatpak-aarch64-20.08-tarball/versions/4.2.19.28/oeSENC-4.2.19.28_flatpak-20.08-aarch64.tar.gz ) is not running the server (it has the x86_64 oeserverd), and thus not detecting the USB. Seems like a Cmake problem rather than a build problem though.
And ocharts (from o-charts_pi/aarch64 https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/o-charts-0.9-flatpak-aarch64-20.08-tarball/versions/0.9.0.17/o-charts-0.9.0.17_flatpak-20.08-aarch64.tar.gz ), is complaining about no internet connection when trying to connect to my account (but there is an active internet connection). So I'm not sure where this comes from, but otherwise the plugin seems to be running.

Do you need the logs? I don't know where the flatpak logs are saved, but I can retrieve them if needed.

Also @bdbcat, what's the difference between oesenc plugin and ocharts plugin? Just a name change for the new version?

leamas, 2021-06-04:
No internet connection...Using another IP when you are on the boat?

I'll look into oesenc, time permitting. On my way to my own boat.

gromain, 2021-06-04:
No, that's weird. It's connected to my phone set as tethering. What's weird is that between the two plugins tests, nothing has changed regarding the internet connection setup, so there should be no further changes. I'll try some things to see what's up.

gromain, 2021-06-04:
Also, for ocharts plugin, oexserverd is for armhf, and is not running on my RPi 4 with a 64 bit OS.

bdbcat, 2021-06-05:
@gromain 
Considering o-charts_pi, OCPN flatpak Beta aarch64 ONLY:
Works fine for me here, with o-charts_pi Version 0.9.0.28.

1.  How did you start OCPN?
2.  Have you updated the o-charts_pi from the Beta catalog?
3. How exactly do you conclude that oexserverd is for armhf?

FYI, oeSENC_PI on flatpak is low priority right now.  Please focus on o-charts_pi.
Thanks
Dave

gromain, 2021-06-05:
Hi,
1. Installed with `flatpak install --user https://flathub.org/beta-repo/appstream/org.opencpn.OpenCPN.flatpakref` and launched with `flatpak run org.opencpn.OpenCPN//beta`
2. Just installed 0.9.0.28 from Plugin Manager, I used the link provided in https://github.com/bdbcat/oesenc_pi/pull/116 , I didn't know there was a newer version directly available through the manager.
3. `file .var/app/org.opencpn.OpenCPN/bin/oexserverd`

This latest command gives the right architecture with the plugin downloaded from the Beta catalog in the plugin manager.
However, I still have a 404 error when trying to connect. Nothing that looks relevant is shown in the logs, but I can paste them here if needed.

I realise oesenc_pi and o-charts_pi are separate, but I don't know the difference between them. As I now understands it, o-charts_pi is the new version.

bdbcat, 2021-06-07:
We are done here.
Please open a new issue for a specific flatpak plugin issue, if required.

closed 2021-06-07 by bdbcat