OpenCPN/OpenCPN #2277: Proposal:  Set AIS Class = AIS_GPSG_BUDDY for Followers

speters, 2021-05-19:
I wonder if it would be worthwile to set AIS target Class to AIS_GPSG_BUDDY for all ships in [g_MMSI_Props_Array with m_bFollower mark](https://github.com/OpenCPN/OpenCPN/blob/master/src/AIS_Decoder.cpp#L2518) to make those targets being [symbolized differently](https://github.com/OpenCPN/OpenCPN/blob/master/src/ais.cpp#L1099).

This suggestion came to my mind due to question in [segeln-forum.de](https://www.segeln-forum.de/board194-boot-technik/board35-elektrik-und-elektronik/board129-opencpn/80077-befreundete-schiffe-ais-optisch-erkennbar-machen/).

Hakansv, 2021-05-19:
The reason to call the MMSI properties "Follower" was to avoid confusion with the VHF DSC BUDDY which is another thing. The follower function is made to be able to suppress CPA alarms for a boat following the same track when e.g. cruising together. 
See more about DSC Buddy here: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais:dsc_buddy_via_ais#dsc_or_buddy_via_ais

So this may not be an issue or bug and would fit better in the rather new "Discussions"?

rgleason, 2021-05-19:
[Group DSC Primer for Fleets and Safety: Group Calls on your VHF and SSB](https://www.cruisingclub.org/communications/dsc-groups)

". A DSC VHF can query another boat’s VHF and automatically receive a reply with the second boat’s position.  Emergency VHF transmissions automatically have the position embedded.  This will enable newer chartplotters to automatically plot the position of other boat positions received by DSC."

This is different than GPS Buddy, but similar.  If the VHF/DSC radio is programmed properly, will the Fleet Positions be shown?   

(Also see the paragraphs about Groups and creating a Group MMSI.)

rgleason, 2021-05-22:
calls. That is off topic

rgleason, 2021-05-29:
Hakan link to the wiki docs shows that GasBuddy, dsc, dsc distress  all have different symbols. I believe dsc group functions will show location as dsc.

Follower is a separate designation.
Perhaps this should be closed.

rgleason, 2021-05-29:
Music "Follower" feature shuts down alerts.
Should "Follower" Icons be identified differently than other ais?

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:options_setting:ships&s[]=mmsi&s[]=follower#this_mmsi_is_my_follower_no_cpa_alert

rgleason, 2021-05-29:
Perhaps this should be moved to "Discussion"

speters, 2021-05-29:
> Should "Follower" Icons be identified differently than other ais?

IMO yes. Besides for the request above, it would make immediately clear that these vessels would not trigger a CPA alert. From a security standpoint, this which might also be the stronger argument for such a feature.

rgleason, 2021-06-20:
I think "followers" should have their own icon to avoid confusion and not be confused with "ais_buddies". This is a separate and distinct type.  Since the CPA is turned off for "followers"  it really must have a unique icon.

rgleason, 2021-06-20:
This should be closed now should it not?

bdbcat, 2021-06-20:
Anyone have a suggestion for an appropriate icon?

rgleason, 2021-07-04:
Please see for suggested icons for "Followers"  Takes GPS Buddy form but adds a broken line, a "V" for Followers
https://github.com/OpenCPN/OpenCPN/issues/2316#issuecomment-873575456

I think this can be closed now as it is resolved in the other Issue.

bdbcat, 2021-07-04:
I would leave it open until actually implemented in O55

commit db39e4e6 (2021-07-09): Symbolize AIS "follower" targets uniquely. Closes #2316 Closes #2277
closed 2021-07-09 by bdbcat