OpenCPN/OpenCPN #2278: Add Github topics to the repository

mendel5, 2021-05-20:
Hi, can you add the following topics to this repository?
* https://github.com/topics/chartplotter
* https://github.com/topics/plotting
* https://github.com/topics/navigation
* https://github.com/topics/sailing
* https://github.com/topics/boating
* https://github.com/topics/marine
* https://github.com/topics/ais
* https://github.com/topics/sdr

Thanks!

rgleason, 2021-05-29:
Do you mean as tags?

mendel5, 2021-05-29:
Yes, like this:
[Github Docs: Classifying your repository with topics](https://docs.github.com/en/github/administering-a-repository/managing-repository-settings/classifying-your-repository-with-topics)

rgleason, 2021-05-29:
I see several github users who base their projects on opencpn, have these topics .

Openplotter for example. I think it is a good idea perhaps bobcat will do it.

rgleason, 2021-05-29:
Signal-k as topic?

mendel5, 2022-04-03:
Hi @bdbcat , can you do this? It only takes a minute and helps other people to find this repository.

rgleason, 2022-04-03:
Dave, I added some topics to Climatology and to some of the other plugins. It is quite easy.
https://github.com/rgleason/climatology_pi

closed 2023-02-03 by nohal