OpenCPN/OpenCPN #2281: In flatpak version no way to choose which object to display on vectorial charts

filochard, 2021-05-22:
Hi
Seems to be some problems
If you compare the linux version : 
[image]

and the flatpak one : 
[image]
Everything is grayed

tai-fun, 2021-05-22:
Change the Display Category on the FlatPak version to All or User Standard.

filochard, 2021-05-22:
> Change the Display Category on the FlatPak version to All or User Standard.
 sorry
I don't understand what you mean 
Is there an option somewhere in the command line to run opencpn ?
Or is there a special way to install opencpn before running it ?

rgleason, 2021-05-23:
Look at the below the bottom of objects afficables.  Select "tout selectionnir" or user standard.

transmitterdan, 2021-05-23:
Click the chart options hamburger on the lower right side of the chart.  The select "Display Category" "All".  It is the only drop down list box at the bottom.

filochard, 2021-05-23:
> Look at the below the bottom of objects afficables. Select "tout selectionnir" or user standard.

The problem is that even the buttons "select all" "select nothing" and "standard" are greyed and not useable

filochard, 2021-05-23:
> Click the chart options hamburger on the lower right side of the chart. The select "Display Category" "All". It is the only drop down list box at the bottom.

Thanks ! that was the trick (unknown... and not default status on my linux version)
I close this issue : the problem was between the chair and the keyboard    ;-)
Sorry for the noise

closed 2021-05-23 by filochard