OpenCPN/OpenCPN #2282: [RFE] Udev rules strategy

leamas, 2021-05-23:
TL; DR We need to figure out how to handle udev rules on Linux.

Udev rules are only used on Linux. We use (should use) such rules for three purposes:

  1) To update permissions  on the serial device(s) used for NMEA input.
  2) To create stable device names which doesn't change on suspend-resume i. e., when the laptop lid is closed and opened again
  2) To update permissions on the chart dongle for users using this.

The chart dongle is a fixed rule, always the same. The rules handling the serial input device is **not** the same, it depends on the device user has installed. 

#### Debian

On Debian we the dongle rule is installed by default,  this is trivial. However, we cannot install a rule handling the input device since the device is unknown. This means that a OpenCPN user must use root permissions to install a specific rule for actual device(s) used.

#### Flatpak
On Flatpak, we cannot install any rule at all, Flatpak applications are unaware of the /etc directory. This means that the dongle rule must be installed  using root permissions outside of the OpenCPN application.

The situation for serial devices rule(s) is the same as for Debian.

#### Common denominator
We need to install  the fixed dongle rules. Although this is done by the OpenCPN package on Debian, we still need to do it manually on Flatpak.

In both Debian and Flatpak we need to adapt rules for the serial devices(s) used and install them using root  permissions.

#### What to do (ideas)

  - Document how to handle it, and link to the docs from the application (will bite off-line users). Will probably include evil  command line stuff.
  - Link to local docs in the application. Works off-line, docs are not updated.
  - Detect that permissions are wrong and hint users about missing rules and docs.
  - Create an evil  command  line python script which let users select device and then writes the necessary rule(s) and installs them (user approves root access).
  - Create a GUI, flatpaked application which does the same as script: user picks device from a drop-down list, application creates rule
  - ...  and what not

#### Summary 

We need figure out how to handle udev rules on Linux, once and for all.

leamas, 2021-05-28:
Perhaps a dialog as this when selecting an input serial device could make the trick. This relies on that it's possible to create a permissions rule based on usb vendorId and deviceId given a /dev device name. As I understand it, it should be possible.

The dialog is assumed to be shown on linux if user selects a device which lacks write access.

```
The device ttyUSB0 exists but cannot be used due to missing permissions.

This problem can be fixed by installing a udev rules file. I have created
such a file for you using ttyUSB0.

Once installed, the rule will create a new device called opencpn-usb0. It
is a new name for ttyUSB0. However, this name will always work while
ttyUSB0 might fail occasionally if you are using a laptop.

                                                                       ╌╌╌
Don't show this dialog again                                          ┆ ✓ ┆
                                                                       ╌╌╌
───────────────────────────────────────────────────────────────────────────

Manual install instructions ➤

Review rule ➤

───────────────────────────────────────────────────────────────────────────

                                                 ╭──────────────╮   ╭──────╮
                                                 │ Install rule │   │ Quit │
                                                 ╰──────────────╯   ╰──────╯

 ```

leamas, 2021-05-29:
A similar situation occurs  if the dongle can be detected but lacks permissions. This happens if the dongle rules file is missing. Basically, this is only about flatpak since the Debian packages installs the rules file in /lib.

We could handle this with a similar dialog at startup:

```
A OpenCPN dongle is detected but cannot be used due to missing permissions.

This problem can be fixed by installing a udev rules file. I have created
such a file for you using ttyUSB0.  Once installed, it will ensure that
the dongle permissions are OK.

                                                                       ╌╌╌
Don't show this dialog again                                          ┆ ✓ ┆
                                                                       ╌╌╌
───────────────────────────────────────────────────────────────────────────

Manual install instructions ➤

Review rule ➤

───────────────────────────────────────────────────────────────────────────

                                                 ╭──────────────╮   ╭──────╮
                                                 │ Install rule │   │ Quit │
                                                 ╰──────────────╯   ╰──────╯

```

bdbcat, 2021-06-07:
Alec...
Are you suggesting an OCPN actual dialog, or a scripted bash thing?

leamas, 2021-06-07:
Actually two "real" dialogs. One if the dongle exists but can't  be used due to read-only access.  Can only happen on Flatpak, but there is really no need to make a difference in the code.  This is a late start-up check.

The other occurs if user selects a device which is read-only. This can happen on both Debian and Flatpak. The check triggering the dialog should thus be done when user selects a device, typically an USB serial port. Debian users have other options here using group permissions, but t IMHO thisi just messes up things. Better to stick to a  solution which works everywhere.

I don't think a scripted solution (bash, python, ruby, whatever) can provide a reasonable UX  for this (?)

bdbcat, 2021-06-07:
OK, no worries.  Just more code bytes.  I'll take a look.

bdbcat, 2021-06-07:
"A similar situation occurs if the dongle can be detected but lacks permissions."
On flatpak, how can we detect the dongle if the udev rules are not installed?

leamas, 2021-06-07:
Using libusb. The general idea is to open the device and check the error code. In practise, it's more complicated: one needs to enumerate the devices and check if any of them is the dongle before trying to open it.

It's way easier using libudev. However, this is not available in flatpak.

I have some lines available in my fork at https://github.com/leamas/OpenCPN/tree/udev-dlg. Just some preliminary tests and design.

mgrouch, 2021-06-08:
Don’t build it into OpenCPN. 
It should be some standalone executable. With its own release cycle.

leamas, 2021-06-08:
why?

mgrouch, 2021-06-08:
Don’t repeat the same mistake as
putting plugin download and map download managers into one monolithic app.

nohal, 2021-06-08:
Seems like nobody but you considers these a mistake, so please refer to the earlier answers you already got and, please, do not start again.

nohal, 2021-06-08:
And same as any other bugs these will be fixed. The discussion about separating OpenCPN into a heap of single purpose executables is closed and it will not happen.

leamas, 2021-08-02:
Fixed in #2303. Thanks for input! 

Closing.

closed 2021-08-02 by leamas