OpenCPN/OpenCPN #2286: S-57/S-52 Withies and Perches for BCNLAT

bcn2, 2021-05-28:
The following thread at CF raises an issue with the portrayal of a specific type of lateral beacons in the German (and Dutch) wadden sea. Those are called Withy or Perch.
https://www.cruisersforum.com/forums/f134/pricken-and-oesenc-charts-250731.html

The chart with the examples is 49-RMAON5 (DE5NOAMR), Amrum Island
[image]

Finally these are beacons, object type BCNLAT, Q90 with attribute types
BCNSHP 1
CATLAM 1 or 2 (port/stbd)
Which are portrayed correctly by OCPN, as BCNSHP=1 does not show
differences for CATLAM.

However, if you look at the other objects there, we find
TOPMAR, with
TOPSHP 15 or 16
which we don't portrait or don't know how to portrait yet. Hence the "?"
[image]

I understand that a BCNLAT shall not have a topmark, but it seems that
the cartographers have chosen here the combination of two object types
instead of portraiying one object BCNLAT with BCNSHP "2".

[image]

[image]

Can we add that type of topmarks that seems not to be implemented right
now? Because asking the German BSH for clarification would not help that
much, I fear. Or we can do that in parallel.

The information is important for safety in the wadden sea.

Hubert

CarCode, 2021-05-28:
I don't know where this example chart 49-RMAON5 (DE5NOAMR) is from. My charts which I use are different however and when something has to be changed it should be changed to support any chart of this area. In the s57data folder of OpenCPN there are the correct symbols in the file rastersymbols.png (second and third symbol in line 5 of the PNG file):
[image]

[image]

But in my charts they are showing like this for port as well as for starbord:
[image]

bcn2, 2021-05-28:
DE5NOAMR is just the most recent official chart issued by the German BSH.
Please observe the objects and their attributes. BCNSHP = 1 is a pile without taking into account Port or Starboard.
BCNSHP = 2  would render withies and perches. But this is not how the cell is encoded. Instead it uses the additional TOPMRK attribute.

CarCode, 2021-05-28:
Seems the symbols from the OpenCPN file rastersymols.png are outdated. See line 92 in the attachment and the symbols in line 91 are correct. But anyway we should not follow a single institute but the s57 standard.
[image]

Hakansv, 2021-05-30:
Just tested the changes to o-charts:  👍 
[image]

bdbcat, 2021-06-07:
Corrections made to S52 PLIB in OCPN master.
Closing now.

closed 2021-06-07 by bdbcat