OpenCPN/OpenCPN #2289: PIM  Flatpak 18.08 and 20.08 support in Testplugin?

rgleason, 2021-05-29:
See Flatpak: Update runtime to 20.08 #116

Is this a test or should we be making changes to testplugin?
Are you supporting 18.08 or just 20.08 

Does not look simple to accomplish after reviewing the changes.

leamas, 2021-05-30:
The overall idea is to provide three flatpak builds: x86_64/runtime 18.08, x64_64/20.08 and aarch64/20.08. I will upstream changes in oesenc_pi to shipdriver to create a reference example and then file a bug againist testplugin.

But first I need to make my yacht in order so it can be launched on Monday.

rgleason, 2021-06-20:
Ship driver flatpak update with 3 builds as example.

https://github.com/Rasbats/shipdriver_pi/commit/f61acf00cde41dc6c1124319efbc898b42cfd6f4

rgleason, 2021-06-29:
Done by bdbcat in pypilot and jg in testplugin  and pushed to PIM

closed 2021-06-29 by rgleason