OpenCPN/OpenCPN #2292: Function 'Filter NMEA/Course and Speed data' not honored if input thru Signal K

LennartG-Sve, 2021-06-04:
If an NMEA 2000 GPS is connected and Signal K is used to read NMEA 2000 into OpenCPN the data will not honor the filter settings for NMEA/Course and Speed data. The course predictor will jump both in length and direction, not damped by the filter setting.

Hakansv, 2021-06-04:
Lennart
I'm not the expert on this but what I can find also SOG/COG from SignalK data stream is passing the filter function. (by PostProcessNMEA())
But the filter function is by-passed if "Calculate SOG and COG from position changes" is enabled in Options->Ships->Own ship. Pls check. I was falling into that trap myself.

LennartG-Sve, 2021-06-04:
Thanks Håkan
   can't remember I've ever touched that setting not even when using USB-GPS. I will check in a day or two at next visit at the boat. Assume the setting should be the same no matter using 0183 or 2000 thru SK.
Leaving this open until confirmed.

Hakansv, 2021-06-04:
It was the same for me. I've "never" touch that one but still once enabled. It would be used for dead calculation I suppose so never use it enabled when position data is coming in.

LennartG-Sve, 2021-06-05:
Was down to the boat this morning to verify settings, the setting "Calculate SOG and COG from position changes" is not enabled. 
Could it be related to the fact that I'm using a Garmin GPS24xd with an update rate of just above 10 hz and between 25 and 32 satellites received?

transmitterdan, 2021-06-05:
If you have the VDR plugin or can otherwise record the NMEA stream we can test here and report what might be happening.

LennartG-Sve, 2021-06-05:
transmitterdan, I do not have the VDR plugin installed nor do I have internet access from the boat computer. Can however generate a candump file and copy using a USB-stick. if that is acceptable.
If not, then I'll try to arrange something somehow getting the VDR installed.

Hakansv, 2021-06-05:
tdan..
Lennart is running via SignalK protocol so no NMEA for the VDR to catch.

I've not much experience of N2K GNSS data via SK but if you look at the SK "Data Browser" -> navigation.position & navigation.SpeedOverGround  & navigation.CourceOverGround if they are stable or fluctuating.

A 10 Hz position update rate seems overkill but would not be problem I suppose. But try 1 Hz if possible. 
Even if the GNSS receiver has detected a lot satellites no more than 12 are normally used for the position fix.

rgleason, 2021-06-05:
nmea2000 ---> signalK --->nmea0183 ---> OpenCPN ---> What GPS signal is going out or it that off?

LennartG-Sve, 2021-06-06:
Håkan,
   The Garmin GPS24xd is using 10 Hz per default and receives all available satellites, up to 32. I have found no configuration options nor can I generate any if I had as that would require tools I do hot have. We should also be capable of using the commercial receivers in their default configuration.

Rick,
   I think you are on to something here and I was actually thinking along the same lines. I'm generating NMEA 0183 RMC and GGA from SK, this for the GPS Odometer that does not read SK. Could be that those are 'colliding' with the SK signals although from the same origin. The GPS Odometer shows no speed fluctuations but is rock stable indicating that SK is working fine and the NMEA 0183 signals generated are correct. 

Further testing is required . . . .

Hakansv, 2021-06-06:
Lennart..
So, you've a mix of SignalK and TCP/UDP NMEA0183 messages for the same thing. For a test disconnect the SK connection and check for more stable position, COG and SOG.

LennartG-Sve, 2021-06-06:
Håkan, yes there is a duplicate of signals from the same source so the content should be the same. My thinking is as everything is time stamped and from the same source it should be ok. Perhaps not.
 So the test should be to disable the SK input totally and run everything thru NMEA0183 TCP/IP. Not sure that all required signals are available that way though, that remains to be seen. Perhaps tomorrow.
This could also be a problem for other plugins not supporting SK directly, Logbookkonni for one. Twocan should work ok as it generates NMEA 0183 directly but I've had other oddities there, probably Linux related.

transmitterdan, 2021-06-06:
There are no time stamps in NMEA data. OpenCPN processes messages as it sees them come in.

If the course and speed have multiple sources you will get "jumpy" performance. And it really taxes the graphic system.

If the Signal-K data drives the "boat" position on the display we can make minor changes to honor the filter on course and speed. But remember if the position updates are 10 times faster you may have to adjust the filter parameter by a factor of 10.

I will look at how the filter interacts with "own-ship" when i get some free time.

LennartG-Sve, 2021-06-07:
I have, after a day with extensive testing, found the golden rule used when not all used plugins can read Signal K natively.
First:
"If you need to generate ONE NMEA 0183 code from Signal K / NMEA 2000 then ALL used signals should be converted into NMEA 0183 format and the direct reading of Signal K data should be deactivated."
The above is a bit harsh but a signal that is available thru BOTH NMEA 0183 and Signal K may cause problems. Some signals may be ok but others may cause problems and I have found no way to select/limit which signals to convert directly. Better safe than sorry. I'm also aware that maybe not all wanted sentences can be generated as NMEA 0183.
Second:
If your GPS delivers more that one RMC sentence (eventually converted form NMEA 2000 into NMEA 0183) then multiply the setting 'Filter NMEA Course and Speed data' with the number of sentences per second. That is, of course, if you use that setting at all.

Maybe both of the above should go into the docs somewhere.

I will be closing this in a day or two as it is not a bug but a handling case and if the above suggestions are accepted.

rgleason, 2021-06-07:
Added here  https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:supplementary_software:signalk

Thanks

closed 2021-06-07 by rgleason

Hakansv, 2021-06-07:
Lennart..
There are no prioritization for any of position, COG or SOG in OCPN so if they are updated O just do so even if two or more sources are involved.
In plugin Dashboard we have such a prioritization but for position, COG and  SOG we honor what's coming from OCPN so in this case it wont help.
There is of course a possibility to add prioritization in OCPN now when we have two main sources; SignalK and the NMEA multiplexer. But that will be a discussion point for the future I suppose.

LennartG-Sve, 2021-06-07:
Håkan, I even removed the SK connection thru port 3000 and it appeared the same as deactivating the SK connection. So just use SK and only connect the NMEA 0183 port 127.0.0.1:10110. 
The signal path NMEA 2000 -> Signal K -> NMEA 0183 should not be able to cause any conflicts, OpenCPN will not even know SK is there as there is no port 3000 connection..

rgleason, 2021-06-07:
https://github.com/OpenCPN/OpenCPN/discussions/2297