OpenCPN/OpenCPN #2295: Modifying Grib_pi GUI with wxFormBuilder

ThomTrab, 2021-06-06:
Hello,

I'm new to OpenCPN and found the software great.

In my opinion, some GUI could benefit some ironing. I mean ensuring that all the boxes are aligned, text centered correctly, and so on. I'm not talking about GUI revamp. I already done this on the WeatherFax and WeatherRouting plugin using wxFormBuilder as mentionned in OpenCPN documentation.

I tried to do the same on the Grib plugin, but apparently some manual modification have been done to the `GribUIDialogBase.cpp`. These modification prevent me to use wxFormBuilder to make changes.

Is there a way to use wxFormBuilder to modify the GUI of the plugin ?

Thank you for your help

transmitterdan, 2021-06-06:
I would say no, wxFormBuilder was used at the beginning but it is slow, limited in features and seems to always be out of step with wxWidgets features. Long ago the dependency on wxFormBuilder was abandoned for the OpenCPN core program and plug-ins.

I would suggest to you that wxFormBuilder isn't that useful in the real world. For a multi-platform program like OpenCPN you often have to compromise on things like perfect dialog item alignment. OpenCPN runs on tiny displays and very large displays. Also, multiple desktop environments like Mac, Linux (10+ flavors), Windows and Android make regression testing a full time job. The time spent adjusting dialog boxes is tiny compared to the time spent testing those modifications on 15+ different environments.

nohal, 2021-06-06:
No, there is not. Either you have to mock your changes up in wxFormBuilder and then modify the code or refactor the whole thing on top of a new source generated from wxFormbuilder.

rgleason, 2021-06-06:
Move "Discussions"  under  "Show and Tell?"

...or close it?

ThomTrab, 2021-06-07:
Thank you all for your input. I may try to refactor the existing code to make it easier to edit manually and then propose GUI improvement.

closed 2021-06-07 by ThomTrab

rgleason, 2021-06-07:
See  https://github.com/OpenCPN/OpenCPN/discussions/2299