OpenCPN/OpenCPN #2296: Trouble with Flatpak builds

rgleason, 2021-06-07:
```
+ flatpak install --user -y http://opencpn.duckdns.org/opencpn/opencpn.flatpakref
GLib-GIO-Message: 00:52:34.094: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
error: Can't load uri http://opencpn.duckdns.org/opencpn/opencpn.flatpakref: Error resolving ?opencpn.duckdns.org?: Name or service not known

Exited with code exit status 1

CircleCI received exit code 1

```

any suggestions?

bdbcat, 2021-06-07:
Temporary glitch in Alec's server, I think.
The referenced URL works right now, so resubmit the job.
http://opencpn.duckdns.org/opencpn/opencpn.flatpakref

leamas, 2021-06-07:
More on this: In general, there should be no need to build against my server in production builds. My server basically serves the devel branch i.e., the last snapshot. In general, production plugins should be built against the flatpak version available in flathub, this is users actually use.

The shipddriver setup uses this for all flatpak builds.

Of course, flathub is much more stable then my private server..

rgleason, 2021-06-07:
Thanks. I've looked over shipdriver and testplugin config.yml and ci files. They are somewhat different and I can't identify where duckdns.org...  is called.   

Jon, in Austrailia, has had his access to circleci curtailed by the US Treasury directive to Circleci to protect assets. Hopefully this will get resolved, but right now he can't use circleci. He is able to use VM on his local servers, but he will have difficulty resolving this if the situation persists.

I am going to try rerunning it.

rgleason, 2021-06-07:
Closing, covered in plugins/ #295

closed 2021-06-07 by rgleason

leamas, 2021-06-07:
>  I can't identify where duckdns.org... is called.

It's not used. The flatpak plugins are built against the version available in flathub i. e., the same version as users use.

rgleason, 2021-06-07:
Thanks. Understood.    Somewhere it is being used.

leamas, 2021-06-07:
> . Somewhere it is being used.

Not in shipdriver.