OpenCPN/OpenCPN #2302: [RFE] Add 'install all updates' and 'install all new' buttons in plugin download manager

mgrouch, 2021-06-11:
Add 'install all updates' and 'install all new' buttons in plugin download manager

It takes too many clicks for a user to do it one by one.

Thanks,
--MG

transmitterdan, 2021-06-12:
There is such a button.  Go to Options->Plugins->ChartDownloader and click the Preferences button.  Check all the boxes found there and click close.  Close OpenCPN and reopen it.  Then a button will appear in the chart downloader that downloads and installs all new and updated charts for all catalogs.

[image]

mgrouch, 2021-06-12:
This request is for plugin downloader. Not chart downloader thought. 

Thanks

pgrawehr, 2021-08-28:
"Update all installed" or "Update all active" would definitely make sense IMHO.

rgleason, 2021-08-28:
Those are  reasonable subsets to use, I think, however the interface is quite involved, so it should be very short.
Update installed  or Update active.