OpenCPN/OpenCPN #2305: flatpak issue when the host system graphic driver is updated !

filochard, 2021-06-13:
I have just encountered a problem with my flatpak version of opencpn
It's not related to OpenCpn itself but that's something that opencpn users must know !!!

I use Mageia as Linux version with proprietary nvidia drivers
Like all other linux system, Mageia offers updates for the basis system and the installed programs

Since January my nvidia driver was the **460.39** release and I had installed opencpn with flatpak it worked well (flatpak had installed its own staff for this nvidia version)

Mageia offered a new version of the nvidia driver 460.84 
After having updated this driver  (2021 june 10th) I couldn't launch opencpn anymore :  GL errors !!

I have had to dig a little to understand where was the problem

Here is the solution : 
1)
I needed to uninstall opencpn, 
running : "flatpak uninstall flathub org.opencpn.OpenCPN"
then uninstall all flatpak own addings (nvidia driver particularly)
running : "flatpak uninstall --all"
2)
Then reinstall opencpn
running : "flatpak install flathub org.opencpn.OpenCPN"
so that **flatpak itself installs its own nvidia driver 460.84**  before installing opencpn to work on my system

Conclusion
Flatpak is not such a sandbox it pretends to be !
It may fail to work if you simply update some of your system own drivers...

Fortunately, all the previous settings of the flatpak opencpn had been preserved and not wiped by the uninstallation

NB my Mageia installed  version of opencpn was not impacted by the update of the nvidia driver

rgleason, 2021-06-13:
Is this something that should be in the User Manual under a new Flatpak Installation page?
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak

@Filochard  If so, could you advise if the above is adequate or write a paragraph.
If you want to become an editor of the wiki, let me know your email somehow. 

Thanks.

filochard, 2021-06-13:
Hi
I didn't try to run this command : 
$ flatpak update org.opencpn.OpenCPN

Since I have uninstalled and reinstalled everything after having updated my nvidia driver I get this list with  "flatpak list"

Name                        Application ID                                   Version                Branch        Installation
Freedesktop Platform        org.freedesktop.Platform                         18.08.39               18.08         system
nvidia-460-84               org.freedesktop.Platform.GL.nvidia-460-84                               1.4           system
html5-codecs                org.freedesktop.Platform.html5-codecs                                   18.08         system
OpenCPN                     org.opencpn.OpenCPN                              5.2.4-0+6b314e6        stable        system

Previously the second line was this : 
nvidia-460-39               org.freedesktop.Platform.GL.nvidia-460-39                               1.4           system

I don't know if "$ flatpak update org.opencpn.OpenCPN"    may work or if it only updates the application and not the flatpak stuff ...

It would be someway difficult for me to downgrade the release of my nvidia driver to 460-39, 
so that I could uninstall all the flatpak stuff, then reinstall it with this nvidia release, then update the nvidia driver, and then try this command ($ flatpak update org.opencpn.OpenCPN)  to test if it works !

Perhaps a question to ask to flatpak developers 

In the Wiki you may just add a warning about the fact that updating the graphic driver of the host system may prevent opencpn to work... and that it needs to uninstall opencpn and all the flatpak stuff and reinstall opencpn (the needed updated flatpak stuff is then installed)

rgleason, 2021-06-13:
Thanks filochard.  I understand about the driver. I'll ask Leamas.

leamas, 2021-06-14:
> I didn't try to run this command :

$ flatpak update org.opencpn.OpenCPN

I don't really see the issue here. Seems that flatpak has all the necessary drivers available, so it should just be a matter of running the `flatpak update` command above.  That you might need to update the flatpak environment after operating system or drivers updates is IMHO logical.

Anyway, this bug is not really an "actionable item"  and should thus IMHO be closed.

rgleason, 2021-06-14:
OK so need to add anything to the wiki.

closed 2021-06-14 by rgleason