OpenCPN/OpenCPN #2307: Plugin Manager too sensitive to environments

jongough, 2021-06-14:
Plugin manager now requires all plugins to be built for every possible supported environment. Currently there are 14 different builds that need to be done to be compatible. This is soon to increase by at least 2 for various flavours of flatpak. Prior to Plugin Manager a plugin built for Ubuntu 16.04 would run quit happily on 18.04, and one built for 20.10 would run quite happily on 20.04, this is no longer the case. Here a build of the plugin for 20.1 gets rejected when being installed on 20.04:
```
07:17:25.689   DEBUG PluginHandler.cpp:198 Host: init: abi: ubuntu-gtk3-x86_64, abi_version: 20.04, major ver: 20
07:17:25.689   DEBUG PluginHandler.cpp:173 Plugin: setting up, name:  OCPN Draw 
07:17:25.689   DEBUG PluginHandler.cpp:174 Plugin: init: abi: linuxmint-gtk3-x86_64, abi_version: 20.1, major ver: 20
07:17:25.689   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED:  OCPN Draw 
```
This is making a maintenance nightmare. Each time there is a new version of one of the supported platforms all the plugins need to be updated and rebuilt to ensure there is an installable version available.

By changing the generated xml produced by 'make build' to have a different 'target-version' the plugin installs and runs quite happily. Both OCPN and OD were built on the same system, but OCPN seemed to think it was on 20.04 so rejected the plugin.

bdbcat, 2021-06-16:
"OCPN seemed to think it was on 20.04"
If OCPN thinks it is "ubuntu-gtk3-x86_64, abi_version: 20.04", and this is not actually the case, then that is a root problem.
What exactly was OCPN running on, and what was it built on?

jongough, 2021-06-16:
Just rebuilt OCPN on Linux Mint 20.10, which is based on Ubuntu 20.10:
```
LSB Version:    core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Linuxmint
Description:    Linux Mint 20.1
Release:        20.1
Codename:       ulyssa
```
but it reports:
```
PluginHandler.cpp:198 Host: init: abi: ubuntu-gtk3-x86_64, abi_version: 20.04, major ver: 20
```
I don't know where it is getting that version from as it is returned from CompatOs from a private string
OCPN cmake returns:
```
-- Building for target (Platform; version; arch): linuxmint-x86_64; 20.1; x86_64
```
To get around the issue for plugins getting the wrong, i.e. linuxmin, platform I build an initial install on a native Ubuntu system, install that then copy over the lib for debugging purposes.

leamas, 2021-06-16:
hm... If you go into Options | Plugins | Settings | Compatibility and select "Default  Setting" from the drop-down,  does this change anything?

jongough, 2021-06-16:
Select Default and it gives `linuxmint-x86_64:20.1`. Try install build on linux mint of plugin fails as it has gtk3 in the abi:
```
16:31:08.618   DEBUG PluginHandler.cpp:198 Host: init: abi: linuxmint-x86_64, abi_version: 20.1, major ver: 20
16:31:08.618   DEBUG PluginHandler.cpp:173 Plugin: setting up, name:  OCPN Draw 
16:31:08.618   DEBUG PluginHandler.cpp:174 Plugin: init: abi: linuxmint-gtk3-x86_64, abi_version: 20.1, major ver: 20
16:31:08.618   DEBUG PluginHandler.cpp:343 Plugin compatibility check Final: REJECTED:  OCPN Draw 
```
The abi of OCPN is wrong as it should have had 'gtk3' in the name as the plugin did. As a test I did:
```
mkdir build-test
cd build-test
cmake ..
make -j 
copy 'uidata' from and install of OCPN
./opencpn -p
stop OCPN
Add CatalogExpert=1 to conf file
./opencpn -p
```
This gave:
Active setting: linuxmint-x86_64:20.1
in the compatability line of the Plugin Catalog Settings.
Looking in the list of flavours there is not Ubuntu 20.1 available, and certainly there is no Linux Mint available.

This is built from git master commit: `65983a64`

leamas, 2021-06-16:
OK: the plugin is linuxmint-gtk3-x86_64, the host is linuxmint-x86_64 and it doesn't match. It could be fixed either way.

To me, the easiest and most straight-forward would be to rebuild the plugins. The shipdriver templates uses the OCPN_TARGET_TUPLE for this purpose; don't know if testplugin has similar provisions. 

Another option would be to use the  OCPN_TARGET_TUPLE  to rebuild main OpenCPN using the linuxmint-gtk3 target OS. 

Using the -gtk3 suffix has the  drawback of keeping a suffix which actually is useless on all versions besides Bionic, the only one where both gtk3 and gtk2 is available. That it is used also on focal is a late mistake which we have to live with.

leamas, 2021-06-16:
> Looking in the list of flavours there is not Ubuntu 20.1 available, and certainly there is no Linux Mint available.

That list is just a digest of the available plugins in the catalog. There are no Mint plugins in the catalog, and thus it's not found in the drop-down list

jongough, 2021-06-16:
But plugins still have to be built for each version of Ubuntu even though they will quite happily run. With the plugins using statically linked libraries the only real issue is the version of wxWidgets installed and gtk2/3 (which you say is only an issue on Bionic). Therefore why be pedantic on the version of Ubuntu?

Linux mint causes me issues with the way that compatibility is worked out. But that is not the thrust of this issue, it is the sheer number of builds that have to be done to satisfy Plugin Manager. Basically there should only be Ubuntu/Debian builds for the different versions of wxWidgets and possibly gtk2/3, there should not be builds for each version of Ubuntu/Debian.

leamas, 2021-06-16:
> But plugins still have to be built for each version of Ubuntu even though they will quite happily run..

Isn't this the same discussion as in #2213?

jongough, 2021-06-16:
No. This is about the need to build for every environment due to the sensitivity of the check. It is nothing to do with mint. I should be able to install a single version of the plugin on Ubuntu 20.04 & 20.10 and probably on 18.04 & 18.10, there should not be a need for 4 different builds.

leamas, 2021-06-16:
There is perhaps a  question how the decision-making process works here (?)

As I see it, this very issue was decided by Dave in this [#2213 comment](https://github.com/OpenCPN/OpenCPN/issues/2213#issuecomment-825341743). Here, he clearly states that we should *not* try to use a  plugin built for for example 19.04 host on 20.04. 

I don't always agree with Dave, for sure. But I respect his right to make the final decisions, and in this particular case I also agree with his reasoning. I could expand on this, but leaving it for now.

But the core problem  is simple: too many builds. I agree with you here. It's noteworthy that this is a Linux only problem. We could cope with it in a number of ways:

  - Using flatpak. We are now exploring it to minimize the number of aarch64 packages. Since this a new platform for us, we might get away with only providing flatpak rather than all sorts of OS:es which are used here.
  - Drop old platforms. Just because the main program builds old outdated stuff like Trusty or Xenial we don't have to support this in plugins. Shipdriver has already dropped Trusty, and I will propose that we drop also unsupported Xenial  before the 5.6.0 release. 
  - Likewise, just because main OpenCPN will support flatpak/18.08 the plugins still don't have to support it. 

Of course, it's important to keep the current trusty/xenial/flatpak-18.08 available in the catalog for yet some time. But that doesn't mean we need to release new plugins on these unsupported platforms.  The download figures provided by @rgleason in https://github.com/OpenCPN/plugins/issues/340 are extremely useful  hints here

rgleason, 2021-06-16:
There is opportunity for changing/improving things as we go forward and the situation changes. Perhaps David will consider simplifying adjusting the sensitivity of the plugin check to just the main version number?

Alec's suggestions along with Jon's would simplify the plugin load for sure, and reduce compute time and storage.

leamas, 2021-06-16:
> Flatpak offers help, if it can work

It's not that flatpak doesn't work, it certainly does. However, it's more picky and reveals existing bugs in some plugins. We need to fix those anyway, flatpak or not. Doing so, we should send a grateful thought to Flatpak which helps us finding these bugs.

bdbcat, 2021-06-16:
Let us dig in to the decision logic a bit.  As example, consider ubuntu.
1.  OCPN 5.6.0 will be released only for LTS versions of Ubuntu.  in Ubuntu's words:
`  "These releases focus on performance enhancement and stability. The LTS is what we recommend to large scale enterprises, general users and businesses."`
2.  Thus, if a user wants to run OCPN on _ANY_  non-LTS (interim) version of ubuntu, they will either a) Compile it themselves, or b) Get it from some third party repo, where it has been built and uploaded, or c) Get it from the supported repo by "fooling" their local apt repo specifications.

So, the implications for Ubuntu plugins are simple.  We plan to formally build and publish Ubuntu plugins for LTS versions only.

Considering other linux repos:
OCPN is not formally supported on other linux repos.  Our "official" position is Flatpak for those repos.
But of course, enthusiastic developers will find that OCPN can be built and run natively on many linux distros.
For these cases, our position is: 
A developer capable and willing to build OCPN for local use should be able to follow some simple rules to allow experimental use of non-supported plugins locally.  If an environment simple cannot be made to work using the available tweak factors built in to the OCPN/plugin build systems, that we should fix that.

Considering the number of plug builds:
1.  We are not (currently) paying cash money for CI builds.  This could change....
2. It is not as though devs have to build each plugin by executing some local commands, n-times..  The CI process, when fully populated, does this for us.
So, I do not see the need to especially "economize" on the plugin build matrix.

I'll stop here.

rgleason, 2021-06-18:
Keep in mind that maintaining the front-end for so many builds has developer time costs and that tweaking of plugins is also often required to get the build to work. It is not automatic

leamas, 2021-06-18:
I have been a bit formal here, talking about Dave's right to make the final decisions when it comes to core OpenCPN. 

However, plugins are something different. They are separate projects, and the one is charge is the plugin developer. Which means that if for example the frontend2 community decides to drop or include builds it's their decision.

Of course, we could still discuss these issues. I actually agree with Rick here: builds come to a cost, especially old not maintained stuff. We should balance these costs against  the user base. With respect to  this balance  plugins should probably be able to drop Ubuntu versions which are EOL, currently Xenial and Trusty.  Debian Stretch is  EOL  July,7 -- we should be able to drop also this when we start to build 11/Bullseye.

Furthermore, it possible to investigate  the download statistics from Cloudsmith before making decisions on dropping builds.

bdbcat, 2021-06-18:
"the one is charge is the plugin developer"
I agree with @leamas here.
The plugin build toolset and catalog matrix selection is entirely the responsibility of the plugin author/packager.  As are any resulting complaints from users that "XXX_pi is not in the catalog for my favorite linux distro".

But, we are a little off track here.
Back to the OP point regarding plugin compatibility policy:
OCPN core does what it does in order to support a large number of possibly inexperienced users.  Key targets here are stability, convenience, and verified compatibility. Any speed bumps that this policy causes to plugin developers should be considered as simply another part of a complex cross-platform development process.

leamas, 2021-06-18:
>  As are any resulting complaints from users that "XXX_pi is not in the catalog for my favorite linux distro".

Since we these days actually knows to what extent a build is used (cloudsmith download statistics...) there should not really be much such surprises if/when dropping an old build like trusty or xenial.

bdbcat, 2021-06-18:
True.  I personally like to keep old legacy builds in the mix since they tend to weed  out subtle compiler incompatibilities and the like.  It is a constant annoyance to aim at "lowest common denominator", and it does preclude use of some new language features.  But I think that this is important for wide cross platform build environments.  To be clear, this is just my opinion when I have my plugin developer hat on.

The more interesting questions for OCPN core will be:
Q: "Where is OCPN for Ubuntu 20.10?"
A:  Adjust your repo sources, and you will (probably) be able to install OCPN as built for Ubuntu 20.04."
Q: "Will the plugins work if I do that?"
A: Yep, because the plugins were built for 20.04.

..and...
Q: But what if I want OCPN on linux Mint-xx.
A:  Two choices:  a) Use Flatpak, or b) build from source (including any plugins you critically need).

leamas, 2021-06-18:
Agreed. 

> Q: "Will the plugins work if I do that?"
A: Yep, because the plugins were built for 20.04.

That is **probably**  true. OTOH, Ubuntu does not offer any ABI guarantee between different releases. A typical plugin might link to  20-30 libraries, and if any of them changes things will break. Which means that the only thing which is actually supported on Ubuntu 20.10 is Flatpak, same as for Mint. 

Or?

jongough, 2021-06-18:
> Let us dig in to the decision logic a bit. As example, consider ubuntu.
>
>     1. OCPN 5.6.0 will be released only for LTS versions of Ubuntu.  in Ubuntu's words:
>        `  "These releases focus on [...]
<Snip> 
This is an invalid assumption about Ubuntu and LTS. Yes the LTS is the 'safe' way to run, however, many people do not use the LTS as they need/want the newer versions to run programs. If you are using a distribution based on Ubuntu you can end up with different names and numbers even though you are on an LTS of the distribution. Distributions often offer to upgrade you to the next version, i.e. from 20.04 to 20.10, to gain 'extras'. This is normally done through the gui and a simple question related to updating the system. There is no need to know anything about the workings or internals.

So, for 'minor' upgrades, i.e. 20.04 to 20.10, OCPN should not require a new version of the plugin. In fact if you have the plugin installed prior to the upgrade it will continue to work, but if you reinstall it it will fail. This is not a good outcome.

Looking at it further, with the current process all the required libs are statically linked into the plugin, so there should be no incompatibility there. The only two changes that will cause an issue are a version change of wxWidgets, i.e. the plugin uses 3.1.2 and OCPN uses 3.0.2, and the move from gtk2 to gtk3. Currently we build gtk2 & gtk3 versions of the plugin and for focal have both available depending on what is currently installed.

As for flatpak, appimage, snap, etc. they are all big and slow. I avoid them where possible as they have caused issues for me in the past (as in a few weeks ago with some apps I use) and I prefer to run real applications built for the platform. On limited machines, i.e. those on the boat, I need the fastest smallest version of the app I can get.

Are there any known issues where a minor upgrade (or even a major one, i.e. 18.04 to 20.04) have caused OCPN or a plugin to suddenly crash or cause problems?

We have replaced an installation method that insured plugins would not be installed on incompatible platforms and could auto-install the required additions if needed. We have also changed the way we link the plugins so that all the libs needed should be linked statically into the plugin so there should be no 'external' libs required. This does require that the base level stuff, i.e. wxWidgets/gtk*, is installed. but that has always been the case.

We need to stop thinking as developers and what we can do to make stuff work and think as users and how to get stuff so it 'just works' and doesn't need to be continually reinstalled when an automated update occurs (Ubuntu can do do automatic updates even on LTS and Server installs).

The ABI is one method of determining things. But I know from OD that the ABI generated by the GCC compiler can 'change' for 'internal' reasons which are not related to anything that is being talked about here, and is non-developer controllable,  but makes plugins incompatible with OCPN.

We are lucky this is currently restricted to Linux. Windows 10 has many patches that are incompatible between themselves, 1709, 1909, etc. as they impact the .Net framework. However we don't bother checking these and 'assume' Win 10 is Win 10 when it is not.

nohal, 2021-06-18:
There is absolutely no warranty that a binary for Ubuntu LTS will work on the next non-LTS release, it is pure luck that it usually does. Actually the non-LTS after a LTS is the point where the ABIs may and do change and things tend to break. Luckily that does not happen too often.

Of course there are known examples where everything we use broke down completely on Ubuntu (and Debian, but there it is not a minor version bump) with a "minor" version update - it was around the migration to GCC5 and newer (where the new C++11 ABI was made default) and GTK3, you sure remember the total mess with several versions of all the GUI libraries - the old ones, the ones with the "v5" suffix and the incomplee set of the ones built against GTK3, with wxWidgets being a nightmare as it lacks competent maintainers in pretty much all the distributions.

The only solution I will admit is something like able to work "reliably" is to implement some sort of compatibility matrix that will be part of the plugin system and will say "If the system reports X and we do not directly have a build for it, it may use plugins built for Y we do have builds for (A somewhat tested to be compatible Ubuntu LTS for example)" - one data file, easy to contribute to, saves work to plugin authors, adds happyness to the users, is not too much of a time consumer for the core team.
Everywhere else, investment to flatpak will pay off longer term as far as I'm concerned.

bdbcat, 2021-06-19:
Jon...
Sorry, still do not get:
"So, for 'minor' upgrades, i.e. 20.04 to 20.10, OCPN should not require a new version of the plugin. In fact if you have the plugin installed prior to the upgrade it will continue to work, but if you reinstall it it will fail...."
Q: If you "reinstall it".  What is "_it_"?  OCPN, or a particular plugin?

leamas, 2021-06-19:
> So, for 'minor' upgrades, i.e. 20.04 to 20.10, OCPN should not require a new version of the plugin. In fact if you have the plugin installed prior to the upgrade it will continue to work, but if you reinstall it it will fail. This is not a good outcome.

This is just not how Linux distros work. They are developed around two basic principles (this includes Ubuntu)
  - There are no ABI or even API compatibility promises between different releases, see for example [Mark Shuttleworths explanation](https://wiki.ubuntu.com/MarkShuttleworth)
  - All software in a distro is re-compiled when a new version is released.
  
Which means that supporting a new version, be it core OpenCPN or plugins, is about recompiling them for the new version. Again, it *might* work without recompiling, but there are no guarantees.

leamas, 2021-06-19:
> The only solution I will admit is something like able to work "reliably" is to implement some sort of compatibility matrix that will be part of the plugin system and will say "If the system reports [...]

Indeed, and this has been discussed. In order to be useful, such a file must be possible to update after the core opencpn release (since we don't know about downstream releases when we make our). The simplest woulld probably be to add this data to the catalog so we can re-use the existing infrastructure to update and distribute the data.

leamas, 2021-06-19:
> As for flatpak, appimage, snap, etc. they are all big and slow.

I see no data anywhere that flatpak applications in general or specifically opencpn are slow. Have you any links supporting your claims?

As for size, flatpak trades compatibility for download and disk size. I think it's a good trade. That said, with the runtime in place the flatpak'ed opencpn application and the plugins are basically the same size as their native counterparts.

leamas, 2021-06-19:
> The only solution I will admit is something like able to work "reliably" is to implement some sort of compatibility matrix

The alternative I can think is to re-package OpenCPN. To rebuild the OpenCPN package in a new Ubuntu version or a downstream distro like Mint is actually simple. It could add a very simple patch which makes it accept plugin for some Ubuntu version which is known to work.  This would be to adapt to the general linux workflow

For Ubuntu minor versions like 20.10 it would just be about setting up a PPA and clone Dave's LTS sources, add a patch, submit and watch the build farms build for all sorts of platforms.

For Mint, it would be about approaching the Mint community. It would probably be easier to use the official package here since this by definition already is reviewed.  If one can handle the relations there, it would just also be about adding a simple patch.

Given how simple and stable these solutions are, it's IMHO probably the right way for someone interested in supporting minor Ubuntu versions or for example Mint without using Flatpak.

bdbcat, 2021-06-19:
Jon...
I cannot reproduce your reported problem.
1.  Provision a bare metal ubuntu 20.04 virtual.
2. Add launchpad OCPN beta PPA (for Ubuntu 20.04).
3. install OpenCPN 5.5 from launchpad PPA.  Starts and runs fine.
4. Update master plugin catalog.
5. Install ODraw plugin (1.8.5.2).  No problem.  Runs fine, no problem.
6. Force an upgrade of Ubuntu from 20.04 to 20.10 (Gorilla...)  Takes about 2 hours on slow net.
7. After final reboot into 20.10, the existing OCPN version had not been replaced or touched.  And it still runs fine.  As do the previously installed plugins.
8. Clear the plugin cache directory, uninstall ODraw plugin.
9. Update the master catalog, install the offered ODraw plugin. (1.8.5.2)
10. And it works fine.

Conclusions:
1.  Despite the distinct possibility that OCPN will be broken on minor upgrades (for reasons stated above), in this particular case we can say that OCPN as built for ubuntu 20.04 is compatible with ubuntu 20.10.
2. For this case, the OCPN  5.5 (i.e. github master) plugin compatibility policy is working exactly as designed and intended.

Finally, if one can demonstrate a practical and functional flaw in the current logic, please do so.

leamas, 2021-06-19:
> install OpenCPN 5.5 from launchpad PPA. Starts and runs fine.

> Force an upgrade of Ubuntu from 20.04 to 20.10 (Gorilla...) T

These things works differently if/when users are using  the official package. Here, we have different versions for 21.04 and 21.10, and upgrading the whole system  to 21.10 will  also update opencpn to a recompiled version. 

Here is to two cases (?)
  - The upgrade is done without any ABI break in dependencies. In this case, the PPA LTS version will still work, as will plugins compiled for LTS. The official package might  reject the plugins as not compatible since the versions mismatch.
  - There is some ABI break somewhere. This means that the upgrade is not possible without removing the PPA opencpn since the upgrade will lead to unresolved references. OTOH, the official, re-compiled package will be installable but the LTS plugins will not work.

Users of the official package might have a better situation of they pin the opencpn package to the LTS version. This would basically make it work as the PPA package.  

Just some preliminary thoughts, heading off to fill  up my boat with food...

bdbcat, 2021-06-19:
"Users of the official package might have a better situation of they pin the opencpn package to the LTS version. This would basically make it work as the PPA package."
That would seem best.  We know that the ubuntu "official" package is always at least one, and maybe two versions behind our release schedule.  Today, I think that the official package for 20.04 is OCPN 5.0.0.  Pretty old, and not what we wish users were installing.

leamas, 2021-06-19:
hm, indeed. We need to handle the backports stuff so there is a more recent version available in the official repos.  I'll start a new discussion on this later, there is not much point before the 5.6.0 release.

hreuver0183, 2021-07-04:
I have one comment on an option I would like to see in the 5.6.x version:
a tarfile plugin with less strict checks than currently available (if only with some "expert" option in opencpn.conf). 
The current tarfile import option is only useable when you have a working cloudsmith repo for managed plugins.

Currently support for deb-packages seems to have been dropped, which practically means there is no leeway in to get a plugin working outside the regulated managed plugins. Another advantage would be that a tarfile might work on specific distribution as well as Flatpak.

jongough, 2021-07-04:
Plugins built with Frontend2 generate "deb", "exe", "pkg" files for Linux, Windows and MacOS and these files are placed into the same cloudsmith repository the tar files are placed. But due to Plugin Manager only working with tar files they do not show up. This support will, however, be dropped over time and most users do not know how to access them and Plugin Manager has precedence so if Plugin Manager plugins are installed package manager packages will not show up.

leamas, 2021-07-05:
@hreuver0183:

> The current tarfile import option is only useable when you have a working cloudsmith repo for managed plugins.

Why? I regularly build plugins and import them without any cloudsmith repo involved. What do I miss?

rgleason, 2021-07-05:
Build locally and Options > Plugins > Import Plugin   then select your tarball from the build directory.

hreuver0183, 2021-07-05:
@jongough 
Very recognizable, if I try to insert a plugin in OpenCPN 5.5.x the deb is not found by OpenCPN when I have other plugins installed via the pluginmanager.
It was to be expected that the deb-files would be dropped sooner or later if I am the only one who uses them.

@leamas 
When I build a plugin as tarball and I try to insert it in 5.2.4 I get "Incompatible import plugin detected".
I see that behaviour changed in 5.5.x. Sorry I did not read all comments last winter. I must have missed it since I haven't been using the tarball option since 5.2.4.

With the changed behaviour of the imported tarballs the dropped deb-files should not hurt as much.
Sorry for the out-of-place comment.

hreuver0183, 2021-07-05:
@leamas Could you point me to one package that produces a wrokable plugin for flatpak 20.08?
The only package I see which produces flatpak is oesenc_pi and that is not likely to be update with flatpak 20.08.

jongough, 2021-07-05:
You could try ocpn_draw in beta, it has all the flatpak builds including 20.08.

Hakansv, 2021-07-06:
All those Linux variants---Phuu. Would I care to mention a in this matter simple system? :)
Jokes(?) away.. Would this post be connected to this issue?
https://www.cruisersforum.com/forums/f134/radar-plugin-5-2-1-released-250046.html#post3438504
Thanks for any support.
Håkan

leamas, 2021-07-06:
@Hakansv:  The issue here is a design issue. OTOH, the CF issue seems like a plain bug to me. As always, what we need in such cases is a logfile, preferably after starting opencpn with the `--loglevel debug` option. 

I'm sailing, and only occasionally on the net. But if you could guide your user to produce such a logfile, I will try to take a look when I have a possiblity.

CarCode, 2021-08-31:
Reading all this above I wonder
- a) is this a general problem of Linux versions and how do handle this problem other software distributors for Linux?
- b) is this a problem with OpenCPN only using Linux?
My conclusion:
- if a) never use Linux
- if b) too many cooks have spoiled OpenCPN at last for Linux

leamas, 2021-08-31:
@CarCode : Given the issue discussed here, in what way is your  comment related to it?

transmitterdan, 2021-08-31:
Someone who refuses to cook complaining about the chef is so 2021.  And tiresome.

matthijskooijman, 2022-05-20:
I originally wrote most of this comment to respond to #2694, but it seems that most of it seems more on-topic in this issue, even if this hasn't seen activitity in a while. My initial issue was that OpenCON did not provide me with good feedback about why the plugin manager showed me no plugins (which is an issue in itself), but from reading up on the code and the metadata, it seems to me that there are broader issues with the way OpenCPN decides if a plugin is compatible (which I think is the topic of this issue). Apologies if I mention things that have been previously discussed, I am not quite familiar with OpenCPN development (but hopefully an outsider view can be helpful as well...).

 1. I'm slightly worried that there is too much complex logic in the code right now. There's a  bunch of exception for various platforms, there's some code that hardcodes which Ubuntu and Debian version are compatible, recently a change was made that makes Ubuntu 20.04 plugins installable on 22.04, and #2702 adds a notice for yet another specific case. I would argue that a lot of this stuff should (ideally) not be in the code at all, but expressed in the downloaded metadata. For example, IIUC #2702 gives a fixed warning on aarch64 and arm64 that no plugins are available and to use flatpak instead. But as mentioned in https://github.com/OpenCPN/OpenCPN/issues/2694#issuecomment-1132223897 "*When a new version of OpenCPN is released, it's up to these communities to create or not to create plugins for various platforms. This is dynamic, and not much could be said in general.*", so I can imagine that in the future, maybe aarch64 *will* be natively supported by OpenCPN plugins (all of them, or maybe just one of them, if that's possible). If that happens, the hardcoded message will no longer be correct, and things will become *more* confusing, rather than less.
 2. More generally, the fact that all these exceptions are actually needed, might be a sign that the way plugin compatibility is expressed might not actually be ideal. I suspect distribution+arch+version (which I think is roughly what is used now) is not the real constraint for compatibility, but just an imperfect stub for the real constraint (which is probably libc and other library versions?). This is probably hard to figure out and implement in the ideal way, so using distribution+arch+version could very well be an acceptable compromise, if the user has enough info and flexibility to fix any imperfections in the matching.
 3. Maybe on easy improvement would be to allow a single plugin tarball to be installed on multiple platforms (i.e. have multiple sets of distribution+arch+version attributes for a single plugin tarball). This would allow expressing the debian-compatible-with-Ubuntu exception entirely server-side, and also the Ubuntu 22.04-is-compatible-with-20.04 exception that was recently added. I suspect this can even be implemented right now server-side without requiring client-side code changes by simply duplicating the entry for a given tarball a few times (at the cost of extra duplication in the XML).
    If the code generating the XML on the server is passed the OpenCPN client version (not sure if this is how it works now, but could be made to work like that), you could even change the XML format to not have this duplication, but dynamically generate a version with the duplication for older OpenCPN versions. (On a related note, unhindered by an understanding of how the plugin publishing process currently works, while revising the XML format, I would consider also combining multiple tarballs of the same plugin (version) into a single XML entry, to remove some duplication of e.g. plugin description, url, etc.).
 4. As a generic way to fix the user-confusion about missing plugins, maybe the plugin manager could simply show *all* plugins, not just the one supported, but mark them as "not supported by the current platform" or something like that (probably not by default, but maybe add a "Show unsupported plugins" checkbox or so to avoid clutter). This would give direct feedback to the user - the metadata is correctly downloaded, the plugins are known to the manager, but somehow their system is not supported. One step further could be to allow viewing the list of platforms that *are* supported (which could help users figure out that they might want to switch to flatpak) and one final step could be to support force-installing a plugin for a different platform (which would e.g. have allowed installing a 20.04 plugin on a 22.04 system (before this was officially supported), or be helpful for users of some Debian-derived or other Linux distro that OpenCPN cannot manage to derive compatibility for.
 5. One final thought is that there might be a distinction between "Built for/on", "Compatible with" and "Supported on" a given platform (I'm not saying that this distinction has to be codified or formalized, but I think that these help with thinking about plugin compatibility). "Built for/on" is very objective - it just means that that particular platform was used to build a plugin, so it should be expected to install and work on that same platform (at least not run into library version issues and missing symbols - it does not necessarily mean the plugin was *tested* and actually *works* on that platform). "Compatible with" would extend this concept to other platforms, where for example a 20.04 plugin would be compatible with 22.04 too (which I can image is applied in a blanket "every 20.04 plugin is compatible with 22.04" way, or on a per-plugin-version basis, or maybe based on automated testing, etc.). Finally "Supported on" would be the strongest label, meaning the plugin author actually considers this a supported platform (meaning they have tested the plugin version on this platform, know that there are sufficient users that do so, or at least that they are in a position to provide support for this platform if it does not work).

leamas, 2022-05-20:
Here are many, many spinning wheels. In order to stop some of them, the current thinking is:
  - For Linux, we only support Debian/Ubuntu/Raspbian and Flatpak (full stop).
  - I agree that the current logic is too complex. Hence I filed #2492, which turned into #2502 in particular the [roadmap](https://github.com/OpenCPN/OpenCPN/discussions/2502#discussioncomment-1904180). The conclusion here is that we should support upstream Debian plugins in Ubuntu and Raspbian.  The current mess is a  transition state where we support both old and new configurations. It is planned to be cleaned up in next version.
  - The plugin compatibility is a simple string. The plugin system does not handle dependencies, and we want to keep it that way.
  - Making it possible for plugin to be installed on multiple platforms is IMHO just too complex, besides the use of upstream Debian on downstream distros. One aspect  is the way the already too big test matrix will explode.
  -  Some sort of feedback on the total number of plugins might be a good idea. Another variant could be a line like _25 plugins available, 3 on this platform ubuntu-22.04-x86_64_ or so. Corresponding info could be displayed when loading the catalog to avoid cluttering the general UI
  - We should IMHO not mess up the compatible thing, we should stick to the KISS principle.  Messing with terms basically like "Really compatible" and "Perhaps compatible" will create all sorts of problems.

One a sidenote, there is a functionality under the Settings button in the plugin panel where a user could lie about the platform compatibility and try to load plugins for other platforms. Since this by definition risks de-stabilizing things, it''s hidden behind some extra steps. This menu also displays the number of plugins available for each platform in the Platform drop-down.