OpenCPN/OpenCPN #2308: flatpak: Access to local files in flatpak e. g., Help.html) fails.

leamas, 2021-06-15:
Discussion start: https://github.com/OpenCPN/OpenCPN/issues/2306#issuecomment-861351620

The core issue seems to be the flatpak sandbox. The basic principle is that the flatpak'ed application (here: Logbook) cannot access host resources like firefox, LibreOffice etc. This also is goes the other way around: the host program like firefox cannot access local data in in sandbox like the Help.html file.

There is a loophole: the sandbox has a xdg-open program which does the same as the "normal" xdg-open i. e., it opens a file. This way a flatpaked application can open a firefox tab on a URI.  The   wxLaunchDefaultBrowser(), the standard wxWidgets tool to open an url copes with this, so normally opening an url from flatpak is no problem.

However, this does not work with local data like Help.html since the host firefox application cannot see the local data.  Of course, we could publish these files on some url and thus have them available. But OpenCPN is designed to work off-line, so this path must be avoided.

I will try to look into this

leamas, 2021-06-15:
Closíng, right bug in wrong place. Moved to https://github.com/rgleason/LogbookKonni_pi/issues/30

closed 2021-06-15 by leamas

rgleason, 2021-06-15:
What? Isn't this a problem for OpenCPN too, if you have html help files?

leamas, 2021-06-15:
Nothing reported. Lets discuss it in the LogBook context for now until we know how to  handle it.