OpenCPN/OpenCPN #2309: ci: macos: Evil, unstable library hacks.

leamas, 2021-06-15:
As of 6273e29d17258d7794a4e9d521b10928a42ca453, the ci script installs some libraries besides this installed by the regular builds process. The need of such hacks comes from libraries which are not properly linked to the application. They are evil because they contains hard-coded  references which might fail when libraries are updated.

We need to properly link these libs and get rid of the hacks.

CarCode, 2021-06-15:
I see in the Frameworks folder of the original last OCPN version 55 libraries, some of them as symbolic links:
libX11.6.dylib
libXau.6.dylib
libXdmcp.6.dylib
libXext.6.dylib
libXrender.1.dylib
libarchive.13.dylib
libb2.1.dylib
libcairo.2.dylib
libexif.12.dylib
libfontconfig.1.dylib
libfreetype.6.dylib
libintl.8.dylib
liblz4.1.9.3.dylib
liblz4.1.dylib
liblzma.5.dylib
libpixman-1.0.dylib
libpng16.16.dylib
libwx_baseu-3.1.2.0.0.dylib
libwx_baseu-3.1.dylib
libwx_baseu_net-3.1.2.0.0.dylib
libwx_baseu_net-3.1.dylib
libwx_baseu_xml-3.1.2.0.0.dylib
libwx_baseu_xml-3.1.dylib
libwx_osx_cocoau_aui-3.1.2.0.0.dylib
libwx_osx_cocoau_aui-3.1.dylib
libwx_osx_cocoau_core-3.1.2.0.0.dylib
libwx_osx_cocoau_core-3.1.dylib
libwx_osx_cocoau_gl-3.1.2.0.0.dylib
libwx_osx_cocoau_gl-3.1.dylib
libwx_osx_cocoau_html-3.1.2.0.0.dylib
libwx_osx_cocoau_html-3.1.dylib
libwx_osx_cocoau_media-3.1.2.0.0.dylib
libwx_osx_cocoau_media-3.1.dylib
libwx_osx_cocoau_propgrid-3.1.2.0.0.dylib
libwx_osx_cocoau_propgrid-3.1.dylib
libwx_osx_cocoau_qa-3.1.2.0.0.dylib
libwx_osx_cocoau_qa-3.1.dylib
libwx_osx_cocoau_ribbon-3.1.2.0.0.dylib
libwx_osx_cocoau_ribbon-3.1.dylib
libwx_osx_cocoau_richtext-3.1.2.0.0.dylib
libwx_osx_cocoau_richtext-3.1.dylib
libwx_osx_cocoau_stc-3.1.2.0.0.dylib
libwx_osx_cocoau_stc-3.1.dylib
libwx_osx_cocoau_webview-3.1.2.0.0.dylib
libwx_osx_cocoau_webview-3.1.dylib
libwx_osx_cocoau_xrc-3.1.2.0.0.dylib
libwx_osx_cocoau_xrc-3.1.dylib
libxcb-render.0.0.0.dylib
libxcb-render.0.dylib
libxcb-shm.0.0.0.dylib
libxcb-shm.0.dylib
libxcb.1.1.0.dylib
libxcb.1.dylib
libzstd.1.4.9.dylib
libzstd.1.dylib
However using mostly the same code of OCPN in my version compiled with Xcode and not "ci" or such things I see only 18 libraries:
Sparkle.framework
libcairo.2.dylib
libfontconfig.1.dylib
libfreetype.6.dylib
liblzma.5.dylib
libpixman-1.0.dylib
libpng16.16.dylib
libwx_baseu-3.1.dylib
libwx_baseu_net-3.1.dylib
libwx_baseu_xml-3.1.dylib
libwx_osx_cocoau_adv-3.1.dylib
libwx_osx_cocoau_aui-3.1.dylib
libwx_osx_cocoau_core-3.1.dylib
libwx_osx_cocoau_gl-3.1.dylib
libwx_osx_cocoau_html-3.1.dylib
libwx_osx_cocoau_qa-3.1.dylib
libwx_osx_cocoau_webview-3.1.dylib
libwx_osx_cocoau_xrc-3.1.dylib
And this version compiled with Xcode works fine.
So what the hell people are doing here? I always suggest to use the tools the system offers and not strange foreign tools. When a software should run with a special OS this OS tools should be used and not any other stuff.
BTW the Sparkle framework is a far better solution for updates than these crasy stuff used by the original OCPN which is still broken.

leamas, 2021-06-15:
In what way is is your comment related to  the original issue here?

CarCode, 2021-06-15:
If that is all what you like to say I'm sorry to have disturb you.

PR #2324 (2021-08-02): ci: macos: Fix evil hack
Macos build: properly link libzstd and libpixman and remove evil hack.

Closes: #2309 

Build logs [here](https://app.circleci.com/pipelines/github/leamas/OpenCPN/1438/workflows/0a1e4e1e-28ef-4482-9a07-32073cb37855/jobs/2579)
closed 2021-08-02 by bdbcat