OpenCPN/OpenCPN #2310: Flatpak Plugin Testing:  Too many security restrictions

rgleason, 2021-06-15:
Ubuntu 18.04  
Cannot copy Plugin tarballs from Desktop to 
Home / .var /app / org.opencpn.OpenCPN / config / opencpn / plugins /cache / tarballs/

not permitted!!!

So where can I copy tarballs so that Option > Import Plugin"  will work in Flatpak????

Otherwise there is no way to quickly test a plugiin.

There has to be some designated "open" directory that permits me to copy files into it that can be used by OpenCPN to load the plugin.

leamas, 2021-06-15:
You can put it anywhere, just not under .var (why would you want to use that directory?).

I usually put my tarballs directly in my home directory and imports them from there.

rgleason, 2021-06-15:
Ok, Desktop or Home should work. Thanks.

closed 2021-06-15 by rgleason