OpenCPN/OpenCPN #2312: GetpPrivateApplicationDataLocation() semantics causes bad bugs.

leamas, 2021-06-17:
For some reason the `GetpPrivateApplicationDataLocation()` returns a pointer to a writable string.  This causes bad, hard to debug bugs bugs where one plugin might affect other plugins.

A simplistic approach for a plugin to get the data directory could be something like this:

```
    wxString s = wxFileName::GetPathSeparator();
    wxString stdPath  = *GetpPrivateApplicationDataLocation();
    stdPath += s + _T("plugins") + s + _T("watchdog");
    return stdPath;
```

This works from the perspective of the plugin. However, after executing this it has actually updated  `m_PrivateDataDir`in core OpenCPN  `OCPNPlatform.cpp `. This will break next plugin which tries to use this function. We just can't have it like this.

One remedy might be to add a new function with proper semantics. However, this will not be available for some time. We should probably still do it, though.

Another possible route would be to update declaration in ocpn_plugins.h like  (simplified patch):

```
diff --git a/include/ocpn_plugin.h b/include/ocpn_plugin.h

--- a/include/ocpn_plugin.h
+++ b/include/ocpn_plugin.h
-extern "C"  DECL_EXP wxString *GetpPrivateApplicationDataLocation();
+extern "C"  DECL_EXP const wxString *GetpPrivateApplicationDataLocation();
```

Adding  `const` to the declaration would effectively block any changes to the private data. It's simple, but breaks our promise to never change an existing API. 

Thoughts?

bdbcat, 2021-06-21:
Not seeing this happening, in o-charts.  m_privateDataDir is not altered after the lines around 497.

```
      g_PrivateDataDir = *GetpPrivateApplicationDataLocation();
      g_PrivateDataDir += wxFileName::GetPathSeparator();
      g_PrivateDataDir += _T("o_charts_pi");

```What am I missing?
>

leamas, 2021-06-21:
You ar emissing the simple fact that I need some form of day-care, again. This issue is just utterly confused, sorry for the noise

closed 2021-06-21 by leamas