OpenCPN/OpenCPN #2314: Feature request: Authenticate to Signal K

johansolve, 2021-06-18:
To be able to get n2k data via Signal K api into OpenCPN on port 3000, public readonly access must be enabled in Signal K. But this exposes very much on the web interface of Signal K without logging in, which is not desirable when Signal K web interface is publicly accessible.

Suggestion: Add support to authenticate with Signal K for access to data. Either with username/password och even better by requesting an application token. 

Related Signal K issue: https://github.com/SignalK/signalk-server/issues/1326

seandepagnier, 2021-07-04:
This is needed anyway for write access unless I am mistaken once opencpn can send signalk as well.

nohal, 2024-02-13:
Token based authentication is now implemented together with TLS encryption.