OpenCPN/OpenCPN #2315: Open CPN plugin manager can download pypilot

cornelledepag, 2021-06-19:
The plugin manager of opencpn is unable to get pypilot returning an error message containing the following message:

"Download error: The requested URL returned error:404 please check system log for more info."

I am assuming that the link in the plugin manager is incorrect, or the destination repository has some issues.

Fault has been constant using terminal to install the plugin works but plugin does not become visible in open cpn and also not possibility to add it through plugin manager. I am trying to find the tar.gz file for manual installation, with no success.

bdbcat, 2021-06-19:
What platform, please?
Windows, linux??
Thanks
Dave

cornelledepag, 2021-06-20:
Hi,

I have used the lysmarine of BBN and openplotter both on the raspberry. Linux

leamas, 2021-06-20:
This is just the tip of the iceberg. Checking all urls in current master shows broken urls in
  - aisradar_pi (1)
  - plots_pi(2)
  - pypilot_pi (5)
  - test_plugin_pi(15)
  - watchdog_pi (1)
  - weather_routing_pi(1)
  - weatherfax_pi (1)
  
Attaching the filtered log from `for f in metadata/*.xml; do echo "======> Checking $f" >&2 ; tools/check-metadata-urls $f; done  |& tee urlcheck.log`

[url-errors.log.gz](https://github.com/OpenCPN/OpenCPN/files/6681906/url-errors.log.gz)

We need to file downstream bugs in the plugins.

leamas, 2021-06-20:
@rgleason : it is not possible to file issues against your plots_pi repo. plots_pi has a broken download linkt for ubuntu-x86_64-18.04-bionic, see above

leamas, 2021-06-20:
@rgleason: Same for pypilot_pi, it is not possible to file issues. Here, the metadata source url is also broken.

leamas, 2021-06-20:
@cornelledepag: Thanks for reporting! That said and FYI this bug is really about the catalog, and such bugs should actually be filed against the plugins project at https://github.com/opencpn/plugins. Eventually this bug might be closed and transferred to this repo. 

Still, the important thing is to have this reported. Thanks again"

rgleason, 2021-06-20:
It appears that seven plugins have bad metadata urls for flatpak and sometimes Ubuntu bionic.  I will see if I can fix these by hand and  run link validation.

cornelledepag, 2021-06-20:
@leamas

I am quite new with github and pypilot and was unsure where to post.

I think the problem will be solved as I saw that you posted the problem in the proper place. Is see options to build plugins for Mac and windows. Is it possible to download the package manually?

If not then I will wait and first will focus on getting the arduino nano ready to receive instructions from pypilot.

rgleason, 2021-06-21:
For pypilot, try this [Cloudsmith repository](https://cloudsmith.io/~opencpn/repos/pypilot-prod/packages/) download the most recent "msvc"  tarball.  Then use OpenCPN  > Options > Plugins   "Import plugin"

Let me know OS and if it works please.

leamas, 2021-06-21:
@rgleason: Please note that @cornelledepag is on linux/raspbian  as of the [comment above](https://github.com/OpenCPN/OpenCPN/issues/2315#issuecomment-864481662)

Looking in that cliudsmith repo, is there any armhf build available for @cornelledepag ?

rgleason, 2021-06-22:
Sorry, I was off the wall. you are right. I need day care too..  no rpi armhf there.   ...yet

rgleason, 2021-06-23:
**Fixed & pushed to Plugins**
- pypilot_pi (5)
- watchdog_pi (1)    (need android builds though)

**Yet to do**
- aisradar_pi (1)
- plots_pi(2)
- test_plugin_pi(15)
- weather_routing_pi(1)
- weatherfax_pi (1)

cornelledepag, 2021-06-23:
@rgleason When do you expect the new change to be available for update?

Indeed, cloudsmith doesn't have any armhf packages available. Thanks for the effort, any suggestions how I can help?

rgleason, 2021-06-23:
Didn't you look here?  https://cloudsmith.io/~opencpn/repos/pypilot-prod/packages/

rgleason, 2021-06-29:
Fixed and pushed to plugins PIM

closed 2021-06-29 by rgleason