OpenCPN/OpenCPN #2316: RFE: AIS. MMSI Followers have own icon

rgleason, 2021-06-20:
See https://github.com/OpenCPN/OpenCPN/issues/2277

Mmsi ships designated as "follower" in the mmsi list have CPA turned OFF. Therefore they should have a unique icon.

rgleason, 2021-07-04:
Proposal regarding the icon to be used:

It should be [normal colors](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais) 

1. red= Danger
2. yellow= No danger. Not identified
3. green= No danger. Identified or Not Identified (Name from cached list)

Then the icon should have a "V" (or peace symbol) similar to a broken "Lost target" icon  coming out of the back of it.

This is appropriate because CPA has been turned off under "Follower" (Friend) function.
Followers (no CPA) can then be identified by the "V" in the symbol.
Then typical  red, yellow, green colors would remain in effect.

An improvement would be to adopt the GPSbuddy shape, but add the "V" off the back for "Followers"
Keeping the 3 color color scheme.

I think this is what I'll draw.

rgleason, 2021-07-04:
[Follower-SVG.zip](https://github.com/OpenCPN/OpenCPN/files/6759840/Follower-SVG.zip)
[image]
[image]
[image]

bdbcat, 2021-07-08:
Rick...
Not sure what you mean regarding "danger" for followers, since we never alert on CPA incursion.  Do you mean that we should change the color (green->red) if a CPA incursion would occur, even though we never alert?
And what exactly is "yellow" to signify?
Thanks
Dave

rgleason, 2021-07-08:
Do you mean that we should change the color (green->red) if a CPA incursion would occur, even though we never alert? 
-Yes I think that would be very prudent, after all the alarm is turned off for "Followers".

Under normal circumstances, yellow is used for a change in status such as:
1. No Danger. Not Identified (Voyage data not received)   
2. Lost Target

For Followers, a delay in receiving a new updated location would probably cause the color to change to yellow wouldn't it? 
However you probably have other equally important thoughts about this.  Thanks.

Here are the symbols for reference
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:toolbar_buttons:ais#ais_target_quick_graphic_references

rgleason, 2021-07-08:
Dave, I had a thought, do you think the colors for green and red  "Followers" should be less intense than the normal icon colors, thus making the Follower icons  less dominant on the screen?

bdbcat, 2021-07-09:
Rick...
We do not use SVG for AIS icons.  We render them in raw stroke graphics, for performance reasons.  All I need to do is add the "inverted v" line strokes to GPSG Buddy symbol rendering.
I'll use the standard colors for now, and we will get some testing done.  I suppose we will need a VDR file with some follower(s) in the record to test this.

rgleason, 2021-07-09:
Great, Ok I'll put together a vdr with followers for a test.

rgleason, 2021-07-09:
I picked a very active AIS file and added a bunch of MMSI followers to the MMSI list..
I hope that is where this is stored, because I inspected the file and did not find any indicatior.
I included a list of mmsi followers too.

[Follower-Files.zip](https://github.com/OpenCPN/OpenCPN/files/6788338/Follower-Files.zip)

closed 2021-07-09 by bdbcat

bdbcat, 2021-07-09:
Code is in github master now.
Try a build, and see what you think.  Works OK for me.

rgleason, 2021-07-09:
Thanks! Will give it a try.

rgleason, 2021-07-09:
Its working after second fetch and build.

This test is using gpsbuddy icon + Follower broken Tails  which is what we had decided.
[image]

This test seems to be using standard icon + Follower broken tails  (it seems to show up pretty well, but  I wonder why it is using the  normal green icon.
[image]

I tried to see if the red version was working but did not find a good situation, it is implemented?

bdbcat, 2021-07-09:
Something wrong with the second image. Followers should always use the "buddy" icon.
Is this using OpenGL?  What OS?  What other AIS settings are set?

rgleason, 2021-07-09:
Windooze 10     Use OpenGL checked,   Options Texture compression, Polygon and Line smoothing on.
AIS settings  see screenshots please

[image]
[image]
[image]

bdbcat, 2021-07-10:
Rick...
OK, I see the problem.  Slight tweak required.
In github master now.  Please test/confirm.
Thanks

rgleason, 2021-07-10:
Shows the GPS buddy + Follower Tails
[HDG_test.txt](https://github.com/OpenCPN/OpenCPN/files/6794575/HDG_test.txt)

Shows the standard green icon + Follower tails
[Hakefjord-Sweden-1m.txt](https://github.com/OpenCPN/OpenCPN/files/6794576/Hakefjord-Sweden-1m.txt)

A couple of nmea vdr files I've been using above.    I just added several ships to the mmsi list and checked follower.

[image]
[image]
[image]

rgleason, 2021-07-10:
TRying to figure out what determines which icon is used.

bdbcat, 2021-07-10:
Rick...
One more tweak in github now, correcting error: "standard green icon + Follower tails"
Thanks
Dave

Hakansv, 2021-07-10:
I can agree the Follower gets the same icon as a Buddy independent if it's a "A" or "B" class target. If anyone gives it the "Follower" tag the class will be known by the user.
I suppose the red mark for a Follower is intentional if within the alarm settings even though the alarm is of course not released.
So, I think this is a "as good as possible" solution for this issue. 
Håkan
Picture: BOB is a "A" and LEDA is a "B". For Leda no alarm but marked red.
[image]

rgleason, 2021-07-10:
@Hakansv  
Leda B  being red is good, is the icon a gpsbuddy type with blunt nose?

Hakansv, 2021-07-10:
LEDA's icon is the blunt nosed gpsbuddy type, yes. 

There's no special column for Followers in the AIS target list and I'm not convinced we need it either. 

Håkan

rgleason, 2021-07-10:
Yes, now I see it is blunt on the PC.  I think I agree, as long as the icons are legible as "Follower" no listing is needed.  I suppose if you hover it might be good to have "Follower" added.

rgleason, 2021-07-10:
I don't think adding "Follower" to hover is necessary.  The blunt nose gpsbuddy icon is being used everywhere in my tests.  I have not found a situation where it turns red like hakan found, but his test shows that it works.  I believe it is working.  Thanks Bdbcat!

Should we be testing yellow or not?

Hakansv, 2021-07-10:
Yellow, i.e. no name in the hash file, is working fine!

rgleason, 2021-07-10:
Good, thanks.  Will have that in manual then.

commit db39e4e6 (2021-07-09): Symbolize AIS "follower" targets uniquely. Closes #2316 Closes #2277