OpenCPN/OpenCPN #2318: feature request: iENC and ENC cells for ECDIS testing

Dirk--, 2021-07-10:
to my (limited) knowledge, ECDIS manufacturer should provide (i)ENC data for testing ...
Found these 3 sets, looking for the right place to integrate (github, wiki, elsewhere):

IHO Presentation Library Edition 3.4
ENC_CheckData (43° 49' N / 68° 59' E)
https://web.archive.org/web/20150604180744if_/http://www.iho.int/iho_pubs/ECDIS/ENC_CheckData.zip

IHO Presentation Library Edition 4.0
ECDIS_Chart_1 (15° 04' N / 5° 04' W)
https://support.navtor.com/support/solutions/articles/48000957435-chart1-iho-test-cells

IENC Presentation Library 2.3
ENC_ROOT (51° N / 9° 45' E)
https://firebasestorage.googleapis.com/v0/b/gitbook-28427.appspot.com/o/assets%2F-LpJCsGXU3njb9uabz2e%2F-LzCHWR6MCz39Kp6G9ie%2F-LzCJOHFgLmGiBvsAeg3%2FENC_ROOT.zip?alt=media&token=3d10b069-1f98-401c-9db6-0eb09ba0208e

rgleason, 2021-07-10:
I believe these should be linked to in the User Manual?   or perhaps it should be the Dev Manual too?
Making an issue at the Dev Manual, so it gets added.

CarCode, 2021-07-10:
Years ago when introducing the S63 plugin there have been published already test data (IHO S-64) which might be outdated now.

Dirk--, 2021-09-29:
Thanks to @rgleason :
https://opencpn-manuals.github.io/development/ocpn-dev-manual/5.3.1/ECDIS_testdata.html

closed 2024-02-17 by nohal