OpenCPN/OpenCPN #2319: iENC > data/s57data/chartsymbols.xml: BOYLAT26 and BOYLAT27 do not show up

Dirk--, 2021-07-10:
just investigated, why those two icons never show up ...

IENC23_lookupTables.zip > pienc23s.dic states:
```
"boylat","catlam15|COLOUR1,3,1,3|","SY(BOYLAT26);TX(OBJNAM,2,1,2,'14106',-1,-1,CHBLK,21)","8","O","DISPLAYBASE","17010"
"boylat","catlam15|COLOUR3,1,3,1|","SY(BOYLAT26);TX(OBJNAM,2,1,2,'14106',-1,-1,CHBLK,21)","8","O","DISPLAYBASE","17010"
"boylat","catlam16|COLOUR1,4,1,4|","SY(BOYLAT27);TX(OBJNAM,2,1,2,'14106',-1,-1,CHBLK,21)","8","O","DISPLAYBASE","17010"
"boylat","catlam16|COLOUR4,1,4,1|","SY(BOYLAT27);TX(OBJNAM,2,1,2,'14106',-1,-1,CHBLK,21)","8","O","DISPLAYBASE","17010"
```

chartsymbols.xml misses the the right colour code:
```
<lookup id="1337" RCID="31389" name="boylat">
			<type>Point</type>
			<disp-prio>Hazards</disp-prio>
			<radar-prio>On Top</radar-prio>
			<table-name>Simplified</table-name>
			<attrib-code index="0">catlam15</attrib-code>
			<attrib-code index="1">COLOUR1,3</attrib-code>
			<instruction>SY(BOYLAT26);TE('by %s','OBJNAM',2,1,2,'15110',-1,-1,CHBLK,21)</instruction>
			<display-cat>Standard</display-cat>
			<comment>17010</comment>
		</lookup>
		<lookup id="1338" RCID="31390" name="boylat">
			<type>Point</type>
			<disp-prio>Hazards</disp-prio>
			<radar-prio>On Top</radar-prio>
			<table-name>Simplified</table-name>
			<attrib-code index="0">catlam16</attrib-code>
			<attrib-code index="1">COLOUR1,4</attrib-code>
			<instruction>SY(BOYLAT27);TE('by %s','OBJNAM',2,1,2,'15110',-1,-1,CHBLK,21)</instruction>
			<display-cat>Standard</display-cat>
			<comment>17010</comment>
		</lookup>
```
reference: https://firebasestorage.googleapis.com/v0/b/gitbook-28427.appspot.com/o/assets%2F-LpJCsGXU3njb9uabz2e%2F-LzCHWR6MCz39Kp6G9ie%2F-LzCHvPV6spwAtXtu6nT%2FIENC23_lookupTables.zip?alt=media&token=a39a41d6-d972-40ec-a664-255ff3ff424f

nohal, 2021-07-10:
Can you please consider submitting pull requests for this stuff instead of just opening the issue?

Dirk--, 2021-07-10:
@nohal not as easy as expected: Do need help on:
`<lookup id="[]">` 
looks like this has to be unique, best way to automate in case of newly inserted objects?

my simple copy paste approach renders just COLOUR4,1,4,1 correct
```
		<lookup id="1338" RCID="31390" name="boylat">
			<type>Point</type>
			<disp-prio>Hazards</disp-prio>
			<radar-prio>On Top</radar-prio>
			<table-name>Simplified</table-name>
			<attrib-code index="0">catlam16</attrib-code>
			<attrib-code index="1">COLOUR1,4,1,4</attrib-code>
			<instruction>SY(BOYLAT27);TE('by %s','OBJNAM',2,1,2,'15110',-1,-1,CHBLK,21)</instruction>
			<display-cat>Standard</display-cat>
			<comment>17010</comment>
		</lookup>
		<lookup id="1338" RCID="31390" name="boylat">
			<type>Point</type>
			<disp-prio>Hazards</disp-prio>
			<radar-prio>On Top</radar-prio>
			<table-name>Simplified</table-name>
			<attrib-code index="0">catlam16</attrib-code>
			<attrib-code index="1">COLOUR4,1,4,1</attrib-code>
			<instruction>SY(BOYLAT27);TE('by %s','OBJNAM',2,1,2,'15110',-1,-1,CHBLK,21)</instruction>
			<display-cat>Standard</display-cat>
			<comment>17010</comment>
		</lookup>	
```

nohal, 2021-07-10:
The `id` and `RCID` attributes must be unique. The `id` is a simple incrementing integer, that is the easy part - pick a number one bigger than the `id` of our last existing `lookup` object, the `RCID` comes originaly from the S52 lookup definitions, of which we have a dated version in https://raw.githubusercontent.com/OpenCPN/OpenCPN/master/data/s57data/S52RAZDS.RLE
As we do not have access to the current versions of the tables, if they exist at all, the best you can do is to invent a for now unused RCID and hope there won't be a conflict in the future.
The objects are loaded serially, so reusing a `RCID` of a previously existing object means you replace it - in your original PR, the second definition would simply replace the first one loaded just before it.

Dirk--, 2021-07-11:
Having unique `id` and `RCID` makes user edits (in context of additions) nearly impossible ... and there is some more missing compared to IENC23_lookupTables [1] or even IES_2_4_LookupTables [2].
@bdbcat @JesperWe @hein-nebel How do you guys handle this?
[1] https://editions.openecdis.org/v/edition-2.3/standard-iecdis/standard-2.3/status-of-the-ienc-presentation-library-2.3/look-up-tables-2.3
[2] https://editions.openecdis.org/standard-iecdis/standard-2.4/status-of-the-ienc-presentation-library-2.4/look-up-tables-2.4

Dirk--, 2021-07-11:
indeed, simple renumbering seems to work:
```
<xsl:stylesheet version="1.0" xmlns:xsl="http://www.w3.org/1999/XSL/Transform">
<xsl:output method="xml" version="1.0" encoding="UTF-8" indent="yes"/>
<xsl:strip-space elements="*"/>

<!-- identity transform -->
<xsl:template match="@*|node()">
    <xsl:copy>
        <xsl:apply-templates select="@*|node()"/>
    </xsl:copy>
</xsl:template>

<!-- renumber lookup id -->
<xsl:template match="lookup/@id" >
    <xsl:attribute name="id">
         <xsl:variable name="n">
          <xsl:number count="lookup" level="any"/>
        </xsl:variable><xsl:value-of select="$n - 1"/>
    </xsl:attribute>
</xsl:template>

<!-- renumber lookup RCID -->
<xsl:template match="lookup/@RCID" >
    <xsl:attribute name="RCID">
        <xsl:text>99</xsl:text>
          <xsl:number count="lookup" level="any" format="001"/>
    </xsl:attribute>
</xsl:template>
</xsl:stylesheet>
```
https://github.com/Dirk--/OpenCPN/blob/master/data/s57data/chartsymbols.xml

nohal, 2021-07-11:
Be VERY careful with this, if you study the file, you will see that there already are intentional rules that replace older versions with the same RCID. You will also destroy the reference to the original IHO data, which will make our life extremely hard if we ever get access to a newer version of it.

nohal, 2021-07-11:
And to your question, I personaly handle it with `grep`, `cut` and `sort`

Dirk--, 2021-07-11:
> Be VERY careful with this, if you study the file, you will see that there already are intentional rules that replace older versions with the same RCID. You will also destroy the reference to the [...]

so how did you design <lookup RCID> in the first place?

Right now, I would like to update inland ECDIS point features (simplified, Edition 2.3) only and imagine, renumbering RCID=23001 upwards will do no harm ....

bdbcat, 2021-07-11:
Dirk...
Please do no renumber the existing RCID's.
You may add records that have unique RCID anywhere in the file.  I suggest simply adding your records starting at some odd number like 57000, and verify that the RCIDs you select are not used elsewhere.  This will work, and is the safest route today.

Background:  RCID is a simple integer used as a hash key in the lookup matching logic.

Some day, we may rebuild this file, incorporating the most current S52 PLIB standards.  But now is not the time...

Dirk--, 2021-07-12:
not exactly on topic:
Just summed up my knowledge: https://wiki.opennauticalchart.org/index.php?title=Category:Data_Sources#ECDIS
Feedback welcome!

Dirk--, 2021-07-12:
here we go

- Edition 4.0 https://vdocuments.mx/annex-a-to-s-52-iho-ecdis-presentation-library-ed-40-a-to-s-52-iho-ecdis-presentation.html
- draft https://legacy.iho.int/iho_pubs/draft_pubs/PresLib_e4.0.0/Part_I/S-52_PresLib_e4.0.0_Part%20I_Redline.pdf
- DAI https://github.com/alios/ecdis/blob/master/config/PresLib_e4.0.0.dai

Dirk--, 2021-07-13:
> Background: RCID is a simple integer used as a hash key in the lookup matching logic.

David, is the hash function  `PresLib.dai > chartsymbols.xml` documented somewhere?

nohal, 2021-07-13:
No, you have to read the code to understand how all this works.

Dirk--, 2021-07-23:
review goes on, more complex then expected!
Set up an extra-repository: https://github.com/Dirk--/openCPN-ECDIS
Hope this helps, Dirk

Dirk--, 2021-08-03:
#2325

closed 2021-08-03 by Dirk--