OpenCPN/OpenCPN #2321: Wind direction arrows in the Grib weather table

bc38, 2021-07-14:
Wish for improvement
The presence of the wind direction arrow will be visually clearer than the direction value in degrees

[image]
[image]

tai-fun, 2021-07-14:
That feature is already there. Click on the word "Dir" and the wind
direction (295) should change to an arrow.
This is an old feature.
BUT there is a hiccup. The arrow does not appear, as it used to do when the
number disappears. The box is all white. If the arrow is there it's white
or transparent.
There is no difference between 5.2.4 and the latest (today) git. Ubuntu
20.04.

bc38, 2021-07-14:
> Эта функция уже есть. Нажмите на слово «Dir», и направление ветра (295) изменится на стрелку. Это старая функция. НО есть икота. Стрелка не появляется, как раньше, когда исчезает число. Коробка вся белая. Если есть стрелка, она белая или прозрачная. Нет разницы между 5.2.4 и последним (на сегодняшний день) git. Ubuntu 20.04.
> […](#)

Thanks for the info!
Works
The function is hidden. It would be better to modify to make the arrow visible all the time and to the right of the arrow the value in degrees. Also, in places where the wind is 0, an arrow is still displayed and there should be a circle (calm)

bc38, 2021-07-14:
> That feature is already there. Click on the word "Dir" and the wind direction (295) should change to an arrow. This is an old feature. BUT there is a hiccup. The arrow does not appear, as it used to [...]
Thanks for the info!
Works
The function is hidden. It would be better to modify to make the arrow visible all the time and to the right of the arrow the value in degrees. Also, in places where the wind is 0, an arrow is still displayed and there should be a circle (calm)

tai-fun, 2021-07-14:
As it works for you, I have to assume I'm hit by one more GtK3 hiccup.

rgleason, 2021-08-02:
I think tai-fun has a good point.
 `It would be better to modify to make the arrow visible all the time and to the right of the arrow the value in degrees. Also, in places where the wind is 0, an arrow is still displayed and there should be a circle (calm)`

closed 2021-11-14 by bc38