OpenCPN/OpenCPN #2322: OpenCPN Flatpak  failing.  Ubuntu 20.04

rgleason, 2021-07-21:
Prior installation was 18.04, upgrade it to 18.0X  as directed. Then had difficulty upgrading to 20.04 and finally got it upgraded.  Have reinstalled the Opencpn Dev setup. and am now trying to get the separate flatpak installation working.

Have removed flatpakk opencpn.   Tried to update and install flatpak, but the installation was recognized. Shutdown the Virtual box.  Started again.  Installed Opencpn.  It keeps popping a message that plugins are incompatible. and I select to simplify or uninstall those plugins. Then tried the other way. No difference. It allways fails and the screen disappears.

Then removed Opencpn using the flatpak button for Remove"

```
```
gleason@rgleason-VirtualBox:~$  flatpak run org.opencpn.OpenCPN
ALSA lib dlmisc.c:226:(snd_dlsym_verify) unable to verify version for symbol _snd_pcm_dsnoop_open
ALSA lib dlmisc.c:348:(snd_dlobj_cache_get0) symbol _snd_pcm_dsnoop_open is not defined inside [builtin]
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
Gtk-Message: 14:29:27.702: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:29:27.703: Failed to load module "canberra-gtk-module"

(opencpn:2): Gdk-ERROR **: 14:29:36.752: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 4571 error_code 2 request_code 131 (MIT-SHM) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
rgleason@rgleason-VirtualBox:~$ 
```

Any suggestions
```

rgleason, 2021-07-21:
Found this[ possible answer](https://askubuntu.com/questions/342202/failed-to-load-module-canberra-gtk-module-but-already-installed)   and  [this one aboutt 20.04 Ubuntu](https://askubuntu.com/questions/1306558/failed-to-load-module-canberra-gtk-module-proxychains)

```
install gtk and gtk3 module should solve it.
sudo apt install libcanberra-gtk-module libcanberra-gtk3-module

```

We have  gtk-3.0 libs why now need libcanberra??? 
Does this have something to do with 32 bit 64bit?

rgleason, 2021-07-21:
(opencpn:2): Gdk-ERROR **: 14:29:36.752: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.  The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 4571 error_code 2 request_code 131 (MIT-SHM) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.

```
**debug your program, run it with the GDK_SYNCHRONIZE environment
variable to change this behavior.** 
```

bdbcat, 2021-07-22:
Rick...
More info needed.
1.  What is linux version on the virtual?  Ubuntu??
2.  Please show the output of $flatpak list
3. "Prior installation was 18.04"  Was that working OK?
4. " upgrade it to 18.0X as directed."   By whom, and what were the directions?
5. "Have reinstalled the Opencpn Dev setup."  You do not need that for simple running/testing plugins.

Dave

rgleason, 2021-07-22:
Dave,
1. Ubuntu 20.04
2.  See below please
3. Yes Flatpak for 18.0x  was working.  (However the entire setup was slower.)
4.  Before I could upgrade to 20.04  (message pop up)  it wanted to update 18.04 to 18.08. Which I did. The entire process was very slow and one of my virtualbox ubuntu failed entirely. The other one made it through. I then double checked that it was a full and complete installation.  Its faster than 18.04 was and seems very stable.
5.  Yes,  but want to be able to build beta Opencpn on Ubuntu.

```
rgleason@rgleason-VirtualBox:~$ flatpak list
Name          Application ID              Version         Branch Installation
Freedesktop … org.freedesktop.Platform    20.08.14        20.08  system
Mesa          …esktop.Platform.GL.default 21.1.4          20.08  system
openh264      …edesktop.Platform.openh264 2.1.0           2.0    system
Ambiance Gtk… org.gtk.Gtk3theme.Ambiance                  3.22   system
Yaru Gtk The… org.gtk.Gtk3theme.Yaru                      3.22   system
OpenCPN       org.opencpn.OpenCPN         5.2.4-0+6b314e6 stable system
rgleason@rgleason-VirtualBo

```

Thanks for looking at this

bdbcat, 2021-07-22:
So, if you run:
$flatpak reun org.opencpn.OpenCPN//stable
What happens?

I'm pretty sure that "5.2.4-0+6b314e6'" is the 18.08 build.  See my other note:
https://github.com/OpenCPN/OpenCPN/discussions/2251#discussioncomment-1017376
Waiting for Alec to return....

rgleason, 2021-07-22:
```
gleason@rgleason-VirtualBox:~$ flatpak run org.opencpn.OpenCPN//stable
ALSA lib dlmisc.c:226:(snd_dlsym_verify) unable to verify version for symbol _snd_pcm_dsnoop_open
ALSA lib dlmisc.c:348:(snd_dlobj_cache_get0) symbol _snd_pcm_dsnoop_open is not defined inside [builtin]
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib setup.c:547:(add_elem) Cannot obtain info for CTL elem (MIXER,'IEC958 Playback Default',0,0,0): No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
Gtk-Message: 07:08:06.787: Failed to load module "canberra-gtk-module"
Gtk-Message: 07:08:06.787: Failed to load module "canberra-gtk-module"

(opencpn:2): Gdk-ERROR **: 07:08:17.625: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 2313 error_code 2 request_code 131 (MIT-SHM) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
rgleason@rgleason-VirtualBox:~$ 
```

Then

rgleason@rgleason-VirtualBox:~$ flatpak list
Name          Application ID              Version         Branch Installation
Freedesktop … org.freedesktop.Platform    20.08.14        20.08  system
Mesa          …esktop.Platform.GL.default 21.1.4          20.08  system
openh264      …edesktop.Platform.openh264 2.1.0           2.0    system
Ambiance Gtk… org.gtk.Gtk3theme.Ambiance                  3.22   system
Yaru Gtk The… org.gtk.Gtk3theme.Yaru                      3.22   system
OpenCPN       org.opencpn.OpenCPN         5.2.4-0+6b314e6 stable system
rgleason@rgleason-VirtualBox:~$

rgleason, 2021-07-22:
See  https://github.com/OpenCPN/OpenCPN/blob/master/flatpak/org.opencpn.OpenCPN.yaml

How does this get changed?

OpenCPN/flatpak/org.opencpn.OpenCPN.yaml 

Shouldn't this be a variable?

See https://github.com/jongough/testplugin_pi/blob/master/cmake/in-files/org.opencpn.OpenCPN.Plugin.yaml.in

```
app-id: org.opencpn.OpenCPN
runtime: org.freedesktop.Platform
runtime-version: 18.08
sdk: org.freedesktop.Sdk
command: opencpn.sh

```

leamas, 2021-07-25:
I cannot reproduce this on a reasonable clean, fully updated  focal/20.04 VM. Possibly something became terribly broken when your upgrade failed (?)

A cleaned up command history below.  Alll of these commands works without errors:

```
  502  flatpak uninstall --user org.opencpn.OpenCPN
  505  flatpak install --user org.opencpn.OpenCPN
  507  flatpak run org.opencpn.OpenCPN
  508  sudo apt-get update
  509  sudo apt-get upgrade
  510  flatpak run org.opencpn.OpenCPN
```

leamas, 2021-07-25:
Furthermore, I can downgrade to the 18.08 version and then start opencpn using commands below. Works fine, shows the complete list of 18.08 plugins. After upgrading (which works fine), just the very short list of 20.08 plugins is displayed,.

```
  512   flatpak update --user --commit 959f5fd700f72e63182eabb9821b6aa52fb12189eddf72ccf99889977b389447 org.opencpn.OpenCPN
  513  flatpak run org.opencpn.OpenCPN
  514   flatpak update --user org.opencpn.OpenCPN
  515  flatpak run org.opencpn.OpenCPN
```

rgleason, 2021-07-28:
Useless

closed 2021-07-28 by rgleason